openwrt/packages #9349: jool: Update Jool to the latest version (4.0.x)

tiagogaspar8, 2019-06-28:
Maintainer: maintainer was removed
Environment: all

Description:

I'd like to ask you guys to update jool to the latest version, I've researched and from the documentation it is compatible with the current kernel versions openwrt currently uses.
Also the jool documentation says the current version used (3.5.8) isn't compatible with 4.19 kernel.

Thanks!

neheb, 2019-06-28:
I tried doing so before but gave up. There are some build changes that need adjustment.

jool is currently unmaintained.

tiagogaspar8, 2019-07-02:
They gave up on jool?

ydahhrk, 2019-09-24:
Hello. Upstream Jool dev here.

Though I'm still maintaining Jool, its OpenWRT package does seem abandoned. I have the email of the [original maintainer](https://openwrt.org/packages/pkgdata_lede17_1/jool-tools) (not sure if it's public domain), so I will try to contact him shortly.

From my understanding of it, Dan Lüdtke was forced to perform some voodoo magic to adapt the then messy build setup to package the software. These quirks should have been removed during the "standardization" I was encouraged to do by [Debian](https://github.com/NICMx/Jool/issues/243), so packaging Jool 4.0.5 for OpenWRT should be hopefully simpler.

(Here's my [Debian package generator along with its debian directory](https://github.com/ydahhrk/packaging/tree/master/Jool), if it helps.)

tiagogaspar8, 2019-09-24:
Hi everyone! I have tried to compile jool by changing to the latest version but it gives out some errors.
I know nothing of compiling so as soon as I can I'll post the outputs here for you guys to see.

Thanks!

neheb, 2019-09-25:
Tried updating. I'm getting some linking error:

```
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: warning: libjoolutil.so.0, needed by ../../../src/usr/argp/.libs/libjoolargp.so, not found (try using -rpath or -rpath-link)
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: warning: libjoolnl.so.0, needed by ../../../src/usr/argp/.libs/libjoolargp.so, not found (try using -rpath or -rpath-link)
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `global_query'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `bib_add'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `blacklist4_foreach'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `json_parse'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `pool4_flush'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `blacklist4_add'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `pool4_rm'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `blacklist4_rm'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `str_to_prefix4'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `l4proto_to_string'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `str_to_prefix6'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `result_success'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `pool4_add'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `global_update'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `iname_validate'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `timeout2str'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `address_query46'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `eamt_add'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `str_to_addr4_port'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `bib_foreach'
[... cut ...] tr_to_addr4'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `blacklist4_flush'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_glibc/lib/gcc/mips-openwrt-linux-gnu/9.2.0/../../../../mips-openwrt-linux-gnu/bin/ld: ../../../src/usr/argp/.libs/libjoolargp.so: undefined reference to `str_to_addr6'
```

ydahhrk, 2019-09-25:
It's not finding the libraries...? Or maybe it's not linking them in the proper order?

Excluding external dependencies, these are the current binaries and their relationships:

| Name | What is it? | Makefile.am location | Direct Dependencies |
|------|-------------|----------------------|---------------------|
| `jool` | userspace client | `src/usr/nat64/` | libjoolargp |
| `jool_siit` | userspace client | `src/usr/siit/` | libjoolargp |
| `joold` | userspace daemon | `src/usr/joold/` | libjoolnl, libjoolutil |
| `libxt_JOOL_SIIT` | iptables shared object | `src/usr/iptables/` | - |
| `libxt_JOOL` | iptables shared object | `src/usr/iptables/` | - |
| `libjoolargp` | library | `src/usr/argp/` | libjoolutil, libjoolnl |
| `libjoolutil` | library | `src/usr/util/` | - |
| `libjoolnl` | library | `src/usr/nl/` | libjoolutil |

Note that the iptables Makefile is plain; it does not use the autotools.

For now and in the foreseeable future, the symbols of the libraries are not really needed to be visible outside of Jool. (In Debian, this concept is referred to as "private libraries," and is simpler to manage than public libraries. Not sure if an equivalent exists in OpenWRT?)

Perhaps it would be simpler to just compile the libraries statically? Just throwing out ideas.

I don't know much about the OpenWRT environment, and I can't do much without replicating, but if you can point me to the steps you're following, I can spend time to try and figure this out.

neheb, 2019-09-25:
Switched computers. Getting a different error now:

```
rm -f /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/.built
touch /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5=jool-4.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/include/fortify -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5=jool-4.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/include/fortify -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/include " LDFLAGS="-L/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/lib -znow -zrelro " make  -C /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5=jool-4.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[3]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5'
Making all in src/common
make[4]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/common'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/common'
Making all in src/usr
make[4]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr'
Making all in util
make[5]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr/util'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr/util'
Making all in nl
make[5]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr/nl'
/bin/bash ../../../libtool  --tag=CC   --mode=compile mips-openwrt-linux-musl-gcc -DPACKAGE_NAME=\"Jool\" -DPACKAGE_TARNAME=\"jool\" -DPACKAGE_VERSION=\"4.0.5\" -DPACKAGE_STRING=\"Jool\ 4.0.5\" -DPACKAGE_BUGREPORT=\"jool@nic.mx\" -DPACKAGE_URL=\"\" -DPACKAGE=\"jool\" -DVERSION=\"4.0.5\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_ARPA_INET_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_UNISTD_H=1 -DHAVE__BOOL=1 -DHAVE_INET_NTOA=1 -DHAVE_MEMSET=1 -DHAVE_STRCASECMP=1 -DHAVE_STRTOL=1 -DHAVE_POW=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -I.   -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/include/fortify -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/include  -Wall -pedantic [... cut ...]
make[2]: *** [/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/.built] Error 2
make[2]: Leaving directory '/home/mangix/devstuff/openwrt/feeds/packages/net/jool'
time: package/feeds/packages/jool/compile#5.37#0.65#5.44
package/Makefile:111: recipe for target 'package/feeds/packages/jool/compile' failed
make[1]: *** [package/feeds/packages/jool/compile] Error 2
make[1]: Leaving directory '/home/mangix/devstuff/openwrt'
/home/mangix/devstuff/openwrt/include/toplevel.mk:216: recipe for target 'package/jool/compile' failed
```

neheb, 2019-09-25:
@ydahhrk it specifically fails at siit when the linking error happens.

neheb, 2019-09-25:
Some of these configure checks need to be fixed. They will most likely fail with -Werror=implicit-function-declaration

```
configure:4491: checking for memset
configure:4491: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-erro$
conftest.c:50:6: warning: conflicting types for built-in function 'memset'; expected 'void *(void *, int,  unsigned int)' [-Wbuiltin-declaration-mismatch]
   50 | char memset ();
      |      ^~~~~~
conftest.c:38:1: note: 'memset' is declared in header '<string.h>'
   37 | # include <limits.h>
   38 | #else
configure:4491: $? = 0
configure:4491: result: yes
configure:4491: checking for strcasecmp
configure:4491: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-erro$
conftest.c:51:6: warning: conflicting types for built-in function 'strcasecmp'; expected 'int(const char *, const char *)' [-Wbuiltin-declaration-mismatch]
   51 | char strcasecmp ();
      |      ^~~~~~~~~~
configure:4491: $? = 0
configure:4491: result: yes
configure:4491: checking for strtol
configure:4491: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-erro$
configure:4491: $? = 0
configure:4491: result: yes
configure:4491: checking for pow
configure:4491: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-erro$
conftest.c:53:6: warning: conflicting types for built-in function 'pow'; expected 'double(double,  double)' [-Wbuiltin-declaration-mismatch]
   53 | char pow ();
      |      ^~~
conftest.c:41:1: note: 'pow' is declared in header '<math.h>'
   40 | # include <limits.h>
   41 | #else
configure:4491: $? = 0
configure:4491: result: yes
configure:4500: checking for library containing pow
configure:4531: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-erro$
conftest.c:38:6: warning: conflicting types for built-in function 'pow'; expected 'double(double,  double)' [-Wbuiltin-declaration-mismatch]
   38 | char pow ();
      |      ^~~
conftest.c:1:1: note: 'pow' is declared in header '<math.h>'
    1 | /* confdefs.h */
configure:4531: $? = 0
configure:4548: result: none required
configure:4556: checking for library containing pthread_create
configure:4587: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-erro$
configure:4587: $? = 0
configure:4604: result: none required
configure:4612: checking for library containing argp_parse
configure:4643: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-erro$
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/lib/gcc/mips-openwrt-linux-musl/9.2.0/../../../../mips-openwrt-linux-musl/bin/ld:$
conftest.c:(.text.startup+0x2): undefined reference to `argp_parse'
collect2: error: ld returned 1 exit status
```

ydahhrk, 2019-09-25:
> Switched computers. Getting a different error now:

These errors imply that it's not finding argp.

> conftest.c:(.text.startup+0x2): undefined reference to `argp_parse'

This too. If argp cannot be found in OpenWRT's standard library set, I suppose you need to add libargp (or equivalent) as an explicit external dependency.

I think this needs to be set at the OpetWRT package level. It wouldn't make sense to request libargp in vanilla Jool since most distros don't seem to need it.

(Still thinking about the other configure checks.)

neheb, 2019-09-25:
argp-standalone is compiled and available. Even adding -largp to the LDFLAGS still has it unavailable.

ydahhrk, 2019-09-25:
Ahh. Looks like Jool is using both argp and getopt. And it's not `#include`ing the latter properly. This explains the "Switched computers" errors, but not the `undefined reference` one.

Mea culpa. I'll push a patch tomorrow, but you can add `#include <unistd.h>` to `src/usr/argp/main.c` if you want a temporary fix.

neheb, 2019-09-25:
The latter error I fixed with:

```
--- a/configure.ac
+++ b/configure.ac
@@ -30,7 +30,7 @@ AC_CHECK_HEADER_STDBOOL
 AC_CHECK_FUNCS([inet_ntoa memset strcasecmp strtol pow])
 AC_SEARCH_LIBS([pow], [m])
 AC_SEARCH_LIBS([pthread_create], [pthread])
-AC_SEARCH_LIBS([argp_parse], [argp])
+AC_CHECK_LIB([argp], [argp_parse])
 
 # Checks for dependencies.
 PKG_CHECK_MODULES(LIBNLGENL3, libnl-genl-3.0 >= 3.1)
```

including unistd.h is not correct for musl. getopt_long is included in <getopt.h> . See: https://git.musl-libc.org/cgit/musl/tree/include/getopt.h

Looks like those are all GNU extensions.

I seem to have gotten further to only hit another error:

```
make[5]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr/iptables'
make libxt_JOOL_SIIT.so;
make[6]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr/iptables'
gcc -Wall -pedantic -std=gnu11 -O2 -I../.. -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5=jool-4.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wall -W -Wmissing-declarations -Wmissing-prototypes  -D_INIT=libxt_JOOL_SIIT_init -fPIC -c -o libxt_JOOL_SIIT.o libxt_JOOL_SIIT.c;
gcc: error: unrecognized command line option '-mno-branch-likely'
gcc: error: unrecognized command line option '-mips32r2'; did you mean '-mpc32'?
gcc: error: unrecognized command line option '-fhonour-copts'; did you mean '-fhonor-std'?
gcc: error: unrecognized command line option '-mips16'; did you mean '-m16'?
gcc: error: unrecognized command line option '-minterlink-mips16'
gcc: error: unrecognized command line option '-fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5=jool-4.0.5'
Makefile:39: recipe for target 'libxt_JOOL_SIIT.o' failed
make[6]: *** [libxt_JOOL_SIIT.o] Error 1
make[6]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr/iptables'
Makefile:28: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/jool-4.0.5/src/usr/iptables'
Makefile:356: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
```

CC is not being passed.

ydahhrk, 2019-09-26:
Ok. I'm fine with patching the Makefile. Is CC the only variable being ignored?

ydahhrk, 2019-09-26:
(`usr/iptables/Makefile` is the one manual Makefile (ie. lacks Makefile.am). It's pretty short so it should be easy to fix.)

neheb, 2019-09-26:
Finally got it. Here's a working version:

https://github.com/neheb/packages/commit/a529c59d47f68edd284e53bfd790e6e9566649a7

I need someone to test it though.

tiagogaspar8, 2019-09-26:
Great! I'll test it as soon as I can.
One question though, shouldn't we be using the stable version, in this case 4.0.1?
From what I read versions from 4.0.2-4.0.5 are transition versions

ydahhrk, 2019-09-26:
It's best to use 4.0.5. At this point I personally think it's more mature than 4.0.1.

neheb, 2019-09-26:
Anyway, the actual patch is here: https://github.com/neheb/packages/blob/a529c59d47f68edd284e53bfd790e6e9566649a7/net/jool/patches/010-musl.patch

Would be great to get upstream.

ydahhrk, 2019-09-26:
https://github.com/NICMx/Jool/commit/8e10bc1a0a00b65933276487650f45f49693c764

Do you need the 4.0.6 release? Or are you using the latest master?

tiagogaspar8, 2019-10-02:
Well If you can release the 4.0.6 version that would be great. I tested neheb's version and it works well.
If you release version 4.0.6 it means that the musl patch can be removed right?

neheb, 2019-10-02:
There is no 4.0.6 version.

tiagogaspar8, 2019-10-02:
From what I can tell ydahhrk can release the 4.0.6 with the patch you made if you need, am I right @ydahhrk ?

neheb, 2019-10-02:
There's no particular need.

ydahhrk, 2019-10-02:
I'd like to postpone the release of 4.0.6 a little, to test a recent and somewhat delicate tweak.

(But I could just remove the tweak and release 4.0.6 as is. That's why I asked. If there's no need, I'll just keep hammering it for a while.)

But either way, 4.0.6 should no longer need the patch (once it's out).

tiagogaspar8, 2019-10-16:
Since jool is updated I'll close this issue

closed 2019-10-16 by tiagogaspar8