openwrt/packages #9352: bind-9.14.3 error in osx environment

porroso, 2019-06-29:
Maintainer: @nmeyerhans
Environment: (ar71xx, OpenWrt Trunk, Osx 10.14.5 compile environment)

I get this error when compiling in Osx, using Debian get compiled with no problems
No way to get it working in Osx?

```
/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/lib/gcc/mips-openwrt-linux-musl/8.3.0/../../../../mips-openwrt-linux-musl/bin/ranlib: /openwrt/build_dir/target-mips_24kc_musl/bind-9.14.3/ipkg-install/usr/lib/libisc.a: error reading backtrace-emptytbl.o: file truncated
make[6]: *** [install] Error 1
make[5]: *** [install] Error 1
make[4]: *** [install] Error 1
```

moralrebuild, 2019-08-10:
Same issue on the MacBook Pro OSX 10.14

ronluna, 2019-09-10:
Same problem here. 
were you able to fix it?

> Maintainer: @nmeyerhans
> Environment: (ar71xx, OpenWrt Trunk, Osx 10.14.5 compile environment)
>
> I get this error when compiling in Osx, using Debian get compiled with no problems
> No way to get it [...]

nebulabox, 2020-01-18:
try `ccache -C`

porroso, 2020-02-02:
now Environment: (ath79, OpenWrt Trunk, Osx 10.15.3 xcode 11.3.1)
tried ccache -C as suggested by nebulabox, but get same error as before

../openwrt/staging_dir/toolchain-mips_24kc_gcc-9.2.0_musl/lib/gcc/mips-openwrt-linux-musl/9.2.0/../../../../mips-openwrt-linux-musl/bin/ranlib: ..openwrt/build_dir/target-mips_24kc_musl/bind-9.14.8/ipkg-install/usr/lib/libisc.a: error reading backtrace-emptytbl.o: file truncated
make[6]: *** [install] Error 1
make[5]: *** [install] Error 1
make[4]: *** [install] Error 1
make[3]: *** [../openwrt/build_dir/target-mips_24kc_musl/bind-9.14.8/.built] Error 2
make[3]: Leaving directory `../openwrt/feeds/packages/net/bind'
time: package/feeds/packages/bind/compile#0.88#0.90#2.08
make[2]: *** [package/feeds/packages/bind/compile] Error 2

neheb, 2020-02-23:
maybe @ldir-EDB0 has an idea.

porroso, 2020-06-01:
and again Environment: (target-arm_cortex-a9, OpenWrt Trunk, Osx 10.15.5 xcode 11.5)
already tried ccache -C as suggested by nebulabox, but get same error as before
error: 
.../bind-9.16.3/ipkg-install/usr/lib/libisc.a: error reading backtrace-emptytbl.o: file truncated
Really no way to fix?

EXSERENS, 2020-09-24:
still not fixed in bind 9.16.6
tested in ramips_mt7621