openwrt/packages #9358: Wifidog v1.3.1 not working

henok400, 2019-06-30:
Wifidog v1.3.1 the last version not working openwrt v18.0.6 please developer fix this issue

Leo-PL, 2019-07-01:
Please provide more details. At first glance looks like duplicate of #9273

henok400, 2019-07-02:
it clear i used openwrt the last version v18.0.6 and wifidog 1.3.1 but it is not work error, not redirect to splash page. but openwrt 15.0.5 and wifidog 1.2.1 is perfect captive portal is working. i google to solve this only i got wifiodog-ng but trial version or underdevelopment. Do u have solution this?

nguyennhatdung, 2019-07-08:
i got the same issue, wifidog not working anymore

[6][Mon Jul  8 03:03:19 2019][29759](gateway.c:531) Forking into background
root@OpenWrt:~# [6][Mon Jul  8 03:03:19 2019][29760](gateway.c:362) Setting started_time
[3][Mon Jul  8 03:03:19 2019][29760](util.c:178) socket(): Invalid argument
[3][Mon Jul  8 03:03:19 2019][29760](gateway.c:377) Could not get IP address information of br-lan, exiting...

henok400, 2019-07-08:
I tried many experment to solve this i cant. It show error socket invalid argument and could not get ip address. I use wifidog old version openwrt v15.0.5 and wifidog v1.2.1 perfect wifi management voucher system, i agree system and user login system. It manage time and data usage limit. My business cloud based called (cloudwifizone) com . I post this issue in freelancer no one cant solve in this case new version of openwrt. All suggust coovacilli but wifidog is batter for me simple way in linux webhosting manage wifi user and deploy many location. Please wifidog developers back it orginal wifidog like version 1.2.1

nguyennhatdung, 2019-07-08:
after trying i just found the old version never work, opkg install wifidog => never work
i had build a new version base on lastest git of wifidog & openwrt 18

1. install wifidog with opkg install wifidog
2. download https://nhatdung.com/wifidog.openwrt_18.06.4.tar.gz
unpack it in to /usr/bin replace original file

3. mkdir -p /usr/local/etc/
cp /etc/wifidog.conf /usr/local/etc/

hope this help any1 who trying make wifidog work again

henok400, 2019-07-08:
i install new openwrt 18.06.4 to my GL-inet router then i follow your step1. opkg update, opkg install wifidog in putty.exe step2. I download & extract files and transfer WinSCP.exe in to /usr/bin then step3. send command into putty.exe then wifidog.conf file like this 
```
GatewayID IND5290
ExternalInterface eth0
GatewayInterface br-lan
AuthServer {
Hostname wan.cloudwifizone.com
HTTPPort 80
Path /
}
HTTPDMaxConn 253
ClientTimeout 10
FirewallRuleSet global {
FirewallRule allow to 8.8.8.8    #S
}
FirewallRuleSet validating-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet known-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet unknown-users {
FirewallRule allow udp port 53
FirewallRule allow tcp port 53
FirewallRule allow udp port 67
FirewallRule allow tcp port 67
FirewallRule block udp port 8000
}
```
then rebooted router 
problem continue it is not solved 
```
Mon Jul  8 15:44:31 2019 daemon.err wifidog[1577]: socket(): Invalid argument
Mon Jul  8 15:44:31 2019 daemon.err wifidog[1577]: Could not get IP address information of br-lan, exiting...
```
can you help me tks

henok400, 2019-07-08:
repeat installation i got error 
```
Mon Jul  8 20:51:27 2019 daemon.info procd: Instance wifidog::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Mon Jul  8 20:51:27 2019 daemon.warn odhcpd[852]: DHCPV6 REQUEST IA_NA from 00010001233f0e3300219b6f2304 on br-lan: ok fda2:5109:8c47:4::e28/128 fd93:765c:40a1::e28/128
```

neheb, 2019-07-08:
I'm quite confused. There is no Wifidog at version 1.3.1. 1.3.0 was the last version. Wifidog-ng was never updated to version 1.3.1

cotequeiroz, 2019-07-09:
This is caused by a socket call, using SOC_RAW with a non-existing protocol.  This has been fixed upstream, so I'll open a PR to fix this.

cotequeiroz, 2019-07-09:
@nguyennhatdung, can you test the package with #9432 applied to make sure it is working correctly?

nguyennhatdung, 2019-07-09:
> can you help me tks

@henok400 : you need remove the old wifidog file first
`ls /usr/bin
mv /usr/bin/wdctl /usr/bin/wdctl.bak
mv /usr/bin/wifidog /usr/bin/wifidog.bak
`
and then tranfer my wifidog.openwrt_18.06.4.tar.gz in to your router, extract it 
`tar -C /usr/bin -zxvf wifidog.openwrt_18.06.4.tar.gz`

just run wifidog -f -d 7 and see what happen

@cotequeiroz : i just recompile (without modify anything) and it work, seem newer openwrt not compatible with old package

henok400, 2019-07-09:
I tried so many times reset my router follow all steps the same error instance wifidog crash loop 6.

neheb, 2019-07-09:
Looks like it can safely be backported.

cotequeiroz, 2019-07-10:
@nguyennhatdung, @henok400 
The patched package has made it to snapshots for most arches.  Give it a try with 18.06 (it should not be a problem).  I'm just assuming you're running a mipsel_24kc (snapshots have been built for it) from the binary built above.  If not, let me know. Note the `1.3.0-4` release number.  The following shell script (copy & paste to a running shell works) should do it:
```sh
#!/bin/sh
export TMP_ARCH="`opkg print-architecture | tail -n1 | cut -f2 -d' '`"
cd /tmp || exit 1
rm -f "wifidog_1.3.0-4_${TMP_ARCH}.ipk"
wget "https://downloads.openwrt.org/snapshots/packages/${TMP_ARCH}/packages/wifidog_1.3.0-4_${TMP_ARCH}.ipk" && \
opkg install --force-reinstall "wifidog_1.3.0-4_${TMP_ARCH}.ipk"
unset TMP_ARCH
```
I'll merge #9439 and close this as soon as you confirm it working.

henok400, 2019-07-10:
@cotequeiroz
you give another job "How to install openwrt shell script"
i cant to install i'm amateurs developer. can you give me help me to install my router is GL-iNet 6416A v1 and Linksys WRT1900ACS v2. tks

cotequeiroz, 2019-07-10:
Login to openwrt using putty/ssh.  When at the prompt, paste the lines I posted above.  

It should install it.  Then check your logs for the error again; it should not happen any more.

henok400, 2019-07-11:
@cotequeiroz it is great job. 
openwrt v18.6.4 and wifidog v1.3.0-4 is perfect working. Thank you again for back wifidog.

closed 2019-07-12 by cotequeiroz

henok400, 2019-08-07:
@cotequeiroz  Dear sir i setup wifidog and internet getaway not WAN port it's wireless Client mode. then the error coming do u have solutions? No problem in WAN port internet source wifidog. Thank you.
```
wifidog[939]: ioctl(): SIOCGIFADDR No such device
wifidog[938]: Could not get IP address information of br-lan
```

henok400, 2019-08-07:
i have got solutions
remove or put # front
`#ExternalInterface eth0`

henok400, 2021-09-09:
Dear @cotequeiroz wifidog1.3.1-6 cant work more than 1 wifi user login the auth server logout all session users. if you can fix this I know ur one of openwrt developers. im from cloudwifizone

cotequeiroz, 2021-09-09:
Hi.  I'm afraid I don't have time to properly do this at the moment.  wifidog's last commit was in April, 2019, so I expect this to be .  Can you provide some more information, so I can try to at least look at the code to spot something easy.  I won't be able to install and test it right now.

henok400, 2021-09-11:
This is system log show wifidog after give internet access the 2nd device login automatically limit internet both or more device. only work single device. i use previous version 1.3.0-4 there is no any issue. 

Fri Sep 10 15:32:24 2021 daemon.info wifidog[1864]: Captured 10.0.1.165 requesting [http%3A%2F%2Fwww.msftncsi.com%2Fncsi.txt] and re-directing them to login page
Fri Sep 10 15:32:25 2021 daemon.info wifidog[1864]: Auth server returned authentication code 1
Fri Sep 10 15:32:25 2021 daemon.info wifidog[1864]: Got ALLOWED from central server authenticating token e6sf19c5i0cwzfrbx0bi21v4jubhjo2d from 10.0.1.165 at 1c:bf:ce:22:3e:56 - adding to firewall and redirecting them to portal

Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Received connection from 10.0.1.165, spawning worker thread
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Got client MAC address for ip 10.0.1.165: 1c:bf:ce:22:3e:56
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Check host 149.154.167.51:80 is in whitelist or not
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: rule mask cloudwifizone.com
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: host 149.154.167.51:80 is not in cloudwifizone.com, continue
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Captured 10.0.1.165 requesting [http%3A%2F%2F149.154.167.51%3A80%2Fapi] and re-directing them to login page

henok400, 2021-09-13:
@cotequeiroz I have requested openwrt package bugs report this link. Please could you help me. Thank you
https://github.com/openwrt/packages/issues/16591

PR #9439 (2019-07-11): [openwrt-18.06] wifidog: fix getting ip from interface, cleanup
Maintainer: @mhaas 
Compile tested: arm, WRT3200ACM, openwrt master
Run tested: same as above, basic testing, see #9432 

Description:
Patch taken from upstream fixes an Invalid argument error while trying
to get the IP address of an interface.
Makefile was updated to current style.
(cherry-picked from 5ab9f3e3576d7fd28cf55ac604dbf485eb1b0de3)

Fixes #9358.