openwrt/packages #9416: postfix: permissions prevent proper execution

flyn-org, 2019-07-07:
Maintainer: @Shulyaka
Environment: x86_64 master 48ada8aa

Description:

Postfix fails to start due to improper permissions.

```
# /etc/init.d/postfix restart
postfix: Postfix is running with backwards-compatible default settings
postfix: See http://www.postfix.org/COMPATIBILITY_README.html for details
postfix: To disable backwards compatibility use "postconf compatibility_level=2" and "postfix reload"
postfix/postfix-script: stopping the Postfix mail system
postfix: Postfix is running with backwards-compatible default settings
postfix: See http://www.postfix.org/COMPATIBILITY_README.html for details
postfix: To disable backwards compatibility use "postconf compatibility_level=2" and "postfix reload"
postfix/postfix-script: warning: not owned by group postdrop: /usr/sbin/postqueue
postfix/postfix-script: warning: not owned by group postdrop: /usr/sbin/postdrop
postfix/postfix-script: warning: not set-gid or not owner+group+world executable: /usr/sbin/postqueue
postfix/postfix-script: warning: not set-gid or not owner+group+world executable: /usr/sbin/postdrop
postfix/postfix-script: starting the Postfix mail system
```

This is fixed by running:

```
# chgrp postdrop /usr/sbin/postqueue
# chgrp postdrop /usr/sbin/postdrop
# chmod g+s /usr/sbin/postqueue
# chmod g+s /usr/sbin/postdrop
# killall -9 postdrop
# /etc/init.d/postfix restart
```

Can the package set these values on install?

Shulyaka, 2019-07-11:
Hi!

Thanks for reporting the issue.
The `postfix set-permissions` command will fix this for you. However it should have been run automatically on package installation.

Did you install postfix as a separate package or did you build your own OpenWRT image with postfix built-in?

Shulyaka, 2019-07-11:
Looks like a reincarnation of #491

flyn-org, 2019-09-09:
I installed the postfix package from the standard repository after building a custom base image. Running `postfix set-permissions` does indeed allow Postfix to start.

flyn-org, 2019-10-05:
@Shulyaka, do you plan to add `postfix-set-permissions` to the post-install script?

Shulyaka, 2019-10-06:
Yes, I just need to figure out how to do it properly for all use cases: a fresh install, an install with copied configuration from a previous system, an install from package, or from an rootfs image.

flyn-org, 2019-11-29:
@Shulyaka: ping.

ojkastl, 2020-03-11:
Hi,

I just installed postfix on my OpenWRT machine, and had similar warnings like in the initial description.

I removed everything and started from scratch, and carefully reimplemented my changes. Turns out I used the wrong directories in the first attempt, and thus earned lots of warnings.

@flyn-org Can you check the directory settings in main.cf?
This is the main.cf directly after installation:
```
# grep -v "^#" main.cf.PAKET |sort -n |uniq

         PATH=/bin:/usr/bin:/usr/local/bin:/usr/X11R6/bin
         ddd $daemon_directory/$process_name $process_id & sleep 5
 
  
command_directory = /usr/sbin
compatibility_level = 2
config_directory = /etc/postfix
daemon_directory = /usr/libexec/postfix
data_directory = /usr/var/lib/postfix
debug_peer_level = 2
debugger_command =
default_database_type = cdb
inet_protocols = ipv4
mail_spool_directory = /usr/var/mail
manpage_directory = no
mydomain = <...>
myhostname = <...>
mynetworks_style = subnet
queue_directory = /usr/var/spool/postfix
shlib_directory = /usr/lib/postfix
unknown_local_recipient_reject_code = 550
```

Shulyaka, 2020-03-12:
The thing is, we already have `postfix set-permissions` in the postinst script, and also in init script if the default configuration was detected. I will try again to reproduce the issue on my environment

flyn-org, 2020-04-26:
I do use an external disk, and thus some of my directories are configured differently:

# grep dir /etc/postfix/main.cf
queue_directory = /mnt/xvdb/var/spool/postfix
data_directory = /mnt/xvdb/var/lib/postfix
mail_spool_directory = /mnt/xvdb/var/spool/mail
html_directory = no
manpage_directory = no
readme_directory = no

I place my Postfix configurations in the "files" directory of my buildroot before building my OpenWrt image.