openwrt/packages #9417: nfs-kernel-server: recursive dependency detected! on 1907

manjuprajna, 2019-07-07:
Maintainer: @Peter Wagner <tripolar@gmx.at>
Environment: Ubuntu1804

Description:
when run make menuconfig, I got below error message:

tmp/.config-package.in:88369:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:88369:   symbol PACKAGE_nfs-kernel-server depends on NFS_KERNEL_SERVER_V4
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
feeds/packages/net/nfs-kernel-server/Config.in:4:       symbol NFS_KERNEL_SERVER_V4 depends on PACKAGE_nfs-kernel-server

but on master branch, no such error message.

hopefully the maintainer @Peter Wagner can port this to 1907, thank you!

manjuprajna, 2019-10-17:
nobody even takes a look at it? 3 monthes past, still nobody fix it.
@tripolar help!

manjuprajna, 2019-12-23:
@mans0n thank you problem fixed.

closed 2019-12-23 by manjuprajna

lantis1008, 2019-12-23:
Can you link the commit which resolves this please?

manjuprajna, 2019-12-23:
@lantis1008 commit 7203a58d7b5102c0c401ac81a5fa85d8dca64dc6
kernel: remove LINUX_4_9 dependency of kmod-dax