openwrt/packages #9427: collectd-exec-mod -> does not execute shell scripts since ~1-2 weeks on startup

camelcamro, 2019-07-09:
collectd-exec-mod
tried current trunk from today ....

collectd - exec-plugin does not execute shell scripts since ~1-2 weeks
was there a change ?

neheb, 2019-07-09:
ping @feckert

camelcamro, 2019-07-09:
i see that the luci_statistic is generating the config for collectd ...
and there is the problem
is total different as the old for few weeks

=> stat-genconfig
i have no clue if that was tested, but with an older version it is working fine. no clue .. what was done :(
also the firewall collectd are not correct, i would recommend to do a rollback

feckert, 2019-07-09:
Indeed there are problem with plugins which need a special treatment!

- network
- iptables
- exec
- curl
- maybe collectd as well
- maybe logfile as well

ping @mberndt123

> i would recommend to do a rollback

@camelcamro If this not get fixed in the next days I will revert the changes

camelcamro, 2019-07-09:
is not urgent i guess, i will use the old from v.18.06 or 04 .. already tested with the old one, and works fine
so, just relax .. tranquilo
 :)

feckert, 2019-07-09:
@camelcamro Please try this
[stat-genconfig.txt](https://github.com/openwrt/packages/files/3372457/stat-genconfig.txt)

The changes should solve your issue :-)

feckert, 2019-07-09:
This should fix the issue
https://github.com/openwrt/luci/pull/2844

feckert, 2019-07-19:
@camelcamro did my change fix your issue? If so please close.

camelcamro, 2019-07-19:
tested it now ....and is OK :)
thx a lot. pls close ticket (not sure if i have rights to close ...)

closed 2019-07-19 by camelcamro