openwrt/packages #9430: chaosvpn: <ERR> tincd manager slave has died with returncode 0, status 0

hydrandt, 2019-07-09:
Maintainer: @neheb @cotequeiroz @TuxCoder
Environment: mips_24kc / TP-Link tl-wdr4300 / 18.06.4

chaosvpn v2.18 (from openwrt 18.06.4 repo)
tincd 1.0.35 (from openwrt 18.06.4 repo)

When started, with valid configuration and keys in place, chaosvpn/tincd crashes:

root@router:/etc/tinc# chaosvpn -r -c /etc/tinc/chaosvpn.conf
<INFO>  ChaosVPN client v2.18 starting.
<ERR>   tincd manager slave has died with returncode 0, status 0.

neheb, 2019-07-18:
In 19.07, there is version 2.19. That may fix your issue.

hydrandt, 2019-07-19:
@neheb cheers. For now we have downgraded to 17.01.7 - chaosvpn/tinc work in this release.

hydrandt, 2019-11-28:
@neheb The problem exists also on openwrt 19.07. It seems to be problem with chaosvpn fetching the configs, not with tinc - if you copy over configs from existing installation, it starts. 

When chaosvpn is called with -d parameter (detach), we get more info, the error is:

`crypto_rsa_decrypt: rsa decrypt failed`

This seems to be resolved by **not using underscores in node name** in my case, so more generally by **only using letters in node name**.

Without the -d parameter, chaosvpn still doesn't start -> the init script must be modified.

neheb, 2019-11-28:
Unfortunately, I don’t use nor maintain this. I looked at git history. I don’t see any changed in chaosvpn that would cause a change in behavior.

neheb, 2020-02-07:
ping @zioproto any idea?

zioproto, 2020-02-15:
I am sorry no idea, I never used chaosvpn