openwrt/packages #9431: pdns-recursor: not compiling under arc

James-TR, 2019-07-09:
Maintainer: me
Environment: buildbot arc700

Description:

I went checking to see if there were compilation errors on the recently merged pdns-recursor package and found some errors for arc700 (and possibly other platforms):

```
In file included from pdns_recursor.cc:53:
syncres.hh: In static member function 'static void SyncRes::pruneNSSpeeds(time_t)':
syncres.hh:451:3: error: unrecognizable insn:
   }
   ^
(insn 18 17 19 5 (set (reg:SI 166)
        (plus:SI (plus:SI (reg:SI 25 r25)
                (reg:SI 168))
            (const_int 20 [0x14]))) "syncres.hh":443 -1
     (expr_list:REG_EQUAL (const:SI (plus:SI (symbol_ref:SI ("_ZN7SyncRes10t_sstorageE") [flags 0xe0] <var_decl 0x7fe52faa8ab0 t_sstorage>)
                (const_int 20 [0x14])))
        (nil)))
during RTL pass: vregs
syncres.hh:451:3: internal compiler error: in extract_insn, at recog.c:2304
```

I've also opened a tracking issue in PowerDNS/pdns#8048

neheb, 2019-07-09:
Not sure I would bother looking into this. GCC8 is the first version of GCC that supports ARC. There are bound to be bugs. An upgrade to GCC9 might fix it. Have not tested.

James-TR, 2019-07-17:
Thanks @neheb I'm closing this for now since it doesn't look like much can really be done.

closed 2019-07-17 by James-TR