openwrt/packages #9448: <python - 2.7.16-7>:ctypes.util.find_library('c') does not work

brucealthompson, 2019-07-10:
Maintainer: @\<jeffreyto> 
Environment: (x86-64, generic, build trunk)

Description:
The python function ctypes.util.find_library() depends on platform specific code that seems to be not working in OpenWrt. The function ctypes.util.find_library() is supposed to return a string which is the filename of the system library referenced in the passed string argument. I have tried ctypes.util.find_library('c') and ctypes.util.find_library('bz2') and neither of them work. The issue can be reproduced using the python code block below
``
root@OpenWrt:/lib# python
Python 2.7.16 (default, Jul 08 2019, 05:38:51) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
>>> import ctypes.util
>>> name = ctypes.util.find_library('c')
>>> print(name)
None
>>> import os
>>> print os.name
posix
>>> import sys
>>> print(sys.platform)
linux2
>>> name = ctypes.util.find_library('bz2')
>>> print(name)
None
>>> 

```
I found libc.so in /lib on my OpenWrt installation and libbz2.so.1.0 is in /usr/lib/.

jefferyto, 2019-07-11:
From the (Python 2) [documentation](https://docs.python.org/2/library/ctypes.html#finding-shared-libraries):

> On Linux, `find_library()` tries to run external programs (`/sbin/ldconfig`, `gcc`, and `objdump`) to find the library file. It returns the filename of the library file.

Considering there is no `ldconfig` for musl, I'm not surprised it doesn't work.

brucealthompson, 2019-07-12:
Jeffrey,

I checked a little further into this. I ran the code snippet above using pdb and the issue is not with ldconfig. Here is the first line of find_library() in ctypes.util:
   >> return _findSoname_ldconfig(name) or _get_soname(_findLib_gcc(name))

The code that causes the failure is trying to use gcc with the trace option to get the name of the requested library. Here's the code that is failing:
`   def _findLib_gcc(name):
        # Run GCC's linker with the -t (aka --trace) option and examine the
        # library name it prints out. The GCC command will fail because we
        # haven't supplied a proper program with main(), but that does not
        # matter.
        expr = r'[^\(\)\s]*lib%s\.[^\(\)\s]*' % re.escape(name)
        cmd = 'if type gcc >/dev/null 2>&1; then CC=gcc; elif type cc >/dev/null 2>&1; then CC=cc;else exit; fi;' \
              'LANG=C LC_ALL=C $CC -Wl,-t -o "$2" 2>&1 -l"$1"'

        temp = tempfile.NamedTemporaryFile()
        try:
            proc = subprocess.Popen((cmd, '_findLib_gcc', name, temp.name),
                                    shell=True,
                                    stdout=subprocess.PIPE)
            [trace, _] = proc.communicate()
        finally:
            try:
                temp.close()
            except OSError, e:
                # ENOENT is raised if the file was already removed, which is
                # the normal behaviour of GCC if linking fails
                if e.errno != errno.ENOENT:
                    raise
        res = re.search(expr, trace)
        if not res:
            return None
        return res.group(0)'

The code is failing because gcc is not creating the trace output that the code expects.

Here is the line where it is failing:
 `res = re.search(expr, trace)`

Here are the values of expr and trace when this line is executed:
`(Pdb) print expr
[^\(\)\s]*libc\.[^\(\)\s]*
(Pdb) print trace
/usr/bin/ld: mode elf_x86_64
/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crt1.o
/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crti.o
/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crtbegin.o
-lc (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libc.so)
libgcc_s.so.1 (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libgcc_s.so.1)
-lc (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libc.so)
libgcc_s.so.1 (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libgcc_s.so.1)
/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crtend.o
/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crtn.o/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crt1.o: In function `_start_c':
(.text._start_c+0x1f): undefined reference to `main'
/usr/bin/ld: link errors found, deleting executable `/tmp/tmpQE0KpF'
collect2: error: ld returned 1 exit status
`

jefferyto, 2019-07-12:
Have you tried running this code to confirm your hypothesis?

brucealthompson, 2019-07-12:
I ran the code in the python debugger (pdb) on the OpenWrt machine. That's how I know what the problem is.

The code that causes the failure is trying to use gcc with the trace option to get the name of the requested library. The problem is that the output of the c compiler does not include the name of the requested library for some reason. 

Here is the output of the c compiler:
/usr/bin/ld: mode elf_x86_64 /usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crt1.o /usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crti.o /usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crtbegin.o -lc (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libc.so) libgcc_s.so.1 (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libgcc_s.so.1) -lc (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libc.so) libgcc_s.so.1 (/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libgcc_s.so.1) /usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crtend.o /usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crtn.o/usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/crt1.o: In function _start_c':
(.text._start_c+0x1f): undefined reference to main' /usr/bin/ld: link errors found, deleting executable /tmp/tmpQE0KpF'
collect2: error: ld returned 1 exit status

I do not see any reference to libc in this output.

brucealthompson, 2019-07-12:
Another comment:
I think the output of the c compiler with the trace option above is not correct.

It looks like the path names it is displaying in the output are for the OpenWrt build environment and not for the target machine. Any idea why this might be?

PR #9532 (2019-07-22): python,python3: Fix ctypes.util.find_library()
Maintainer: me, @commodo 
Compile tested: x86-generic, 2019-07-21 snapshot sdk
Run tested: x86-generic, 2019-07-21 snapshot

Description:
Python's `ctypes.util.find_library()` function currently [doesn't work for musl libraries/systems][1].

This adds a patch to fix this function, based on a [patch from Alpine Linux][2].

Fixes #9448.

[1]: https://bugs.python.org/issue21622
[2]: https://git.alpinelinux.org/aports/tree/main/python2/musl-find_library.patch
commit e14c4e5c (2019-07-22): python,python3: Fix ctypes.util.find_library()

Python's ctypes.util.find_library() function currently doesn't work for
musl libraries/systems[1].

This adds a patch to fix this function, based on a patch from Alpine
Linux[2].

Fixes #9448.

[1]: https://bugs.python.org/issue21622
[2]: https://git.alpinelinux.org/aports/tree/main/python2/musl-find_library.patch
PR #9533 (2019-07-22): [openwrt-19.07] python,python3: Fix ctypes.util.find_library()
Maintainer: me, @commodo 
Compile tested: none (see #9532)
Run tested: none (see #9532)

Description:
Python's `ctypes.util.find_library()` function currently [doesn't work for musl libraries/systems][1].

This adds a patch to fix this function, based on a [patch from Alpine Linux][2].

Fixes #9448.

[1]: https://bugs.python.org/issue21622
[2]: https://git.alpinelinux.org/aports/tree/main/python2/musl-find_library.patch
commit 2d6700ed (2019-07-22): python,python3: Fix ctypes.util.find_library()

Python's ctypes.util.find_library() function currently doesn't work for
musl libraries/systems[1].

This adds a patch to fix this function, based on a patch from Alpine
Linux[2].

Fixes #9448.

[1]: https://bugs.python.org/issue21622
[2]: https://git.alpinelinux.org/aports/tree/main/python2/musl-find_library.patch
closed 2019-07-22 by BKPepe