openwrt/packages #9456: acme: change from curl->wget dependency must be explicit to PACKAGE_wget

pleasantone, 2019-07-11:
# Issue template (remove lines from top till here)

Maintainer: @neheb
Environment: archer c7v2 ath79, master branch, after jun 5th
Description:

Unfortunately, the dependency change from curl->wget breaks some configurations.

The package uclient-fetch also provides wget for some architectures, by default, but the wget image that is included as part of uclient-fetch doesn't support the --headers command line.

We need to explicitly depend upon PACKAGE_wget somehow, if you want this.
```
Thu Jul 11 12:25:28 2019 daemon.info acme: Running ACME for xxx
Thu Jul 11 12:25:28 2019 daemon.err run-acme[31125]: acme: Running ACME for xxx
Thu Jul 11 12:25:28 2019 daemon.info acme: Found previous cert config. Issuing renew.
Thu Jul 11 12:25:28 2019 daemon.err run-acme[31125]: acme: Found previous cert config. Issuing renew.
Thu Jul 11 12:25:29 2019 daemon.info run-acme[31125]: Renew: 'xxx'
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: wget: unrecognized option: header
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: Usage: wget [options] <URL>
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: Options:
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	-4				Use IPv4 only
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	-6				Use IPv6 only
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	-q				Turn off status messages
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	-O <file>			Redirect output to file (use "-" for stdout)
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	-P <dir>			Set directory for output files
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--user=<user>			HTTP authentication username
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--password=<password>		HTTP authentication password
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--user-agent|-U <str>		Set HTTP user agent
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--post-data=STRING		use the POST method; send STRING as the data
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--spider|-s			Spider mode - only check file existence
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--timeout=N|-T N		Set connect/request timeout to N seconds
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--proxy=on|off|-Y on|off	Enable/disable env var configured proxy
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]:
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: HTTPS options:
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--ca-certificate=<cert>		Load CA certificates from file <cert>
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: 	--no-check-certificate		don't validate the server's certificate
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]:
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: Please refer to https://www.gnu.org/software/wget/manual/html_node/Exit-Status.html for error code: 1
Thu Jul 11 12:25:29 2019 daemon.err run-acme[31125]: Can not init api.
```

neheb, 2019-07-11:
ping @yousong 

uclient-fetch has PROVIDES:=wget but the full GNU wget is needed here. What's the solution?

yousong, 2019-07-12:
It's still name collision.  I prefer we change the abstract package name to something else like `wgeter`.    That way `opkg install wget` will still match existing expectation and intuition.  Coordinated multiple treewide changes are required to achieve this.

pleasantone, 2019-07-12:
Opkg install wget works fine, but uclient-fetch provides wget to satisfy dependencies. If wget packages provided wget-full you could depend upon wget-full, or you can switch back to curl.

yousong, 2019-07-12:
I thought names in `PROVIDES` should be for minimum common.  But I think you are right that we should let opkg know that `wget` and `wget-nossl` both provide `wget-full`, or `gnu-wget`.  That fits in just well ;)

neheb, 2019-07-12:
We could also change the package name to wget-ssl and add PROVIDES:=wget as well as change the dependency in acme.sh.

yousong, 2019-07-12:
Changing package name has the undesirable effect that users may get lost and will need to search things like the internet, changelog to find out the new name use in `opkg install xx`.

neheb, 2019-07-18:
Another option would be to implement the missing options. I have no idea what is needed.

pleasantone, 2019-07-19:
I don’t think we should be doing local changes to a package when there are two other packages that solve the problem with no changes.

PR #9941 (2019-09-09): acme: depends on gnu-wget
Maintainer: wget @tripolar , acme @tohojo 
Compile tested: n/a
Run tested: n/a

Description:

"net/wget/Makefile" was reworked to provide "gnu-wget" so that packages like acme requiring features from it can depend on it explicitly, not the more basic "wget" which is also provided by
"uclient-fetch"

Fixes #9456

Only checked the dependency resolution.
closed 2019-09-09 by tripolar