openwrt/packages #9469: ffmpeg:

lantis1008, 2019-07-14:
Maintainer: @thess @antonlacon 
Environment: x86-geode, OpenWrt 19.07

Description:
Compiles fine on other x86 profiles, but not under geode. 

My current workaround. Not sure if this is the best solution.
```diff
--- a/makefile
+++ b/makefile
@@ -485,11 +485,15 @@
 endif
 
 ifneq ($(CONFIG_TARGET_x86),)
+ifeq ($(CONFIG_TARGET_x86_geode),)
 ifeq ($(CONFIG_NASM),y)
   FFMPEG_CONFIGURE += --enable-x86asm
 else
   FFMPEG_CONFIGURE += --disable-x86asm
 endif
+else
+   FFMPEG_CONFIGURE += --disable-x86asm
+endif
 endif

 ifeq ($(BUILD_VARIANT),full)
```

neheb, 2019-07-14:
It doesn't compile as nasm is totally broken under that platform and generates invalid files. Why, I don't know.

lantis1008, 2019-07-14:
So the fix is further up the line i guess. 
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=toolchain/Config.in;hb=f1875e902d0afb7d9b9e5285b4fd8da7f6c5e30a#l227
?

neheb, 2019-07-14:
Yeah probably. Nasm probably should not be compiled for legacy x86 as it’s broken there: http://downloads.openwrt.org/snapshots/faillogs/i386_pentium/packages/

diizzyy, 2019-07-14:
You probably want something like this...

```
ifeq ($(CONFIG_i386),y)
  ifneq ($(findstring pentium4,$(CONFIG_TARGET_ARCH_PACKAGES)),)
    ifeq ($(CONFIG_NASM),y)
      CONFIGURE_ARGS += --enable-gpl
    else
      CONFIGURE_ARGS += --disable-x86asm
    endif
  endif
endif
```

diizzyy, 2019-07-14:
Something like this should work (not tested)
```
ifeq ($(CONFIG_i386),y)
  ifneq ($(findstring pentium4,$(CONFIG_TARGET_ARCH_PACKAGES)),)
    ifeq ($(CONFIG_NASM),y)
      CONFIGURE_ARGS += --enable-gpl
    else
      CONFIGURE_ARGS += --disable-x86asm
    endif
  else
    CONFIGURE_ARGS += --disable-x86asm
  endif
endif
```

diizzyy, 2019-07-16:
@lantis1008 
I know it's not a direct fix but if you want and have time, try https://github.com/openwrt/packages/pull/9475 ?

diizzyy, 2019-07-22:
@lantis1008 
Hi, I see.
Regarding size it depends, about 2-2.5Mbyte if I recall excluding external libs and if you disable optimization.

neheb, 2020-01-12:
I found a better solution:

```
diff --git a/include/target.mk b/include/target.mk
index a813ba2d2d..a30400c5bc 100644
--- a/include/target.mk
+++ b/include/target.mk
@@ -177,8 +177,8 @@ ifeq ($(DUMP),1)
     CPU_CFLAGS_octeonplus = -march=octeon+ -mabi=64
   endif
   ifeq ($(ARCH),i386)
-    CPU_TYPE ?= pentium
-    CPU_CFLAGS_pentium = -march=pentium-mmx
+    CPU_TYPE ?= pentium-mmx
+    CPU_CFLAGS_pentium-mmx = -march=pentium-mmx
     CPU_CFLAGS_pentium4 = -march=pentium4
   endif
   ifneq ($(findstring arm,$(ARCH)),)
```

lantis1008, 2020-01-12:
@neheb are you suggesting that instead of your patch sent to mailing list? Or should NASM still be disabled exclusively.

neheb, 2020-01-12:
instead of. I got rid of the patch on the mailing list.

That being said, runtime testing of ffmpeg with x86-asm enabled would be great.

lantis1008, 2020-01-12:
I don't have any users that i know of with them anymore... we produce the images as kind of a legacy thing. I'm sure someone will come out of the woodwork and complain if it doesn't work...

neheb, 2020-01-12:
Now that I think about it, that NASM patch will be needed for 19.07 as I'm pretty sure we can't just change the toolchain.

neheb, 2020-02-17:
I fixed this in master and 19.07.

closed 2020-02-17 by neheb