openwrt/packages #9487: transmission: uses unreasonable amount of CPU time on torrents with web seeds

patrakov, 2019-07-15:
# Issue template (remove lines from top till here)

Maintainer: @neheb 
Environment: TP-Link Archer C7 v2, ar71xx/generic, OpenWrt SNAPSHOT r10061-0f6b944
Version: transmission-daemon 2.94 (d8e60ee44f)

Description:

I have limited the download and upload speeds of my instance of Transmission to 200 kilobytes per second. `transmission-daemon` behaves sanely (eats less than 2% CPU) when downloading torrents that do not have any web seeds. However, with web seeds, it always eats 100% CPU.

Example torrent that triggers the issue: https://torrents.endlessm.com/torrents/eos-amd64-amd64-base-3.6.1-iso.torrent

If the torrent link becomes dead, just get any torrent from https://endlessos.com/download/

neheb, 2019-07-18:
Cannot reproduce locally.

My theory is that the router is being memory limited. The 100 CPU usage you're seeing is probably the router swapping memory in and out. Try adding a swap device.

In the config, there's a mem_percentage parameter. Try setting it to more than 50 (the default).

patrakov, 2019-07-18:
Increased it to 75, stopped all services unneeded at the moment, still see the same bug. Will try later (not today) to reproduce on x86 in kvm, just to see if it is device-specific.

patrakov, 2019-07-18:
```
Mem: 99040K used, 25184K free, 3868K shrd, 4676K buff, 28448K cached
CPU:  75% usr  21% sys   0% nic   0% idle   0% io   0% irq   2% sirq
Load average: 1.27 1.07 0.59 2/71 5712
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 3713     1 nobody   S    18320  15%  72% transmission-daemon -g /mnt/bigdisk/.settings -f
```

neheb, 2019-07-18:
My attempt was on an mt7621 system with 512MB of RAM + swapfile.

patrakov, 2019-07-18:
Was it on a release or snapshot version? In my case, it is snapshot.

neheb, 2019-07-18:
snapshot

neheb, 2019-07-18:
Could also try running with strace. It might uncover something.

patrakov, 2019-07-18:
strace did not uncover anything obvious (like syscalls redone over and over again, or errors, or EAGAIN).

neheb, 2019-07-18:
No real idea.

Maybe it's an ath79 issue instead of a transmission issue. I do know the USB ports on there use a lot of CPU.

patrakov, 2019-07-18:
Let me remind you: there is no issue at all if I download only torrents that have no web seeds. So this rules out any USB issue.

Epicteto, 2019-08-02:
Similar issue.

Environment: Raspberry pi 4b+ 2GB RAM, dietpi@latest
Version: transmission-daemon 2.94 (d8e60ee44f)

Description:
Transmission is using one core, and CPU% bottlenecks at consistent 100% for a regular download speed of 20MB/s. If download speed is slower, CPU% is lower. Tried changing UDP/encryption/peer/cache with no success.
Samba transfers reach 90MB/s on the same device, 
Internet connection speed test is about the same.

Raspberry pi 3b+, Transmission 2.92 (14714), dietpi: reaches 20-30MB/s (the USB max) with CPU% between 20% and 70%.

neheb, 2019-08-02:
A different user reported that this happens on OpenWrt but not a diferent distro on his Raspberry Pi.

The only possibility is that there are patches we should not be carrying. One of them is for DHT. So that's not it. One is for Safari so that's not it either.

It might be the TLS patch. Try adding

```
procd_set_param env TR_CURL_SSL_NO_VERIFY=1
```

to transmission.init after the command line:

https://github.com/openwrt/packages/blob/master/net/transmission/files/transmission.init#L118

patrakov, 2019-08-02:
Does not help:

```
Mem: 103772K used, 20452K free, 20K shrd, 2100K buff, 9196K cached
CPU:  83% usr   8% sys   0% nic   0% idle   0% io   0% irq   8% sirq
Load average: 6.59 2.91 2.07 2/83 17124
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
16892     1 nobody   S    20148  16%  83% transmission-daemon -g /mnt/bigdisk/.settings -f
```

diizzyy, 2019-08-02:
Have you tried the OpenSSL variant of transmission? I've read somewhere that someone noticed much lower performance with mbed TLS when trying to add support for a http service. I think it boiled down to how it handles network traffic in some way but I might be completely wrong

patrakov, 2019-08-02:
The ticket is not about low performance - Transmission can download at least 5 megabytes per second. It is about some kind of busy waiting, 100% CPU no matter how low I set the speed limit as long as there is at least one web seed.

With transmission-daemon-openssl, the situation is the same:

```
Mem: 92992K used, 31232K free, 4680K shrd, 3652K buff, 20852K cached
CPU:  88% usr   2% sys   0% nic   6% idle   0% io   0% irq   2% sirq
Load average: 1.49 1.83 0.97 4/86 5781
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 5769     1 nobody   S    17312  14%  90% transmission-daemon -g /mnt/bigdisk/.settings -f
```

zorun, 2019-08-02:
@patrakov this is a long-standing upstream issue, unfortunately.

There were already some reports on the old Trac bugtracker, but here are some recent upstream tickets:

https://github.com/transmission/transmission/issues/586
https://github.com/transmission/transmission/issues/553
https://github.com/transmission/transmission/issues/784

neheb, 2019-11-04:
Hmmm https://github.com/transmission/transmission/issues/553 has a patch that disables webseeds. Good idea or not?

patrakov, 2019-11-23:
I do not know Transmission code base well enough to judge the patch. If it does cleanly disable the broken feature, let's apply it.

Anyway the bug is not OpenWRT specific, I have reproduced it on Arch Linux. So, reported upstream: https://github.com/transmission/transmission/issues/1061

neheb, 2019-11-25:
I pushed the fix.

closed 2019-11-25 by neheb

patrakov, 2019-11-29:
I confirm the fix.