openwrt/packages #9492: procps-ng is removed from download repos?

rezad1393, 2019-07-16:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

I found out that recently this package in not in stable repos anymore.

neheb, 2019-07-16:
On a quick glance I see them for mips24. Any platform that they are missing on?

rezad1393, 2019-07-16:
sorry that was embarrassing.
I checked this:
https://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a9_vfpv3/packages/

neheb, 2019-07-16:
Fail logs are here: https://downloads.openwrt.org/releases/faillogs/arm_cortex-a9_vfpv3/

I can't see any dependencies that are not being built. Maybe python.

ping @commodo @jeffferyto

rezad1393, 2019-07-16:
I dont understand?
I just dont see the package.
if it is not there is means the build server failed to build it?
so the guys you pinged are the maintainer of repo?

jefferyto, 2019-07-16:
The actual faillog is here: https://downloads.openwrt.org/releases/faillogs-18.06/arm_cortex-a9_vfpv3/packages/procps-ng/

I checked the other architectures and arm_cortex-a9_vfpv3 is the only one that failed. It also had a [pretty weird error](https://downloads.openwrt.org/releases/faillogs-18.06/arm_cortex-a9_vfpv3/packages/procps-ng/compile.txt):

```
configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."
```

I would suggest waiting to see if it continues to fail in the next rebuild, but I'm not sure if there is automatic rebuild for release packages :joy:

neheb, 2019-08-04:
This should be available now.

closed 2019-08-04 by neheb