openwrt/packages #9493: pdns-recursos fails on mips64

neheb, 2019-07-16:
https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/pdns-recursor/compile.txt

@James-TR 

looks like some boost detection bug.

James-TR, 2019-07-17:
Thanks I'll take a look into this one and get input from the developers :)

James-TR, 2019-07-17:
pdns-recursor 4.2.0 has been released. This may still have the same issue. We can see if it's still there or try to fix prior to bumping.

James-TR, 2019-07-24:
PowerDNS/pdns#7951 introduced changes to boost.m4 which fixed the detection of boost, however now it simply fails during configure instead of falsely showing boost_context as available. Does octeonplus support System V ucontexts?

neheb, 2019-07-24:
It should.

neheb, 2019-08-13:
I looked at this again. I can't figure it out. I'm guessing it's a boost problem.

Note that the compiler for this platform has a + in its name. This has caused issues with other packages as well. This is an example: https://github.com/openwrt/packages/blob/master/multimedia/gst1-libav/Makefile#L145

ping @ClaymorePT

ClaymorePT, 2019-08-14:
Rosen, I'm currently on vacations and I don't have my PC with me. I can
only check it next week, which I will do it anyway since the new boost
version is about to be released.

ClaymorePT, 2019-08-26:
Ok, I have been analysing this issue. For what I can tell, this is an issue regarding the mips64 arch and the lack of support of f_context for it ([f_context arch support](https://www.boost.org/doc/libs/1_71_0/libs/context/doc/html/context/architectures.html))

As an alternative, for mips64 I could try to use u_context. [It will be slower but it's an alternative](https://www.boost.org/doc/libs/1_71_0/libs/context/doc/html/context/ff/implementations__fcontext_t__ucontext_t_and_winfiber.html#implementation).

ClaymorePT, 2019-08-26:
I also thought that, but its what the documentation sugests :/

ClaymorePT, 2019-08-26:
Ok, the fix that I suggested is a no-go. 

After trying that approach, I get this 
`libboost_context.so: undefined reference to `getcontext'`
which happens because 'getcontext' was deprecated since POSIX.1-2003.

@neheb The best thing I can do is to disable boost.context for mips64 :/ 
What are your thoughts about this?

ClaymorePT, 2019-08-26:
> From reading the documentation again, it seems that nobody has written ASM for MIPS64. Probably because of a lack of devices.

Yep, I noticed that when I looked at the [src/asm](https://github.com/boostorg/context/tree/develop/src/asm) folder and noticed the same.
Nothing to do here :/

> Disabling boost.context for mips64 and ARC seems correct here.
Doing so.

ClaymorePT, 2019-08-26:
Ok. I've opened a PR @ https://github.com/openwrt/packages/pull/9838 which will disable Boost.Context for arc and mips64.

closed 2019-12-31 by neheb

neheb, 2021-02-13:
@ClaymorePT I see https://github.com/boostorg/context/pull/92 has a bunch of mips64 code for context. Any idea about that?

ClaymorePT, 2021-02-13:
@neheb Are you talking about [this](https://github.com/boostorg/context/commit/e3f744a1862164062d579d1972272d67bdaa9c39)?

ClaymorePT, 2021-02-13:
@neheb Ah! I see that you I've found the PR. I'm not monitoring such changes. They seem to be recent ones.