openwrt/packages #9494: prometheus fails on mips64

neheb, 2019-07-16:
https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/prometheus/compile.txt

ping @aparcar and @jefferyto 

Any idea what's happening? Closest thing I see to an error are these lines:

```
# gopkg.in/fsnotify/fsnotify.v1
pkg/mod/gopkg.in/fsnotify/fsnotify.v1@v1.3.1/inotify.go:39:15: undefined: unix.InotifyInit
```

jefferyto, 2019-07-18:
@aparcar are you looking into this?

aparcar, 2019-07-18:
I'm not really a go person but a prometheus user person :( I'll try to look into it but likely can't find a solution...

aparcar, 2019-07-18:
@neheb which target/subtarget is this actually?

neheb, 2019-07-18:
octeonplus

neheb, 2019-07-20:
This is also an issue on 19.07: https://downloads.openwrt.org/releases/faillogs-19.07/mips64_octeonplus/packages/prometheus/compile.txt

PR #9543 (2019-07-25): prometheus: Fix build for mips64/mips64el
Maintainer: @aparcar 
Compile tested: octeon (2019-07-18 snapshot sdk), ath79-generic / x86-generic / armvirt-64 (2019-07-22 snapshot sdk)
Run tested: **none**

Description:
Prometheus uses an older version of fsnotify, which uses a system call (`unix.InotifyInit`) that is not implemented for mips64/mips64el.

This patches Prometheus to use a newer version of fsnotify that uses a different system call (`unix.InotifyInit1`) that should be available on all Linux systems.

Fixes #9494.
commit 79b19421 (2019-07-25): prometheus: Fix build for mips64/mips64el

Prometheus uses an older version of fsnotify, which uses a system call
(unix.InotifyInit) that is not implemented for mips64/mips64el.

This patches Prometheus to use a newer version of fsnotify that uses a
different system call (unix.InotifyInit1) that should be available on
all Linux systems.

Fixes #9494.
PR #9544 (2019-07-25): [openwrt-19.07] prometheus: Fix build for mips64/mips64el
Maintainer: @aparcar 
Compile tested: none (see #9543)
Run tested: none (see #9543)

Description:
Prometheus uses an older version of fsnotify, which uses a system call (`unix.InotifyInit`) that is not implemented for mips64/mips64el.

This patches Prometheus to use a newer version of fsnotify that uses a different system call (`unix.InotifyInit1`) that should be available on all Linux systems.

Fixes #9494.
commit 43a9da01 (2019-07-25): prometheus: Fix build for mips64/mips64el

Prometheus uses an older version of fsnotify, which uses a system call
(unix.InotifyInit) that is not implemented for mips64/mips64el.

This patches Prometheus to use a newer version of fsnotify that uses a
different system call (unix.InotifyInit1) that should be available on
all Linux systems.

Fixes #9494.
closed 2019-07-25 by neheb