openwrt/packages #9495: net/kea: Unusable -- Requires "all" of Python 3

jeffsf, 2019-07-16:
Trying to build an image on the ipq40xx platform including `kea` results in a 55 MB image, rather than 14 MB. `kea` does not have a run-time dependency on Python 3, which gets dragged in my the current package definition.

This is potentially due to the boost library being built with Python support.

Python 3 is not a run-time dependency of kea.

https://ftp.isc.org/isc/kea/1.3.0/doc/kea-guide.html#required-software

neheb, 2019-07-23:
dependency is here: https://github.com/openwrt/packages/blob/master/net/kea/Makefile#L79

neheb, 2019-07-23:
ping @rosysong @hbl0307106015

rosysong, 2019-07-24:
@jeffsf  all right, will drop it, Thanks!

closed 2019-07-25 by neheb

tiagogaspar8, 2019-08-18:
After removing the python3 dependency Kea disappears from the make menuconfig

commit 44442042 (2019-07-25): kea: remove python3 dependency of kea-admin

fix #9495
commit 07538f53 (2020-05-28): kea: remove python3 dependency of kea-admin

fix #9495

(cherry picked from commit 444420423fbbd707c41343e043d00fa2e843e483)