openwrt/packages #9499: CircleCI is failing due to gpg: no valid OpenPGP data found.

BKPepe, 2019-07-16:
This issue started to happen today. CircleCI is failing on Download the SDK.

Output:
```
#!/bin/bash -eo pipefail
curl "https://$SDK_HOST/$SDK_PATH/sha256sums" -sS -o sha256sums
curl "https://$SDK_HOST/$SDK_PATH/sha256sums.asc" -sS -o sha256sums.asc
gpg --with-fingerprint --verify sha256sums.asc sha256sums
rsync -av "$SDK_HOST::downloads/$SDK_PATH/$SDK_FILE" .
sha256sum -c --ignore-missing sha256sums
gpg: no valid OpenPGP data found.
gpg: the signature could not be verified.
Please remember that the signature file (.sig or .asc)
should be the first file given on the command line.
Exited with code 2
```

ping @thess

diizzyy, 2019-07-16:
Since the generated binaries are discarded we skip the PGP step since it seems to break from time to time.

diizzyy, 2019-07-16:
Fixed

closed 2019-07-16 by diizzyy