openwrt/packages #9503: minicom (2.7.1-1) not work -S script function

juankr10s, 2019-07-17:
Please make sure that the issue subject starts with `<minicom (2.7.1-1)>: `
in OpenWrt 18.06.4, 

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2019-07-18:
There's no enough info in this.

diizzyy, 2019-08-01:
Closing due to lack of information by submitter and timeout

closed 2019-08-01 by diizzyy