openwrt/packages #9506: <watchcat>: many instances in uci - add after each reboot or upgrade

kofec, 2019-07-17:
Maintainer: @Roger D <rogerdammit@gmail.com>
Environment: Openwrt Snapshoot

Description:

```
root@Xiaomi3G:~# uci show | grep watchca
system.@watchcat[0]=watchcat
system.@watchcat[0].mode='ping'
system.@watchcat[0].pinghosts='8.8.8.8'
system.@watchcat[0].forcedelay='30'
system.@watchcat[0].period='300'
system.@watchcat[0].pingperiod='20'
ucitrack.@system[1].init='watchcat'
ucitrack.@system[2].init='watchcat'
ucitrack.@system[3].init='watchcat'
ucitrack.@system[4].init='watchcat'
ucitrack.@system[5].init='watchcat'
ucitrack.@system[6].init='watchcat'
ucitrack.@system[7].init='watchcat'
ucitrack.@system[8].init='watchcat'
ucitrack.@system[9].init='watchcat'
ucitrack.@system[10].init='watchcat'
ucitrack.@system[11].init='watchcat'
ucitrack.@system[12].init='watchcat'
ucitrack.@system[13].init='watchcat'
ucitrack.@system[14].init='watchcat'
ucitrack.@system[15].init='watchcat'
ucitrack.@system[16].init='watchcat'
ucitrack.@system[17].init='watchcat'
ucitrack.@system[18].init='watchcat'
ucitrack.@system[19].init='watchcat'
ucitrack.@system[20].init='watchcat'
ucitrack.@system[21].init='watchcat'
ucitrack.@system[22].init='watchcat'
ucitrack.@system[23].init='watchcat'
ucitrack.@system[24].init='watchcat'
ucitrack.@system[25].init='watchcat'
ucitrack.@system[26].init='watchcat'
ucitrack.@system[27].init='watchcat'
ucitrack.@system[28].init='watchcat'
ucitrack.@system[29].init='watchcat'
ucitrack.@system[30].init='watchcat'
ucitrack.@system[31].init='watchcat'
ucitrack.@system[32].init='watchcat'
ucitrack.@system[33].init='watchcat'
ucitrack.@system[34].init='watchcat'
ucitrack.@system[35].init='watchcat'
ucitrack.@system[36].init='watchcat'
ucitrack.@system[37].init='watchcat'
ucitrack.@system[38].init='watchcat'
 ```
```

feckert, 2019-07-21:
I think the problem has todo with this lines
https://github.com/openwrt/luci/blob/ca94c7805c2edd4ac29f27b16d8bee6ae09eb1d0/applications/luci-app-watchcat/root/etc/uci-defaults/40_luci-watchcat#L4-L6

Others scripts do a delete  before add
See
https://github.com/openwrt/luci/blob/ca94c7805c2edd4ac29f27b16d8bee6ae09eb1d0/applications/luci-app-adblock/root/etc/uci-defaults/40_luci-adblock#L4

kofec, 2019-07-22:
I will test it and provide feedback here

kofec, 2019-07-22:
I'm not sure how it looks like regarding adblock but for section system there are also other instances:
root@Xiaomi3G:~# uci show | grep track | grep system
```
ucitrack.@system[0]=system
ucitrack.@system[0].init='led'
ucitrack.@system[0].affects='luci_statistics' 'dhcp'
ucitrack.@system[1]=system
ucitrack.@system[1].init='watchcat'
ucitrack.@system[2]=system
ucitrack.@system[2].init='watchcat'
ucitrack.@system[3]=system
ucitrack.@system[3].init='watchcat'

```
It is possible that it can be removed "led.. or luci..." ?

kofec, 2019-07-22:
As i understand I should add:
`delete ucitrack.@system[-1]`

feckert, 2019-07-22:
I am not sure if the whole section is correct.
I think this has to be

```
#!/bin/sh

uci -q batch <<-EOF >/dev/null
 	delete ucitrack.@watchcat[-1]
 	add ucitrack watchcat
 	set ucitrack.@watchcat[-1].init=watchcat
 	commit ucitrack
 EOF
 
 rm -f /tmp/luci-indexcache
 exit 0
```

kofec, 2019-07-30:
It seems to be OK. After reboot, the process was initiated and parameters are fine

```
Tue Jul 30 23:13:14 2019 user.notice ucitrack: Setting up /etc/config/watchcat reload trigger for non-procd /etc/init.d/watchcat
Tue Jul 30 23:13:16 2019 daemon.notice procd: /etc/rc.d/S96led: setting up led power
Tue Jul 30 23:13:16 2019 user.info watchcat: starting all tasks
Tue Jul 30 23:13:16 2019 user.info watchcat: started task (mode=ping;period=300;pinghosts=192.168.1.1;pingperiod=20;forcedelay=30)
Tue Jul 30 23:13:16 2019 user.info watchcat: all tasks started
......
root@XiaomiMiniAP:~# uci show | grep watch
system.@watchcat[0]=watchcat
system.@watchcat[0].mode='ping'
system.@watchcat[0].forcedelay='30'
system.@watchcat[0].pinghosts='192.168.1.1'
system.@watchcat[0].pingperiod='20'
system.@watchcat[0].period='300'
ucitrack.@watchcat[0]=watchcat
ucitrack.@watchcat[0].init='watchcat'
....
1956 root      1204 S    /bin/sh /usr/bin/watchcat.sh period 300 30 192.168.1.1 20
```

closed 2019-08-01 by feckert

kofec, 2019-08-01:
Thanks a lot for fix