openwrt/packages #9509: php7: failing to build on aarch64_cortex-a53

bluewavenet, 2019-07-17:
Maintainer: Michael Heimpold <mhei@heimpold.de>
Environment: aarch64_cortex-a53 OpenWrt Snapshot

Description: php7 and all its modules are failing to build for the aarch64_cortex-a53 architecture
from ~16th July 2019.

neheb, 2019-07-17:
It's not just php. The issue happens as the buildbot is not building the kernel correctly.

bluewavenet, 2019-07-17:
Indeed not just PHP.
I will assume it is on someone's todo list and be patient :-)

bluewavenet, 2019-07-17:
@bawantha

neheb, 2019-07-20:
I just realized the buildbot is trying to build LEDE packages for this platform.

https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53_neon-vfpv4/packages/libwebsockets/cyassl/compile.txt

note the version. This is 2.2.1, not 3.1.0 in master.

bluewavenet, 2019-07-22:
Is the common factor for all these failures the fact that openssl is failing to build?
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/base/openssl/compile.txt

cotequeiroz, 2019-07-22:
You have to go back one more step.  `cryptodev-linux` (along with other out-of-tree kernel modules) fails, which is the cause of the `openssl` failure:
```
/builder/shared-workdir/build/sdk/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_musl/bin/aarch64-openwrt-linux-musl-ld: cannot open linker script file ./arch/arm64/kernel/module.lds: No such file or directory
scripts/Makefile.modpost:129: recipe for target '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-cryptodev-linux-1.10/cryptodev.ko' failed
```
mac80211, and nat46, for example, fail with the same error message.

bluewavenet, 2019-07-22:
How can it be fixed?

bawantha, 2019-07-22:
exactly what I want to fix

neheb, 2019-08-13:
Closing. It seems buildbot is fixed now.

closed 2019-08-13 by neheb

bluewavenet, 2019-08-13:
@neheb Indeed yes. Thankyou.