openwrt/packages #9524: opkg: Cannot install package busybox due to a conflict with diff tools

fantom-x, 2019-07-19:
Maintainer: @yousong
Environment: Image builder with 19.07-SNAPSHOT

Description: opkg_install_cmd: Cannot install package busybox. 

```
Collected errors:
 * check_data_file_clashes: Package busybox wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/usr/bin/cmp
	But that file is already provided by package  * diffutils
 * opkg_install_cmd: Cannot install package busybox.
Makefile:152: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:111: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:195: recipe for target 'image' failed
make: *** [image] Error 2
```

yousong, 2019-07-20:
Changes are now backported to both 19.07 and 18.06 .  Thanks for the report.

closed 2019-07-20 by yousong

fantom-x, 2019-07-20:
Thx, will it take time before the commit shows up here: https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=refs/heads/openwrt-19.07 ?

yousong, 2019-07-20:
The change is on diffutils.  It depends on buildbot to pick up the change and build new .ipk for use with imagebuilder.

yousong, 2019-07-20:
If I understand it right, this should be it, http://buildbot.openwrt.org/openwrt-19.07/packages/builders/mips_24kc

fantom-x, 2019-07-20:
Thx, I see the commit in github now. Thx again for prompt fixes.