openwrt/packages #9525: Adblock: Starts too early and before fstab

fantom-x, 2019-07-20:
Maintainer: @dibdot
Environment: 19.07-SNAPSHOT

Adblock (START=30) starts before fstab (START=40) and if I want to store adblock files on a USB drive, then there is an issue as the drive is not yet mounted when adblock starts.

Can adblock start be moved to 60 or even 80 so that all systems are up and operational before it starts?

dibdot, 2019-07-20:
During boot adblock only implements a startup trigger (usually for your wan interface), nothing more. Whenever your interface comes up, adblock processing begins (with a default delay of 2 seconds). If your usb device needs more time, then raise the 'adb_triggerdelay' e.g. to '30' ('Trigger Delay' in LuCI under Additonal fields of the extra section).

closed 2019-07-20 by dibdot

fantom-x, 2019-07-20:
Well, this is current behaviour and it can change in the future: i.e. you or someone else might add more logic to the start up phase, which would require access to the file system and this might become a bigger issue. Why does adblock need to start this early?

fantom-x, 2019-07-20:
The “start” action calls “f_main” and it is in turn using “adb_tmpdir”, which can point at a allocation on a usb drive. Also, some of us might not want to have “/tmp” reside in memory, so even if “adb_tmpdir” is not set and “/tmp” is not in memory, then that is another race condition.

dibdot, 2019-07-20:
> Why does adblock need to start this early?
as explained before, the adblock service will be triggered on boot solely by an "interface trigger" which reacts on ifup events of the network interface. To get all possible trigger events you have to register this procd service as soon as possible.

> The “start” action [...]
There is no "start" action on boot. On boot only the procd service will be registered. If you change multiple OpenWrt defaults (e.g. filesystem changes) than you have to change the adb_triggerdelay as well.

fantom-x, 2019-07-20:
> To get all possible trigger events you have to register this procd service as soon as possible.

Thank you, that makes sense. Maybe a comment added to the init script would help to avoid these questions later.