openwrt/packages #9537: libressl: build fails with musl

anibalportero, 2019-07-22:
Seems like libressl fails on musl targets after 2.9.2:

https://github.com/gentoo/gentoo/pull/12078

```
libtool: compile:  gcc -DPACKAGE_NAME=\"libressl\" -DPACKAGE_TARNAME=\"libressl\" -DPACKAGE_VERSION=\"2.9.2\" "-DPACKAGE_STRING=\"libressl 2.9.2\"" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"libressl\" -DVERSION=\"2.9.2\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_SYMLINK=1 -DHAVE_ERR_H=1 -DHAVE_ASPRINTF=1 -DHAVE_MEMMEM=1 -DHAVE_STRLCAT=1 -DHAVE_STRLCPY=1 -DHAVE_STRNDUP=1 -DHAVE_STRNLEN=1 -DHAVE_STRSEP=1 -DHAVE_TIMEGM=1 -DHAVE_SYSLOG=1 -DHAVE_ACCEPT4=1 -DHAVE_PIPE2=1 -DHAVE_POLL=1 -DHAVE_SOCKETPAIR=1 -DHAVE_EXPLICIT_BZERO=1 -DHAVE_GETAUXVAL=1 -DHAVE_GETAUXVAL=1 -DHAVE_DL_ITERATE_PHDR=1 -DHAVE_CLOCK_GETTIME=1 -DHAVE_VA_COPY=1 -DHAS_GNU_WARNING_LONG=1 -DSIZEOF_TIME_T=8 -I. -I../include -I../include/compat -DLIBRESSL_INTERNAL -D__BEGIN_HIDDEN_DECLS= -D__END_HIDDEN_DECLS= -I../crypto/asn1 -I../crypto/bn -I../crypto/ec -I../crypto/ecdsa -I../crypto/evp -I../crypto/modes -I../crypto -I/builds/pantacor/pv-platforms/openwrt-base/openwrt/staging_dir/host/include -D_DEFAULT_SOURCE -D_BSD_SOURCE -D_POSIX_SOURCE -D_GNU_SOURCE -D__STRICT_ALIGNMENT -O2 -I/builds/pantacor/pv-platforms/openwrt-base/openwrt/staging_dir/host/include -fpic -Wall -std=gnu99 -fno-strict-aliasing -fno-strict-overflow -D_FORTIFY_SOURCE=2 -fstack-protector-strong -DHAVE_GNU_STACK -Wno-pointer-sign -MT compat/getprogname_linux.lo -MD -MP -MF compat/.deps/getprogname_linux.Tpo -c compat/getprogname_linux.c -o compat/getprogname_linux.o
compat/getprogname_linux.c: In function 'getprogname':
compat/getprogname_linux.c:32:2: error: #error "Cannot emulate getprogname"
 #error "Cannot emulate getprogname"
  ^~~~~
```

anibalportero, 2019-07-22:
https://github.com/openwrt/openwrt/pull/2261

neheb, 2019-07-23:
@anibalportero wrong repository. Issue should be reported at https://bugs.openwrt.org

closed 2019-07-23 by neheb