openwrt/packages #9538: wireguard can

wackejohn, 2019-07-22:
Maintainer: @feckert

Environment: 
```
OpenWrt SNAPSHOT, r10551-d616b2c906
MWAN3 	2.8.0-1
Running on turris omnia with four pppoe wan interface,two wireguard vpn client and one wireguard vpn server.
```

Description:
```
When I using the OpenWRT as a wireguard server with multi pppoe wan interface, the wireguard 
client failed handshake. And I found that the inbound interface and outbound interface was different.
eg: My wireguard client connect the wireguard server from pppoe-wan4, but the handshake data was sent from pppoe-wan1 to the client.
```

The tcpdump output:
```
root@HOME-Router:~# tcpdump -i pppoe-wan4 -vv host  49.94.153.160
tcpdump: listening on pppoe-wan4, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
21:40:42.419873 IP (tos 0x0, ttl 113, id 5756, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:40:47.635469 IP (tos 0x0, ttl 113, id 5757, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:40:52.534148 IP (tos 0x0, ttl 113, id 5758, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:40:57.697818 IP (tos 0x0, ttl 113, id 5759, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:02.728237 IP (tos 0x0, ttl 113, id 5760, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:07.937823 IP (tos 0x0, ttl 113, id 5761, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:13.142458 IP (tos 0x0, ttl 113, id 5762, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:18.267116 IP (tos 0x0, ttl 113, id 5763, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:23.496362 IP (tos 0x0, ttl 113, id 5764, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:28.698172 IP (tos 0x0, ttl 113, id 5765, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:33.800883 IP (tos 0x0, ttl 113, id 5766, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:38.899820 IP (tos 0x0, ttl 113, id 5767, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:43.998023 IP (tos 0x0, ttl 113, id 5768, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:49.254995 IP (tos 0x0, ttl 113, id 5769, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:54.349569 IP (tos 0x0, ttl 113, id 5770, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:41:59.327974 IP (tos 0x0, ttl 113, id 5771, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:42:04.368634 IP (tos 0x0, ttl 113, id 5772, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
21:42:09.491494 IP (tos 0x0, ttl 113, id 5773, offset 0, flags [none], proto UDP (17), length 176)
    49.94.153.160.12572 > 112.3.86.249.5888: [udp sum ok] UDP, length 148
^C
18 packets captured
18 packets received by filter
0 packets dropped by kernel
```
```
root@HOME-Router:~# tcpdump -i pppoe-wan1 -vv host  49.94.153.160
tcpdump: listening on pppoe-wan1, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
21:40:42.421591 IP (tos 0x88, ttl 64, id 1206, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:40:47.637268 IP (tos 0x88, ttl 64, id 1711, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:40:52.535810 IP (tos 0x88, ttl 64, id 1987, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
[... 47 lines cut ...]
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:41:54.395633 IP (tos 0x0, ttl 55, id 8927, offset 0, flags [none], proto ICMP (1), length 148)
    49.94.153.160 > 114.228.200.208: ICMP 49.94.153.160 udp port 12572 unreachable, length 128
        IP (tos 0x0, ttl 55, id 5110, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:41:59.329684 IP (tos 0x88, ttl 64, id 5361, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:41:59.368025 IP (tos 0x0, ttl 55, id 9212, offset 0, flags [none], proto ICMP (1), length 148)
    49.94.153.160 > 114.228.200.208: ICMP 49.94.153.160 udp port 12572 unreachable, length 128
        IP (tos 0x0, ttl 55, id 5361, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:42:04.370263 IP (tos 0x88, ttl 64, id 5540, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:42:04.409760 IP (tos 0x0, ttl 55, id 9586, offset 0, flags [none], proto ICMP (1), length 148)
    49.94.153.160 > 114.228.200.208: ICMP 49.94.153.160 udp port 12572 unreachable, length 128
        IP (tos 0x0, ttl 55, id 5540, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:42:09.493346 IP (tos 0x88, ttl 64, id 5932, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
21:42:09.532424 IP (tos 0x0, ttl 55, id 9895, offset 0, flags [none], proto ICMP (1), length 148)
    49.94.153.160 > 114.228.200.208: ICMP 49.94.153.160 udp port 12572 unreachable, length 128
        IP (tos 0x0, ttl 55, id 5932, offset 0, flags [none], proto UDP (17), length 120)
    114.228.200.208.5888 > 49.94.153.160.12572: [udp sum ok] UDP, length 92
```
The output showed that my client(49.94.153.160) connect server (112.3.86.249:5888), but the handshake data was sent from

lucize, 2019-07-22:
did you tried to add a rule so a specific port/udp to only use one interface ? you can find it in web interface

wackejohn, 2019-07-23:
> did you tried to add a rule so a specific port/udp to only use one interface ? you can find it in web interface
The wireguard itself can't bind to specific interface, and I've tried to specific 5888 port only use one interface, but it didn't work.

feckert, 2019-08-13:
This is because the wireguard runs directly on the router. For outgoing router traffic the nat prerouting hook is not passed through by the firewall, because it does not exist! Only for forwarded traffic. So a rule for on mwan3 for router traffic make no sense.
See:
https://upload.wikimedia.org/wikipedia/commons/3/37/Netfilter-packet-flow.svg

wackejohn, 2019-08-13:
@feckert 
> This is because the wireguard runs directly on the router. For outgoing router traffic the nat prerouting hook is not passed through by the firewall, because it does not exist! Only for forwarded [...]

So there is no way to get wireguard working with mwan3?

yousong, 2019-08-14:
My guess is that wireguard kernel module selects a source address to use after looking up routes.  It does not record which dst addr it previously received peer traffic and send response from that address.  If that's what happened, try to connect to the server by address `112.3.86.249`

wackejohn, 2019-08-14:
@yousong 
> My guess is that wireguard kernel module selects a source address to use after looking up routes. It does not record which dst addr it previously received peer traffic and send response from that address. If that's what happened, try to connect to the server by address `114.228.200.208 `

The `114.228.200.208` was my wan1 ip address, I've tested that the wireguard only working with wan1.
PS: Are you chinese?

yousong, 2019-08-14:
> @yousong
>
>
> The `114.228.200.208` was my wan1 ip address, I've tested that the wireguard only working
with wan1.

Then very likely my guess is right ;)  Note that the use of udp as transport plays a role here.  Many user space services use IP_PKTINFO for fetching the local address it received the msg and send response from the same address.

The issue will need to be handled by the wireguard kernel module.

> PS: Are you chinese?

Yes, I am ;)

wackejohn, 2019-08-14:
@yousong 
So you meen that I should report this to the wireguard team?
PS: I'm chinese too, so may I get your qq number or email address? Thanks.

yousong, 2019-08-14:
After searching it a bit more, I found that WireGuard designers already took the issue into consideration [1].  There should be other causes.

 [1] see the slide about "Sticky Sockets", https://www.wireguard.com/talks/eindhoven2018-slides.pdf

yousong, 2019-08-14:
Maybe try adding a source based policy route rule

```
ip rule add from 112.3.86.249 lookup N
```

where N should be the route table number corresponding to `pppoe-wan1` as setup by mwan3.

wackejohn, 2019-08-15:
> Maybe try adding a source based policy route rule
> 
> ```
> ip rule add from 112.3.86.249 lookup N
> ```
> 
> where N should be the route table number corresponding to `pppoe-wan1` as setup by mwan3.

Tried this, but it wasn't working. I think wireguard just use the main routing table.

yousong, 2019-08-15:
It's generic linux route lookup.  It should work.

Just to be sure, please share the output of the following commands

```
ip rule
ip route show table N
```

wackejohn, 2019-08-15:
@yousong 
I want to use wireguard as server on pppoe-wan4, so the output:
```
root@HOME-Router:~# ip rule
0:      from all lookup local
1001:   from all iif pppoe-wan1 lookup 1
1002:   from all iif pppoe-wan2 lookup 2
1003:   from all iif pppoe-wan3 lookup 3
1004:   from all iif pppoe-wan4 lookup 4
1005:   from all iif wana lookup 5
1006:   from all iif wanb lookup 6
2001:   from all fwmark 0x100/0xff00 lookup 1
2002:   from all fwmark 0x200/0xff00 lookup 2
2003:   from all fwmark 0x300/0xff00 lookup 3
2004:   from all fwmark 0x400/0xff00 lookup 4
2005:   from all fwmark 0x500/0xff00 lookup 5
2006:   from all fwmark 0x600/0xff00 lookup 6
2253:   from all fwmark 0xfd00/0xff00 blackhole
2254:   from all fwmark 0xfe00/0xff00 unreachable
32766:  from all lookup main
32767:  from all lookup default
```
```
root@HOME-Router:~# ip route show table 4
default via 112.1.96.1 dev pppoe-wan4 metric 21 
112.1.96.1 dev pppoe-wan4 proto kernel scope link src 112.1.99.41 
114.226.46.1 dev pppoe-wan1 proto kernel scope link src 114.226.46.109 
114.228.157.1 dev pppoe-wan2 proto kernel scope link src 114.228.157.158 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.120.0/24 dev wana proto static scope link metric 30 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn1 proto kernel scope link src 192.168.195.1 
192.168.195.4 dev vpn1 proto static scope link 
192.168.196.0/24 dev tun0 proto kernel scope link src 192.168.196.1 
223.66.36.1 dev pppoe-wan3 proto kernel scope link src 223.66.36.116 
```

yousong, 2019-08-15:
Hmm, I cannot see where the problem is.  The `from x.x.x.x lookup N` rule should work.

I am curious and had an experiment [1].  Hope it helps.

 [1] https://gist.github.com/yousong/e6ebd3c9f838286d6fda3228655c1f90

wackejohn, 2019-08-15:
> Hmm, I cannot see where the problem is. The `from x.x.x.x lookup N` rule should work.
> 
> I am curious and had an experiment [1]. Hope it helps.
> 
> [1] https://gist.github.com/yousong/e6ebd3c9f838286d6fda3228655c1f90

I think it won't work, in my opinion the source address is fixed by the wireguard when wireguard reply the handshake data, the `from x.x.x.x lookup N` rule won't change the source address. And that's why the openvpn should listen on lan ip to work with mwan3.

wackejohn, 2019-08-27:
My wireguard config:
```
root@HOME-Router:/etc/config# wg showconf vpn1
[Interface]
ListenPort = 5888
PrivateKey = XXFzF4RpuJyzv8LLl4jyF8YgkC+11odAEM18QrAOhEs=

[Peer]
PublicKey = XXz3RNu3kvuQkrjNF7MJOHdABwvfwJFNDfD3ul54FR8=
PresharedKey = XX+BdnxHq+up3NIGFCh5tf33m8w+7DScUDrHBKzLrsY=
AllowedIPs = 192.168.195.3/32, dd5d:189b:b5f9:3::3/128
PersistentKeepalive = 25

[Peer]
PublicKey = XXQ3aVzXp2PKObEInwZ1+v4w0mCHsjDQK/6onPLwxUk=
PresharedKey = XXpDE20HLOU2lRwUD0Dfu8H3U4a6XVI1e5s7mziXvyU=
AllowedIPs = 192.168.195.2/32, dd5d:189b:b5f9:3::2/128
PersistentKeepalive = 25

[Peer]
PublicKey = XXJHl6mKCkh+++pWYRVxHkSntaKnZAUuwQsgGDCEuDM=
PresharedKey = XXpDE20HLOU2lRwUD0Dfu8H3U4a6XVI1e5s7mziXvyU=
AllowedIPs = 192.168.195.4/32, dd5d:189b:b5f9:3::4/128
```

wackejohn, 2019-08-27:
> And the server side configuration?

This is the server side configuration...

fox85, 2019-09-02:
> > And the server side configuration?
> 
> This is the server side configuration...

Hello, wackejohn,  我遇到与你同样的issue发生在使用UDP协议上的OpenVPN。
我发现该问题产生的条件是在同时跑mwan3和监听着UDP端口的程序上。
我试着把OpenVPN换成TCP协议，问题就消失，数据就能从指定的wan口进出。
但一旦转换成UDP协议，数据就会从wan2入，wan1出。
我试了mwan3-2.17.3 和 2.8.0-2都依旧。

wackejohn, 2019-09-02:
> > > And the server side configuration?
> > 
> > 
> > This is the server side configuration...
> 
> Hello, wackejohn, 我遇到与你同样的issue发生在使用UDP协议上的OpenVPN。
> 我发现该问题产生的条件是在同时跑mwan3和监听着UDP端口的程序上。
> 我试着把OpenVPN换成TCP协议，问题就消失，数据就能从指定的wan口进出。
> 但一旦转换成UDP协议，数据就会从wan2入，wan1出。
> 我试了mwan3-2.17.3 和 2.8.0-2都依旧。

配置OpenVPN监听在lan口，该问题就解决了，wireguard目前只能监听在0.0.0.0，所以目前来说似乎无解。。。

mwan3官方文档：https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#openvpn

fox85, 2019-09-03:
> 配置OpenVPN监听在lan口，该问题就解决了，wireguard目前只能监听在0.0.0.0，所以目前来说似乎无解。。。
>
> mwan3官方文档：https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#openvpn

非常感谢你给出的指引。
我调整了OpenVPN的配置，bind在local lan地址上，然后在防火墙做端口转发，这样就能从WAN2入，WAN2出了。
同样地我也调整了shadowsocks server bind lan ip的配置，和在防火墙做端口转发，也能OK了。

feckert, 2019-09-03:
@fox85 @wackejohn
Come on please talk in English!!

wackejohn, 2019-09-03:
@feckert 
Sorry, @fox85 got the similar issue on OpenVPN and mwan3, and I told him that OpenVPN should bind to lan and then the issue would gone ...

fox85, 2019-09-03:
> @fox85 @wackejohn
> Come on please talk in English!!

Sorry. @feckert , thanks @wackejohn has translated my issue to english. :)

feckert, 2020-05-29:
@wackejohn @fox85 If the problem persists please reopen the ticket

psergei, 2020-11-28:
I had the same issue with my mwan3 config balancing 2 WAN with 50/50 rule.
This is how I solved it

`iptables -t mangle -I mwan3_hook 2 -p udp --dport <your_wireguard_port> -j MARK --set-mark 0x100`

Please put corresponding ifmark for your wireguard listening WAN interface. 
I added this script to /etc/mwan2.user

`if [ "$ACTION" = "ifup" ]; then
	# add Wireguard port rule if not exists already
	if ! iptables -t mangle -S mwan3_hook | grep 9999 &> /dev/null; then
		iptables -t mangle -I mwan3_hook 2 -p udp --dport 9999 -j MARK --set-mark 0x100
	fi
fi
`

As I have traced the packets, issue happens on PREROUNTING -> wman3_hook chain. Incoming handshake packet from WAN1 came to the router and wman decided to mark it with 0x100 or 0x200 - after that, the whole connection will have same mark.

wackejohn, 2020-11-29:
@psergei 
I've tried your script, but did not work. As I posted before, my issue is that the wireguard server always choose the lowest metric wan interface (in my config, wan1 had the lowest metric) to reply the handshake data, no matter where the handshake data come from (just like the openvpn listen on 0.0.0.0, and with mwan3, it should listen on lan), but the wireguard was designed to always listen on 0.0.0.0.

psergei, 2020-11-29:
I had the same issue. WG server accepted connection on WAN1 but in 50% of cases tried to respond via the WAN2 and I'm sure the mwan3 rules never comes into play.
WAN1 has the lowest metric.

Can you show the output of
iptables -t mangle -L

wackejohn, 2020-11-29:
@psergei 

The output:
```
root@HOME-Router_CZ:~# iptables -t mangle -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
mwan3_hook  all  --  anywhere             anywhere            

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
RRDIPT_INPUT  all  --  anywhere             anywhere            

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
RRDIPT_FORWARD  all  --  anywhere             anywhere            
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
RRDIPT_OUTPUT  all  --  anywhere             anywhere            
mwan3_hook  all  --  anywhere             anywhere            

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain RRDIPT_FORWARD (1 references)
target     prot opt source               destination         

Chain RRDIPT_INPUT (1 references)
target     prot opt source               destination         
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            

Chain RRDIPT_OUTPUT (1 references)
target     prot opt source               destination         
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            
RETURN     all  --  anywhere             anywhere            

Chain mwan3_connected (2 references)
target     prot opt source               destination         
MARK       all  --  anywhere             anywhere             match-set mwan3_connected dst MARK or 0x3f00

Chain mwan3_hook (2 references)
target     prot opt source               destination         
CONNMARK   all  --  anywhere             anywhere             mark match 0x0/0x3f00 CONNMARK restore mask 0x3f00
MARK       udp  --  anywhere             anywhere             udp dpt:5888 MARK set 0x400
mwan3_ifaces_in  all  --  anywhere             anywhere             mark match 0x0/0x3f00
mwan3_connected  all  --  anywhere             anywhere             mark match 0x0/0x3f00
[... 219 lines cut ...]
target     prot opt source               destination         
mwan3_rule_AntiGFW_DNS  all  --  anywhere             dns.google           mark match 0x0/0x3f00
mwan3_rule_AntiGFW_DNS  all  --  anywhere             dns.google           mark match 0x0/0x3f00
mwan3_rule_AntiGFW_DNS  all  --  anywhere             resolver1.opendns.com  mark match 0x0/0x3f00
mwan3_rule_AntiGFW_DNS  all  --  anywhere             resolver2.opendns.com  mark match 0x0/0x3f00
mwan3_rule_AntiGFW_DNS  all  --  anywhere             199.91.73.222        mark match 0x0/0x3f00
mwan3_rule_AntiGFW_DNS  all  --  anywhere             li187-110.members.linode.com  mark match 0x0/0x3f00
mwan3_rule_ChinaNet_DNS  all  --  anywhere             dns2.ctcdma.com      mark match 0x0/0x3f00
mwan3_rule_ChinaNet_DNS  all  --  anywhere             dns1.ctcdma.com      mark match 0x0/0x3f00
mwan3_rule_ChinaNet_DNS  all  --  anywhere             public1.114dns.com   mark match 0x0/0x3f00
mwan3_rule_ChinaNet_DNS  all  --  anywhere             public1.alidns.com   mark match 0x0/0x3f00
[... cut ...]
mwan3_policy_wanb_only  all  --  anywhere             tsa01s09-in-f4.1e100.net  mark match 0x0/0x3f00
mwan3_policy_wana_wanb  all  --  anywhere             anywhere             match-set gfwlist dst mark match 0x0/0x3f00
mwan3_rule_CM_IPV4  all  --  anywhere             anywhere             match-set cmcc dst mark match 0x0/0x3f00
mwan3_rule_ChinaNet_IPV4  all  --  anywhere             anywhere             match-set cnnocmcc dst mark match 0x0/0x3f00
mwan3_rule_IPV4_WEB  all  --  anywhere             anywhere             mark match 0x0/0x3f00
```

psergei, 2020-11-29:
well, first - I don't see my rule there.
second - can you show output of 

iptables -t mangle -L -v

Further looking at the rules I see an interesting thing.
When the packets comes to the router for WG, it went to PREROUTING table, then to mwan3_hook
It has no mark yet. But magic happens in mwan3_ifaces_in chain, specifically in its target chains, e.g. mwan3_iface_in_wan1 
There is a record
`MARK       all  --  anywhere             anywhere             mark match 0x0/0x3f00 /* wan1 */ MARK xset 0x100/0x3f00`
Actually this record is bound to specific interface (verbose output will show this). So if WG udp packet comes to router via the interface of WAN1, then connection will be marked with 0x100 and handshake will 100% happen on WAN1. Balance rules or metrics do not play any role here.

wackejohn, 2020-11-29:
@psergei 
The output:
```
root@HOME-Router_CZ:~# iptables -t mangle -L -v
Chain PREROUTING (policy ACCEPT 1346K packets, 1501M bytes)
 pkts bytes target     prot opt in     out     source               destination         
  31M   36G mwan3_hook  all  --  any    any     anywhere             anywhere            

Chain INPUT (policy ACCEPT 63096 packets, 17M bytes)
 pkts bytes target     prot opt in     out     source               destination         
1687K  347M RRDIPT_INPUT  all  --  any    any     anywhere             anywhere            

Chain FORWARD (policy ACCEPT 1282K packets, 1484M bytes)
 pkts bytes target     prot opt in     out     source               destination         
  30M   35G RRDIPT_FORWARD  all  --  any    any     anywhere             anywhere            
 220K   13M TCPMSS     tcp  --  any    pppoe-wan1  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
 3987  229K TCPMSS     tcp  --  pppoe-wan1 any     anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
95322 5687K TCPMSS     tcp  --  any    pppoe-wan2  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
14621  805K TCPMSS     tcp  --  pppoe-wan2 any     anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
56114 3329K TCPMSS     tcp  --  any    pppoe-wan3  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
 7086  387K TCPMSS     tcp  --  pppoe-wan3 any     anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
   14   760 TCPMSS     tcp  --  any    pppoe-wan4  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
   13   696 TCPMSS     tcp  --  pppoe-wan4 any     anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
  154  9240 TCPMSS     tcp  --  any    wana    anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU
   12   720 TCPMSS     tcp  --  wana   any     anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU
    0     0 TCPMSS     tcp  --  any    wanb    anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU
    0     0 TCPMSS     tcp  --  wanb   any     anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wireguard MTU fixing */ TCPMSS clamp to PMTU

Chain OUTPUT (policy ACCEPT 61864 packets, 26M bytes)
 pkts bytes target     prot opt in     out     source               destination         
1675K  695M RRDIPT_OUTPUT  all  --  any    any     anywhere             anywhere            
1675K  695M mwan3_hook  all  --  any    any     anywhere             anywhere            

Chain POSTROUTING (policy ACCEPT 1344K packets, 1510M bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain RRDIPT_FORWARD (1 references)
 pkts bytes target     prot opt in     out     source               destination         

Chain RRDIPT_INPUT (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 RETURN     all  --  eth0   any     anywhere             anywhere            
 247K  144M RETURN     all  --  pppoe-wan1 any     anywhere             anywhere            
96955   42M RETURN     all  --  pppoe-wan2 any     anywhere             anywhere            
57273 4226K RETURN     all  --  pppoe-wan3 any     anywhere             anywhere            
58687 4668K RETURN     all  --  pppoe-wan4 any     anywhere             anywhere            
30608 2658K RETURN     all  --  wana   any     anywhere             anywhere            
29001 2436K RETURN     all  --  wanb   any     anywhere             anywhere            

Chain RRDIPT_OUTPUT (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 RETURN     all  --  any    eth0    anywhere             anywhere            
 349K  183M RETURN     all  --  any    pppoe-wan1  anywhere             anywhere            
29875 2510K RETURN     all  --  any    pppoe-wan2  anywhere             anywhere            
29833 2506K RETURN     all  --  any    pppoe-wan3  anywhere             anywhere            
29862 2508K RETURN     all  --  any    pppoe-wan4  anywhere             anywhere            
28926 2430K RETURN     all  --  any    wana    anywhere             anywhere            
29024 2438K RETURN     all  --  any    wanb    anywhere             anywhere            

Chain mwan3_connected (2 references)
 pkts bytes target     prot opt in     out     source               destination         
  14M  813M MARK       all  --  any    any     anywhere             anywhere             match-set mwan3_connected dst MARK or 0x3f00

Chain mwan3_hook (2 references)
 pkts bytes target     prot opt in     out     source               destination         
  32M   36G CONNMARK   all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00 CONNMARK restore mask 0x3f00
[... cut ...]
config policy 'wana_6_only'
        list use_member 'wana_6_m1_w1'
        option last_resort 'unreachable'

config policy 'wanb_6_only'
        list use_member 'wanb_6_m1_w1'
        option last_resort 'unreachable'

config policy 'wana_6_wanb_6'
        list use_member 'wana_6_m1_w1'
        list use_member 'wanb_6_m2_w1'
        option last_resort 'unreachable'

config policy 'wanb_6_wana_6'
        list use_member 'wanb_6_m1_w1'
        list use_member 'wana_6_m2_w1'
        option last_resort 'unreachable'
```

psergei, 2020-11-29:
Could you try add some iptables logging to mwan3_hook chain?
Pls put LOG target there before and after the "mwan3_ifaces_in" target.

```
 iptables -t mangle -I mwan3_hook 2 -p udp --sport <your wg port> -j LOG --log-prefix "WG hook before: "
 iptables -t mangle -I mwan3_hook 4 -p udp --sport <your wg port> -j LOG --log-prefix "WG hook after: "
```

Then run 
`logread -f | grep "WG hook" `

and try connect to WG on pppoe-wan4

wackejohn, 2020-11-29:
@psergei 
output:
```
Sun Nov 29 11:34:12 2020 kern.warn kernel: [139284.766327] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=30035 PROTO=UDP SPT=5888 DPT=41502 LEN=156 MARK=0x100
Sun Nov 29 11:34:12 2020 kern.warn kernel: [139284.782218] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=30035 PROTO=UDP SPT=5888 DPT=41502 LEN=156 MARK=0x100
Sun Nov 29 11:34:17 2020 kern.warn kernel: [139290.525901] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=30153 PROTO=UDP SPT=5888 DPT=41502 LEN=156 MARK=0x100
Sun Nov 29 11:34:17 2020 kern.warn kernel: [139290.541785] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=30153 PROTO=UDP SPT=5888 DPT=41502 LEN=156 MARK=0x100
Sun Nov 29 11:34:21 2020 kern.warn kernel: [139294.190274] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30259 PROTO=UDP SPT=5888 DPT=41547 LEN=100
Sun Nov 29 11:34:21 2020 kern.warn kernel: [139294.205210] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30259 PROTO=UDP SPT=5888 DPT=41547 LEN=100
Sun Nov 29 11:34:26 2020 kern.warn kernel: [139299.229406] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30396 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:26 2020 kern.warn kernel: [139299.245280] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30396 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:31 2020 kern.warn kernel: [139304.235762] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30582 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:31 2020 kern.warn kernel: [139304.251655] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30582 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:36 2020 kern.warn kernel: [139309.248556] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30615 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:36 2020 kern.warn kernel: [139309.264429] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30615 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:41 2020 kern.warn kernel: [139314.248386] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30950 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:41 2020 kern.warn kernel: [139314.264279] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=30950 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:46 2020 kern.warn kernel: [139319.267588] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=31202 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:46 2020 kern.warn kernel: [139319.283467] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=31202 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:51 2020 kern.warn kernel: [139324.314472] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=31610 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:51 2020 kern.warn kernel: [139324.330362] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=31610 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:56 2020 kern.warn kernel: [139329.253918] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=32079 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:34:56 2020 kern.warn kernel: [139329.269796] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=32079 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:35:01 2020 kern.warn kernel: [139334.313888] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=32583 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:35:01 2020 kern.warn kernel: [139334.329783] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=32583 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:35:06 2020 kern.warn kernel: [139339.293120] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=33053 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:35:06 2020 kern.warn kernel: [139339.309011] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=33053 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
[... cut ...]
Sun Nov 29 11:35:16 2020 kern.warn kernel: [139349.328452] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=33378 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:35:21 2020 kern.warn kernel: [139354.373581] WG hook before: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=33638 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
Sun Nov 29 11:35:21 2020 kern.warn kernel: [139354.389494] WG hook after: IN= OUT=pppoe-wan1 SRC=114.228.153.243 DST=49.78.147.251 LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=33638 PROTO=UDP SPT=5888 DPT=41547 LEN=100 MARK=0x100
```

psergei, 2020-11-29:
Those are all outgoing packets, I don't see any connection attempts from outside

wackejohn, 2020-11-29:
@psergei 
If you don't mind, can you tell me your email address? I'll send my router login info, then you can ssh in my router.

psergei, 2020-11-29:
[image]

dsander, 2020-12-27:
I am having the same problem on `19.07.5` with `mwan3  2.8.14-2`. My setup is a bit different though. I want the wireguard server to respond on both WAN interfaces in case the primary (for wireguard which isn't the main primary interface)  goes down, however it always sends it's responses through the "global" primary interface.

More details about the setup are in this forum post https://forum.openwrt.org/t/wireguard-server-can-only-successfully-be-used-via-one-wan-interface/83374

I added added some debugging using these commands:
```
iptables -t mangle -I mwan3_hook 2 -p udp --sport 13377 -j LOG --log-prefix "sport WG hook before: "
iptables -t mangle -I mwan3_hook 2 -p udp --dport 13377 -j LOG --log-prefix "dport WG hook before: "
iptables -t mangle -I mwan3_hook 5 -p udp --sport 13377 -j LOG --log-prefix "sport WG hook after: "
iptables -t mangle -I mwan3_hook 5 -p udp --dport 13377 -j LOG --log-prefix "dport WG hook after: "
```

```
root@openwrt:~# logread -f | grep "WG hook"
Sun Dec 27 21:13:13 2020 kern.warn kernel: [23839.155196] dport WG hook before: IN=pppoe-vdsl OUT= MAC= SRC=80.xxx.xxx.xxx DST=83.135.89.19 LEN=176 TOS=0x00 PREC=0x00 TTL=51 ID=53830 PROTO=UDP SPT=30607 DPT=13377 LEN=156
Sun Dec 27 21:13:13 2020 kern.warn kernel: [23839.170841] dport WG hook after: IN=pppoe-vdsl OUT= MAC= SRC=80.xxx.xxx.xxx DST=83.135.89.19 LEN=176 TOS=0x00 PREC=0x00 TTL=51 ID=53830 PROTO=UDP SPT=30607 DPT=13377 LEN=156
Sun Dec 27 21:13:13 2020 kern.warn kernel: [23839.187875] sport WG hook before: IN= OUT=eth1 SRC=185.xxx.xxx.xxx DST=80.xxx.xxx.xxx LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=21203 PROTO=UDP SPT=13377 DPT=30607 LEN=100
Sun Dec 27 21:13:13 2020 kern.warn kernel: [23839.202828] sport WG hook after: IN= OUT=eth1 SRC=185.xxx.xxx.xxx DST=80.xxx.xxx.xxx LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=21203 PROTO=UDP SPT=13377 DPT=30607 LEN=100

```
Forcing the mark like @psergei suggested does not seem to help. `0x200` is the mark for the `pppoe-vdsl` inteface, but the response is send via `eth1`
```
$ iptables -t mangle -I mwan3_hook 4 -p udp --sport 13377 -j MARK --set-mark 0x200
$ iptables -t mangle -L --line-numbers -n
Chain mwan3_hook (2 references)
num  target     prot opt source               destination
1    CONNMARK   all  --  0.0.0.0/0            0.0.0.0/0            CONNMARK restore mask 0x3f00
2    LOG        udp  --  0.0.0.0/0            0.0.0.0/0            udp dpt:13377 LOG flags 0 level 4 prefix "dport WG hook before: "
3    LOG        udp  --  0.0.0.0/0            0.0.0.0/0            udp spt:13377 LOG flags 0 level 4 prefix "sport WG hook before: "
4    MARK       udp  --  0.0.0.0/0            0.0.0.0/0            udp spt:13377 MARK set 0x200
5    LOG        udp  --  0.0.0.0/0            0.0.0.0/0            udp dpt:13377 LOG flags 0 level 4 prefix "dport WG hook after: "
6    LOG        udp  --  0.0.0.0/0            0.0.0.0/0            udp spt:13377 LOG flags 0 level 4 prefix "sport WG hook after: "
7    mwan3_ifaces_in  all  --  0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
8    mwan3_connected  all  --  0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
9    mwan3_rules  all  --  0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
10   CONNMARK   all  --  0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0x3f00
11   mwan3_connected  all  --  0.0.0.0/0            0.0.0.0/0            mark match ! 0x3f00/0x3f00
```

```
Sun Dec 27 21:13:46 2020 kern.warn kernel: [23872.306149] dport WG hook before: IN=pppoe-vdsl OUT= MAC= SRC=80.xxx.xxx.xxx DST=83.135.89.19 LEN=176 TOS=0x00 PREC=0x00 TTL=51 ID=7230 PROTO=UDP SPT=31046 DPT=13377 LEN=156
Sun Dec 27 21:13:46 2020 kern.warn kernel: [23872.321711] dport WG hook after: IN=pppoe-vdsl OUT= MAC= SRC=80.xxx.xxx.xxx DST=83.135.89.19 LEN=176 TOS=0x00 PREC=0x00 TTL=51 ID=7230 PROTO=UDP SPT=31046 DPT=13377 LEN=156
Sun Dec 27 21:13:46 2020 kern.warn kernel: [23872.338708] sport WG hook before: IN= OUT=eth1 SRC=185.xxx.xxx.xxx DST=80.xxx.xxx.xxx LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=22078 PROTO=UDP SPT=13377 DPT=31046 LEN=100
Sun Dec 27 21:13:46 2020 kern.warn kernel: [23872.353665] sport WG hook after: IN= OUT=eth1 SRC=185.xxx.xxx.xxx DST=80.xxx.xxx.xxx LEN=120 TOS=0x08 PREC=0x80 TTL=64 ID=22078 PROTO=UDP SPT=13377 DPT=31046 LEN=100 MARK=0x200
```

Since the same setup and configuration worked on `18.06.9` I looked through the changes made to `mwan3` and this PR stands out because it changed the way how outgoing requests are routed: https://github.com/openwrt/packages/pull/6515

idarek, 2021-05-17:
I see that this issue seams sorted (off course as long there is no "Router allowed IP" selected at WireGuard config for peers.

I got two connections (fibre and 4G with external IP). It didn't work when I hot-plugged 4G but after reboot got both working with mwan3 in failsafe wan_4G and can connect on both addresses and getting handshake correctly.

jamesmacwhite, 2021-05-30:
This is still a problem for IPv4 and multi homed WANs. I've confirmed that I connect to the wireguard server with the LAN IP (confirming configuration works) and directly over IPv6 (setting the endpoint to an IPv6 address), but with IPv4 externally, I can't get a handshake. I suspect it is the same problem the handshake going over the wrong interface.

theAeon, 2021-06-03:
Echoing this, I'm on a trunk build from 5/31/2021 and wireguard is only responding over my WWAN despite being unbound and open on both WANs.

jamesmacwhite, 2021-06-03:
In my case, my default WAN policy is wan_wanb_wanc in failover, rather than a balanced mode. As per the info from @psergei, if I add this rule into /etc/firewall.user:

```
iptables -t mangle -I mwan3_hook 2 -p udp --dport 51820 -j MARK --set-mark 0x100
```

And add the 0x100 FW mark onto the Wireguard server interface, I can then handshake over IPv4. However it only works for the primary WAN IPv4 address. Despite having a firewall rule that should allow any incoming UDP connection to 51820 on all my other external IPv4 addresses, the handshake does not occur.

It is better than nothing but, I still think the handshake response ends up going over the primary WAN, because of the default mwan3 policy configured.

psergei, 2021-06-03:
In fact my configuration now works without this rule. I spend some time debugging it before and found that issue was with CONNMARK, had to reinstall some packages, unfortunately do not remember the details.
When the very first request is made, it comes to a specific static IP always and connection is marked with 0x100. And then WG always uses this connection. This is standard configuration, works like a charm. OpenWrt 19.07.3

jamesmacwhite, 2021-06-03:
@psergei Interesting, do share further info if you happen to recall the steps. I'm grateful for your post earlier around fwmark, this has allowed external connections with valid handshake on at least one IP for me at least.

psergei, 2021-06-03:
@jamesmacwhite As I said I had problems with CONNMARK, probably I installed corresponding packages.
If you have the problems - just try to debug the iptables and see if packets are marked properly in INPUT chain.

jamesmacwhite, 2021-06-03:
@psergei Thank you! It isn't the first time CONNMARK has been a problem with mwan3 before, I'll look into it more at some point and see what I can find.

ingenium13, 2021-09-08:
I'm having a related issue. I'm trying to have a wireguard client on openwrt 19.07.6 connect to a remote wireguard server. I have mwan3 configured with failover for two interfaces (wwan and wan). wwan is the preferred interface in mwan3, with wan being the failover, however wan has a lower metric in the kernel routing table.

The result is that wireguard will only use wan, and traffic won't go over wwan. It seems that mwan3 rules don't apply to wireguard when it initiates the connection, and it instead will only use the kernel routing table. Other than the fact that it's not using the preferred (faster) interface, this is also a problem for when wan goes down, since it won't switch over to wwan instead, and I end up locked out of the router unless I'm on site (wwan is CGNAT, so the tunnel provides a way to remotely access it when wan goes down).

I found this related post https://forum.openwrt.org/t/wireguard-server-can-only-successfully-be-used-via-one-wan-interface/83374 which seems to indicate that wireguard basically binds to the interface directly, and mwan3 can't change the routing on them.

Has there been any progress with this? Or has anyone figured out a way to get the mwan3 iptables rules to correctly apply to wireguard? Looking at /proc/net/nf_conntrack, it seems that the wireguard traffic isn't getting marked (or rather, it has mark=16128 which I think corresponds to the mask?). Traffic set to prefer wwan and failover to wan has mark=256 and traffic set to prefer wan and failover to wwan has mark=768. Those are my only two mwan3 policies.

psergei, 2021-09-08:
So far I have seen, all problems when WG is not using the correct WAN are related to conntrack, no exceptions. I have 3 WANs but only 1 with real IP is used for WG - all is working just fine for about a year.

ingenium13, 2021-09-08:
@psergei so you managed to get the wireguard connections marked correctly? My issue is a bit different from others here, since conntrack seems to be tracking it correctly and it passes data and such. Rather, the connection seems to not get marked by the mwan3 rules, and instead uses the kernel routing table. The only other connections in /proc/net/nf_conntract with mark=16128 are connections to devices in my LAN (such as a LAN device making a DNS query to the router).

psergei, 2021-09-08:
@ingenium13 Then you have to debug the rules by putting a log statement to find out why they are skipped.

ingenium13, 2021-09-09:
@psergei I figured out the issue I think. It seems that wireguard is adding a static route to the IP of the remote wireguard server to send it over wan (eth0.2 in my case). And because of that, mwan3 rules apparently don't apply. The route even persists after stopping the tunnel. If I manually remove the route, it's added back again as soon as I start the tunnel. I've tried setting route_allowed_ips to 0, and that stops adding allowed IPs to the routing table, but it still adds the route to the wireguard server for some reason.

Do you how to prevent wireguard from adding this route? It of course makes sense to add it if you're routing all traffic over the VPN, but I'm not doing that. I'm hoping that with the route removed, then the regular mwan3 rules will start working.

Or would adding a specific mangle iptables rule to force mark the connection work? If so, where should it be inserted?

psergei, 2021-09-09:
@ingenium13 It is something else. My config is different - I have 3 normal WANs and other networks connect to 1st WAN because it has real IP address. So all negotiation should happen through this WAN only.

idarek, 2021-09-30:
Hi All,
I just do some checks and found, that this is not really **mwan3** fault that **wireguard** or any other service cannot communicate back over the right interface.

**Without mwan3**: If you got two connections, 1st with Metric 1 and 2nd with Metric 10, Internet in-network is used through the lowest metric. Wireguard connected through 1st connection is working. Through 2nd connection not really as cannot handshake.

If If I change metrics around, 1st to 10 and 2nd to 1, Wireguard will handshake through 2nd with lower metric, but not through 1st with a higher metric.

Just my observation, but think it's going back into how OpenWrt is handling packages and responses through interfaces. May be wrong, but not **mwan3** issue?

Similar if you want to **iperf3** through specific interface. Its always responding back through interface with lower metric.

idarek, 2021-09-30:
**Observation.**
When I connect to 2nd connection (with set metric 100) via wireguard and when connection 1st is working (with metric 1), then I got a handshake on Wireguard. It doesn't matter if is coming back from 1st connection, wireguard is ok with that.

Not a good solution in a failsafe scenario, as you will connect to 2nd connection, the router will still try to handshake with 1st.

idarek, 2021-10-01:
Following my initial thought, in the instance of failsafe, when WAN is down run WireGuard over WAN2 by implementing actions in mwan3 to change metrics over. In theory, this shall work but will need to try.

```
if [ "${ACTION}" = "ifdown" ] && [ "${INTERFACE}" = "wan" ] ; then
 uci set network.wan.metric='100'
 uci set network.wan2.metric='1'
 uci commit
 conntrack -F
 /etc/init.d/network reload
fi

if [ "${ACTION}" = "ifup" ] && [ "${INTERFACE}" = "wan" ] ; then
 uci set network.wan.metric='1'
 uci set network.wan2.metric='100'
 uci commit
 conntrack -F
 /etc/init.d/network reload
fi
```

Rough idea. Anybody would like to participate?

wackejohn, 2021-12-01:
@feckert  @aaronjg 
Hi all,
According to mailing list of wireguard:
https://lists.zx2c4.com/pipermail/wireguard/2021-January/006243.html
https://lists.zx2c4.com/pipermail/wireguard/2021-October/007217.html

I tried to write a script to use dport routing rule:
```
root@HOME-Router_Backup[14:23]:~# cat /etc/init.d/wireguard 
#!/bin/sh /etc/rc.common
 
USE_PROCD=1
 
START=95
STOP=01
 
start_service() {
    procd_open_instance
    procd_set_param command /bin/sh "/usr/bin/wireguard_dport.sh"
    procd_set_param stdout 1
    procd_set_param stderr 1
    procd_close_instance
}
```
```
root@HOME-Router_Backup[14:23]:~# cat /usr/bin/wireguard_dport.sh
#!/bin/sh
listen_port=$(uci get network.vpn0.listen_port)
while true
do
        for i in $(seq 1 4)
        do
                if [ $(iptables -t mangle -L mwan3_iface_in_wan$i | grep -c $listen_port) = "0" ]; then
                        iptables -t mangle -I mwan3_iface_in_wan$i 3 -p udp --dport $listen_port -j LOG --log-prefix "WG_WAN$i: " 
                fi
                port=$(logread | grep IN=pppoe-wan$i | grep WG_WAN$i | tail -1 | sed 's/^.*\(SPT.*\).*$/\1/' | cut -d " " -f 1 | sed 's/SPT=//')
                case $i in
                        1)
                        rt=1
                        ;;
                        2)
                        rt=3
                        ;;
                        3)
                        rt=5
                        ;;
                        4)
                        rt=7
                        ;;
                esac
                if [ "$port" != "" ] && [ $(ip rule show | grep -c $port) = "0" ] && [ "$(wg show vpn0 | grep endpoint | cut -d ":" -f 3 | grep -c $port)" != "0" ] ; then
                        ip rule add dport $port lookup $rt
                fi
        done

        sleep 1

        ip rule show | grep dport | sed 's/^.*\(dport.*\).*$/\1/' | sed 's/^/ip rule del &/g' > /tmp/wireguard_iprule.sh
        chmod +x /tmp/wireguard_iprule.sh

        peer_num=$(wg show vpn0 | grep -c endpoint)
        if [ $peer_num != "0"  ]; then
                for i in $(seq 1 $peer_num)
                do
                        endpoint_port=$(wg show vpn0 | grep endpoint | cut -d ":" -f 3 | sed -n ${i}p)
                        sed -i "/$endpoint_port/d" /tmp/wireguard_iprule.sh
                done
                /tmp/wireguard_iprule.sh
        else
                /tmp/wireguard_iprule.sh
        fi
done
```
It works for ervery wan, but the performance was very bad (too high cpu load when running), is there bettery way to make it work? Many Thanks.

feckert, 2021-12-01:
The question is first of all, does it have to be a service? Can it not be a hotplug script under /etc/hotplug.d/iface? This is called when the wg0 interface goes up or down.

I don't like the diversions with the log. Couldn't you consult the mwan3 configuration to get the needed informations?

wackejohn, 2021-12-01:
> The question is first of all, does it have to be a service? Can it not be a hotplug script under /`etc/hotplug.d/iface`? This is called when the wg0 interface goes up or down.
> 
> I don't like the diversions with the log. Couldn't you consult the mwan3 configuration to get the needed informations?

I'm not good at shell scripts and iptables, accroding to the mailing list, there are two ways to make the wireguard working for multi wan at same time: sport policy routing or dport policy routing.

sport policy routing is very simple but it should get multi wireguard server running (or is there anyway to get it running on  the same sport?).
dport policy routing, we should know the connection  port number and the connection from which wan interface at the first time, and as a server on openwrt, when a client behind nat try to connect to the server, it shoud creat the dport policy ASAP. 

So , is there any other way to know the source port of connection and connect from which wan when the client try to connect the server, and then trigger a script to creat the dport routing policy?

wackejohn, 2022-01-01:
@feckert
help needed.
I'm trying to hack the source code of wireguard based on the patch form the mailinglist:
`https://lists.zx2c4.com/pipermail/wireguard/2021-September/007157.html`

My idea is try to save the firewall mark of incoming data (eg: the data incoming from wan2 ,and the mark based on mwan3 is MARK xset 0x300/0x3f00 ) and then apply it (0x300) when wireguard sendding the reply data, just like set fwmark manually in conf file. But when I try to save the firewall mark value, my result was got the mark mask only. (eg: the firewall mask of mwan3 was 0x3F00, and the skb->mark value was as 0x3F00.)

the code was like this:
```
wg_receive_handshake_packet @ recevice.c

static void wg_receive_handshake_packet(struct wg_device *wg,
					struct sk_buff *skb)
{
	enum cookie_mac_state mac_state;
	struct wg_peer *peer = NULL;
	/* This is global, so that our load calculation applies to the whole
	 * system. We don't care about races with it at all.
	 */
	static u64 last_under_load;
	bool packet_needs_cookie;
	bool under_load;

 	mark_for = skb->mark;
	pr_err("recive.c after mark show %x", mark_for);

	if (SKB_TYPE_LE32(skb) == cpu_to_le32(MESSAGE_HANDSHAKE_COOKIE)) {
		net_dbg_skb_ratelimited("%s: Receiving cookie response from %pISpfsc\n",
					wg->dev->name, skb);
		wg_cookie_message_consume(
			(struct message_handshake_cookie *)skb->data, wg);
		return;
	}

```
test result:

```
[  108.277464] wireguard: recive.c after mark show 3f00
```

I'm not good at source code, so my question is why the `skb->mark` only contain the firewall mark mask? Or am I mess something up?

wackejohn, 2022-01-02:
@feckert 
The fwmark mask was added by mwan3:
```root@HOME-Router_Backup[14:19]:~# iptables -t mangle -L mwan3_connected
Chain mwan3_connected (2 references)
target     prot opt source               destination         
MARK       all  --  anywhere             anywhere             match-set mwan3_connected dst MARK or 0xff00
```
Tried add this rule to the mwan3_connected chain:
```
iptables -t mangle -I mwan3_connected 1 -p udp --dport 5999 -j RETURN
```
Then tested working ( the client can handshake succefully both wan1 and wan2 ):
```
[14917.393899] wireguard: socket.c after  wg_receive 100
[14917.398312] wireguard: recive.c after mark show 100
[14917.404916] wireguard: socket.c send4 skb 100
[14917.608679] wireguard: socket.c after  wg_receive 100
[14917.613077] wireguard: recive.c after mark show 100
[14917.619756] wireguard: socket.c send4 skb 100
[14917.645015] wireguard: socket.c after  wg_receive 100
[14917.649479] wireguard: socket.c send4 skb 100
[14918.022897] wireguard: socket.c after  wg_receive 100
[14918.028152] wireguard: socket.c send4 skb 100
[14918.044822] wireguard: socket.c after  wg_receive 100
[14918.049441] wireguard: socket.c send4 skb 100
[14918.073152] wireguard: socket.c after  wg_receive 100
[14918.077901] wireguard: socket.c send4 skb 100
[14918.095085] wireguard: socket.c after  wg_receive 100
[14918.099680] wireguard: socket.c send4 skb 100
[14918.118863] wireguard: socket.c after  wg_receive 100
[14918.123425] wireguard: socket.c send4 skb 100
[14918.155415] wireguard: socket.c after  wg_receive 100
[14918.160002] wireguard: socket.c send4 skb 100
[14918.180084] wireguard: socket.c after  wg_receive 100
[14918.212089] wireguard: socket.c after  wg_receive 100
[14919.205620] wireguard: socket.c after  wg_receive 100
[14919.211116] wireguard: socket.c send4 skb 100
[14919.229388] wireguard: socket.c after  wg_receive 100
[14920.228727] wireguard: socket.c after  wg_receive 100
[14920.233982] wireguard: socket.c send4 skb 100
[14920.257990] wireguard: socket.c after  wg_receive 100
[14923.183045] wireguard: socket.c after  wg_receive 100
[14923.188292] wireguard: socket.c send4 skb 100
[14923.215949] wireguard: socket.c after  wg_receive 100
[14923.220617] wireguard: socket.c send4 skb 100
[14923.240624] wireguard: socket.c after  wg_receive 100
[14923.245183] wireguard: socket.c send4 skb 100
[14923.260741] wireguard: socket.c after  wg_receive 100
[14923.763810] wireguard: socket.c after  wg_receive 100
[14933.428209] wireguard: socket.c send4 skb 100
[14959.027002] wireguard: socket.c send4 skb 100
[14968.631990] wireguard: socket.c after  wg_receive 300
[14968.636881] wireguard: recive.c after mark show 300
[14968.643500] wireguard: socket.c send4 skb 300
[14968.842114] wireguard: socket.c after  wg_receive 300
[14968.846518] wireguard: recive.c after mark show 300
[14968.853196] wireguard: socket.c send4 skb 300
[14968.869090] wireguard: socket.c after  wg_receive 300
[14969.473264] wireguard: socket.c after  wg_receive 300
[14969.478729] wireguard: socket.c send4 skb 300
[14969.497938] wireguard: socket.c after  wg_receive 300
[14969.502494] wireguard: socket.c send4 skb 300
[14969.518026] wireguard: socket.c after  wg_receive 300
[14969.522596] wireguard: socket.c send4 skb 300
[14969.539061] wireguard: socket.c after  wg_receive 300
[14969.543639] wireguard: socket.c send4 skb 300
[14969.560049] wireguard: socket.c after  wg_receive 300
[14969.564730] wireguard: socket.c send4 skb 300
[14969.582020] wireguard: socket.c after  wg_receive 300
[14969.586632] wireguard: socket.c send4 skb 300
[14969.603054] wireguard: socket.c after  wg_receive 300
[14971.512712] wireguard: socket.c after  wg_receive 300
[14971.517972] wireguard: socket.c send4 skb 300
[14971.540136] wireguard: socket.c after  wg_receive 300
[14971.544643] wireguard: socket.c send4 skb 300
[14971.572148] wireguard: socket.c after  wg_receive 300
[14973.898984] wireguard: socket.c after  wg_receive 300
[14973.904249] wireguard: socket.c send4 skb 300
[14973.928359] wireguard: socket.c after  wg_receive 300
[14973.932968] wireguard: socket.c send4 skb 300
[14973.951092] wireguard: socket.c after  wg_receive 300
[14974.619210] wireguard: socket.c after  wg_receive 300
[14974.624481] wireguard: socket.c send4 skb 300
[14976.585498] wireguard: socket.c after  wg_receive 300
[14976.590010] wireguard: socket.c send4 skb 300
[14976.609269] wireguard: socket.c after  wg_receive 300
[14976.613782] wireguard: socket.c send4 skb 300
[14976.631186] wireguard: socket.c after  wg_receive 300
[14978.973251] wireguard: socket.c after  wg_receive 300
[14978.978582] wireguard: socket.c send4 skb 300
[14978.998833] wireguard: socket.c after  wg_receive 300
[14979.003415] wireguard: socket.c send4 skb 300
[14979.029004] wireguard: socket.c after  wg_receive 300
[14981.691732] wireguard: socket.c after  wg_receive 300
[14981.696972] wireguard: socket.c send4 skb 300
[14981.721899] wireguard: socket.c after  wg_receive 300
[14981.726418] wireguard: socket.c send4 skb 300
[14984.043485] wireguard: socket.c after  wg_receive 300
[14984.048159] wireguard: socket.c send4 skb 300
[14984.066364] wireguard: socket.c after  wg_receive 300
[14984.070990] wireguard: socket.c send4 skb 300
[14984.102831] wireguard: socket.c after  wg_receive 300
[14984.107520] wireguard: socket.c send4 skb 300
```

wackejohn, 2022-01-02:
The patch:
```
diff -uNr linux-5.15.12_orig/drivers/net/wireguard/queueing.h linux-5.15.12/drivers/net/wireguard/queueing.h
--- linux-5.15.12_orig/drivers/net/wireguard/queueing.h	2021-12-29 19:29:03.000000000 +0800
+++ linux-5.15.12/drivers/net/wireguard/queueing.h	2022-01-02 19:46:00.000000000 +0800
@@ -20,6 +20,8 @@
 struct prev_queue;
 struct sk_buff;
 
+static u32 mark;
+
 /* queueing.c APIs: */
 int wg_packet_queue_init(struct crypt_queue *queue, work_func_t function,
 			 unsigned int len);
@@ -73,15 +75,19 @@
 	return real_protocol && skb->protocol == real_protocol;
 }
 
-static inline void wg_reset_packet(struct sk_buff *skb, bool encapsulating)
+static inline void wg_reset_packet(struct sk_buff *skb, bool encapsulating,
+				   bool xnet)
 {
 	u8 l4_hash = skb->l4_hash;
 	u8 sw_hash = skb->sw_hash;
 	u32 hash = skb->hash;
-	skb_scrub_packet(skb, true);
+
+	skb_scrub_packet(skb, xnet);
+	mark = skb->mark;
 	memset(&skb->headers_start, 0,
 	       offsetof(struct sk_buff, headers_end) -
 		       offsetof(struct sk_buff, headers_start));
+	skb->mark = mark;
 	if (encapsulating) {
 		skb->l4_hash = l4_hash;
 		skb->sw_hash = sw_hash;
diff -uNr linux-5.15.12_orig/drivers/net/wireguard/receive.c linux-5.15.12/drivers/net/wireguard/receive.c
--- linux-5.15.12_orig/drivers/net/wireguard/receive.c	2021-12-29 19:29:03.000000000 +0800
+++ linux-5.15.12/drivers/net/wireguard/receive.c	2022-01-02 19:46:17.000000000 +0800
@@ -477,7 +477,7 @@
 		if (unlikely(wg_socket_endpoint_from_skb(&endpoint, skb)))
 			goto next;
 
-		wg_reset_packet(skb, false);
+		wg_reset_packet(skb, false, !net_eq(dev_net(peer->device->dev), dev_net(skb->dev)));
 		wg_packet_consume_data_done(peer, skb, &endpoint);
 		free = false;
 
diff -uNr linux-5.15.12_orig/drivers/net/wireguard/send.c linux-5.15.12/drivers/net/wireguard/send.c
--- linux-5.15.12_orig/drivers/net/wireguard/send.c	2021-12-29 19:29:03.000000000 +0800
+++ linux-5.15.12/drivers/net/wireguard/send.c	2022-01-02 19:46:27.000000000 +0800
@@ -296,7 +296,7 @@
 		skb_list_walk_safe(first, skb, next) {
 			if (likely(encrypt_packet(skb,
 					PACKET_CB(first)->keypair))) {
-				wg_reset_packet(skb, true);
+				wg_reset_packet(skb, true, true);
 			} else {
 				state = PACKET_STATE_DEAD;
 				break;
diff -uNr linux-5.15.12_orig/drivers/net/wireguard/socket.c linux-5.15.12/drivers/net/wireguard/socket.c
--- linux-5.15.12_orig/drivers/net/wireguard/socket.c	2021-12-29 19:29:03.000000000 +0800
+++ linux-5.15.12/drivers/net/wireguard/socket.c	2022-01-02 19:47:16.000000000 +0800
@@ -20,11 +20,19 @@
 static int send4(struct wg_device *wg, struct sk_buff *skb,
 		 struct endpoint *endpoint, u8 ds, struct dst_cache *cache)
 {
+	u32 fwmark;
+	if (!wg->fwmark) {
+		fwmark = mark;
+	}
+	else {
+		fwmark = wg->fwmark;
+	}
+    
 	struct flowi4 fl = {
 		.saddr = endpoint->src4.s_addr,
 		.daddr = endpoint->addr4.sin_addr.s_addr,
 		.fl4_dport = endpoint->addr4.sin_port,
-		.flowi4_mark = wg->fwmark,
+		.flowi4_mark = fwmark,
 		.flowi4_proto = IPPROTO_UDP
 	};
 	struct rtable *rt = NULL;
@@ -33,7 +41,7 @@
 
 	skb_mark_not_on_list(skb);
 	skb->dev = wg->dev;
-	skb->mark = wg->fwmark;
+	skb->mark = fwmark;
 
 	rcu_read_lock_bh();
 	sock = rcu_dereference_bh(wg->sock4);
@@ -98,11 +106,19 @@
 		 struct endpoint *endpoint, u8 ds, struct dst_cache *cache)
 {
 #if IS_ENABLED(CONFIG_IPV6)
+	u32 fwmark;
+	if (!wg->fwmark) {
+		fwmark = mark;
+	}
+	else {
+		fwmark = wg->fwmark;
+	}
+    
 	struct flowi6 fl = {
 		.saddr = endpoint->src6,
 		.daddr = endpoint->addr6.sin6_addr,
 		.fl6_dport = endpoint->addr6.sin6_port,
-		.flowi6_mark = wg->fwmark,
+		.flowi6_mark = fwmark,
 		.flowi6_oif = endpoint->addr6.sin6_scope_id,
 		.flowi6_proto = IPPROTO_UDP
 		/* TODO: addr->sin6_flowinfo */
@@ -113,7 +129,7 @@
 
 	skb_mark_not_on_list(skb);
 	skb->dev = wg->dev;
-	skb->mark = wg->fwmark;
+	skb->mark = fwmark;
 
 	rcu_read_lock_bh();
 	sock = rcu_dereference_bh(wg->sock6);
@@ -321,6 +337,7 @@
 	wg = sk->sk_user_data;
 	if (unlikely(!wg))
 		goto err;
+	mark = skb->mark;
 	skb_mark_not_on_list(skb);
 	wg_packet_receive(wg, skb);
 	return 0;

```

And don't forget add the iptable rules, or it won't work:
```
iptables -t mangle -I mwan3_connected 1 -p udp --dport <wireguard listen port> -j RETURN
ip6tables -t mangle -I mwan3_connected 1 -p udp --dport <wireguard listen port> -j RETURN
```

@feckert 
Is there any better way to mark the wireguard incoming connections?

feckert, 2022-01-03:
If you change wireguard source code, you should contact them. And describe your problem. I don't think I can contribute much unless I look into it more. Maybe next week!

[27 more replies]

closed 2022-02-21 by wackejohn