openwrt/packages #9541: [RFC] Split setuptools and pip packaging from the python package?

jefferyto, 2019-07-22:
Maintainer: me, @commodo 
Environment: N/A

Description:
#9534 and #9539 brought up something I've thought about before: should we separate the setuptools and pip packages from the python package? (This would be for target packages only; for host python I think it's fine to install the bundled versions.)

These packages can be compiled the same way as other python packages (they are available from PyPI), and separating them means we can control the versioning (make updates for master, hold back for release branches).

@commodo Any thoughts? Are there (historical) reasons why they are packaged as they are now?

Also anyone else I should ping for this?

commodo, 2019-07-22:
> @commodo Any thoughts? Are there (historical) reasons why they are packaged as they are now?

historical, yes:

When i started packaging/supporting Python, I went straight for Python3 [I named it Python], and thought "we don't need Python[2]"; but I was proven wrong in short time, because people still wanted Python2; I had to rename Python to Python3, and introduce Python2 as Python; it was only me, so ¯\_(ツ)_/¯ ; there weren't [or did not seem to be] many users of Python & OpenWrt.

After Python[2] & Python3 were up and working, I thought "how do I let people use Python packages freely, without me having to work too much?". Answer: "offer pip + setuptools & pip3 & setuptools3"; they were packaged separately at some point in time. See commit https://github.com/openwrt/packages/commit/a53d0c5a403d1669e2cf6c59c2be6a9d3ed633a0  [this is when they were merged with the interpreters]. 

This proved annoying to maintain [as a single person], as you need to account for 6 Makefiles **[python{2,3}, setuptools{2,3}, pip{2,3}]**, 3 of which were duplicate. At that time, the neat PyPackage + Py3Package mechanisms did not exist. So, for each Python package you would have a Makefile, and for each Python3 package you would have another Makefile. And there were some bits of both of other Python & Python3 packages [but mostly Python2 packages].

Not to mention that pip & setuptools have [or used to have] really active development. Things would break [or get broken] in either setuptools or pip, or sometimes I would need to dig into Python's distutils. It seemed a bit too much work.

What I started to think was: "oh, I know: I just stick with Python's own versions of setuptools & pip". This simplified the support matrix quite a lot, as I would be sure that Python had validated the versions of `setuptools` & `pip` that they are shipping, and this meant reducing complexity from 6 packages to 2.

Now, moving forward: I am not against moving back to exporting setuptools & pip outside the Python packaging. It could be that things are better now.
One thing about doing that maybe, is that I would like to **not be** the maintainer for them [ if possible :) ] at least for a while. I would still help of course, as much as I can. It could be a new experience [a better experience or a learning experience].
[Personally] I really don't mind having `setuptools` & `pip` packaged with Python; the only downside [now as they are] is that they update slower than the projects put out releases.

jefferyto, 2019-07-22:
I can see how combining the packages would reduce the support complexity :joy: It would be easier now with Py[3]Package but I don't know if the compatibility situation has gotten better.

I'll sit on this for a few days... I'll take another look if I really can't stop myself :laughing:

commodo, 2019-07-22:
well, just as an idea; i am pretty fuzzy [memory-wise] about how difficult it was to maintain all 6 packages;
i also had other stuff to do/work on;
now they would be 3 Makefiles [vs 6 back then]; so the thing should be slightly easier to maintain;

jefferyto, 2019-10-04:
I still think being able to manage target setuptools/pip versioning is a good idea, but I don't think I'll be able to get around to it anytime soon. Closing this issue (for now).

closed 2019-10-04 by jefferyto