openwrt/packages #9554: pptpd can not

rufengsuixing, 2019-07-24:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

feckert, 2019-08-13:
Please add a description how to reproduce this issue.

neheb, 2020-01-28:
It's not starting because "enabled" is 0 in /etc/config/pptpd . Closing.

closed 2020-01-28 by neheb