openwrt/packages #9572: freeradius3 rlm_python module not built/missing

DomAmato, 2019-07-25:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: mips_24kc OpenWrt 18.06.1 r7258-5eb055306f 

Description:

Right now the python module isn't built for freeradius, it looks like an older patch was submitted for freeradius2: https://dev.archive.openwrt.org/ticket/12289.html

neheb, 2020-01-28:
@Robby- can you comment on this?

Robby-, 2020-01-30:
@neheb I tried compiling the python module but cannot get the freeradius rlm_python configure script to find `Python.h` even though its looking in the right path and the file is actually there when I go to check it manually.
`/home/robby/openwrt-master/staging_dir/hostpkg/include/python2.7/Python.h`
```
[...]
checking for arm-openwrt-linux-muslgnueabi-gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... arm-openwrt-linux-muslgnueabi-gcc -E
checking for python2.7... python2.7
configure: Python sys.prefix "/home/robby/openwrt-master/staging_dir/hostpkg"
configure: Python sys.exec_prefix "/home/robby/openwrt-master/staging_dir/hostpkg"
configure: Python sys.version "2.7"
configure: Python local_mod_libs "-lz"
configure: Python base_mod_libs ""
configure: Python other_libs "-Xlinker -export-dynamic -lpthread -ldl -lutil"
checking for Python.h in /home/robby/openwrt-master/staging_dir/hostpkg/include/python2.7/... no
checking for Python.h... no
checking for Python.h in /usr/local/include/... no
checking for Python.h in /opt/include/... no
checking for Py_Initialize in -lpython2.7 in /home/robby/openwrt-master/staging_dir/target-arm_xscale_musl_eabi/usr/lib/python2.7... yes
checking for dl_iterate_phdr... yes
configure: error: set --without-rlm_python to disable it explicitly.
configure: error: ./configure failed for src/modules/rlm_python
Makefile:762: recipe for target '/home/robby/openwrt-master/build_dir/target-arm_xscale_musl_eabi/freeradius-server-release_3_0_20/.configured_059ce960b95c68443337f43a915dbd53' failed
make[2]: *** [/home/robby/openwrt-master/build_dir/target-arm_xscale_musl_eabi/freeradius-server-release_3_0_20/.configured_059ce960b95c68443337f43a915dbd53] Error 1
make[2]: Leaving directory '/home/robby/openwrt-master/feeds/packages/net/freeradius3'
time: package/feeds/packages/freeradius3/compile#7.38#1.64#17.93
package/Makefile:111: recipe for target 'package/feeds/packages/freeradius3/compile' failed
make[1]: *** [package/feeds/packages/freeradius3/compile] Error 2
make[1]: Leaving directory '/home/robby/openwrt-master'
/home/robby/openwrt-master/include/toplevel.mk:225: recipe for target 'package/freeradius3/compile' failed
make: *** [package/freeradius3/compile] Error 2
robby@vopenwrt:~/openwrt-master$
```

I've attached a diff to the Makefile that is in net/freeradius3/Makefile that I made and I think is actually correct so far (but it might need more changes, not sure, I'm merely an advanced user, not a developer). I've renamed it to a .txt because GitHub doesn't accept a .diff file.
[fr3-python.txt](https://github.com/openwrt/packages/files/4134590/fr3-python.txt)

Maybe someone else can continue with the diff file I made and get the python module to compile.

jefferyto, 2020-01-30:
I tried compiling with your changes and configure was able to complete the sub-configure for rlm_python and rlm_python3. (configure later failed for ./drivers/rlm_sql_mongo and src/modules/rlm_sql.)

I suspect configure failed for you because it actually detected host Python instead of target Python (as it did for me). You can see it in this output (host Python is in `staging_dir/hostpkg`):

```
configure: Python sys.prefix "/home/robby/openwrt-master/staging_dir/hostpkg"
configure: Python sys.exec_prefix "/home/robby/openwrt-master/staging_dir/hostpkg"
```

I suggest patching [these lines](https://github.com/FreeRADIUS/freeradius-server/blob/release_3_0_20/src/modules/rlm_python/configure.ac#L68-L75) in configure.ac (or perhaps better the [corresponding lines](https://github.com/FreeRADIUS/freeradius-server/blob/release_3_0_20/src/modules/rlm_python/configure#L2931-L2941) in configure) to use target Python paths/version from the Makefile, instead of letting it call host Python. (And also make the corresponding change for rlm_python3.)

Robby-, 2020-02-02:
Thank you for taking a look at it.

> I suggest patching [these lines](https://github.com/FreeRADIUS/freeradius-server/blob/release_3_0_20/src/modules/rlm_python/configure.ac#L68-L75) in configure.ac (or perhaps better the [...]

How to patch this in a platform/architecture-agnostic way, though? Because I have no idea how to go about that.

I've managed to compile successfully by patching rlm_python's configure manually:
`nano +2974 /home/robby/openwrt-master/build_dir/target-arm_xscale_musl_eabi/freeradius-server-release_3_0_20/src/modules/rlm_python/configure`
And change the `smart_include_dir` line to:
`smart_include_dir="/home/robby/openwrt-master/staging_dir/target-arm_xscale_musl_eabi/usr/include/python2.7"`

And the same for rlm_python3:
`nano +2974 /home/robby/openwrt-master/build_dir/target-arm_xscale_musl_eabi/freeradius-server-release_3_0_20/src/modules/rlm_python3/configure`
And change the `smart_include_dir` line to:
`smart_include_dir="/home/robby/openwrt-master/staging_dir/target-arm_xscale_musl_eabi/usr/include/python3.8"`

By the way, I also had to update my Makefile diff because of an error with the conffiles section that only now occurred since it was finally going through.

jefferyto, 2020-02-02:
> How to patch this in a platform/architecture-agnostic way, though?

I think you can patch configure so that `PY_PREFIX` and `PY_EXEC_PREFIX` are set to `$STAGING_DIR`:

```
PY_PREFIX=$STAGING_DIR
...
PY_EXEC_PREFIX=$STAGING_DIR
```

Because `$(STAGING_DIR)` is [explicitly exported](https://github.com/openwrt/openwrt/blob/8216766ad97db2f20e30b6622343d8d473ad9d68/rules.mk#L242) in rules.mk, the `$STAGING_DIR` shell variable should be available.

Or if you want it to be more flexible, you can patch configure to set `PY_PREFIX` and `PY_EXEC_PREFIX` to a different environment variable, e.g.:

```sh
PY_PREFIX=$OPENWRT_PY_PREFIX
...
PY_EXEC_PREFIX=$OPENWRT_PY_PREFIX
```

Then set that variable in `CONFIGURE_VARS` in the package Makefile:

```makefile
CONFIGURE_VARS+= \
        ... \
        OPENWRT_PY_PREFIX=$(STAGING_DIR)
```

(Or you can even completely patch out the `PY_PREFIX` and `PY_EXEC_PREFIX` assignments in configure and set them directly in `CONFIGURE_VARS`.)

It is similar with `PY_SYS_VERSION`, except you would want to set it to `$(PYTHON_VERSION)` / `$(PYTHON3_VERSION)` (these variables are available through python-package.mk/python3-package.mk), but these variables are not exported by default.

Personally, I would use intermediate environment variables to set `PY_PREFIX`, `PY_EXEC_PREFIX`, and `PY_SYS_VERSION`, then set those variables in `CONFIGURE_VARS`.

Robby-, 2020-02-04:
Thank you so much @jefferyto for your insights and guidance. I went with the intermediate environment variables as I think it is a simple and clean solution.

I have created 2 pull requests to fix this (and another) issue in both the master and 19.07 branches.

jefferyto, 2020-02-04:
Oops I didn't see your comment edit - I should have let you investigate it :joy:

PR #11244 (2020-02-09): freeradius3: Fix proxy.conf file conflict and enable the rlm_python and rlm_python3 modules
Maintainer: no maintainer at this time
Compile tested: ARM, Linksys EA3500, master
Run tested: No

Description:
- Fix proxy.conf file conflict. This closes #11169.
- Enable the rlm_python and rlm_python3 modules. This closes #9572.
PR #11245 (2020-02-09): [19.07] freeradius3: Fix proxy.conf file conflict and enable the rlm_python and rlm_python3 modules
Maintainer: no maintainer at this time
Compile tested: ARM, Linksys EA3500, 19.07
Run tested: No

Description:
- Fix proxy.conf file conflict. This closes #11169.
- Enable the rlm_python and rlm_python3 modules. This closes #9572.
closed 2020-02-09 by neheb