openwrt/packages #9579: samba4 cross compile error...

adaaaaaa, 2019-07-27:
`Traceback (most recent call last):
  File "./buildtools/bin/waf", line 75, in <module>
    import Scripting
  File "/home/captain/openwrt/build_dir/hostpkg/samba-4.9.11/third_party/waf/wafadmin/Scripting.py", line 146
    except Utils.WafError,e:
                         ^
SyntaxError: invalid syntax
make[3]: *** [Makefile:474: /home/captain/openwrt/build_dir/hostpkg/samba-4.9.11/.configured] Error 1
make[3]: Leaving directory '/home/captain/openwrt/feeds/packages/net/samba4'
time: package/feeds/packages/samba4/host-compile#0.26#0.10#0.34
make[2]: *** [package/Makefile:107: package/feeds/packages/samba4/host/compile] Error 2
make[2]: Leaving directory '/home/captain/openwrt'
make[1]: *** [package/Makefile:103: /home/captain/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/captain/openwrt'
make: *** [/home/captain/openwrt/include/toplevel.mk:218：world] 错误 2
`
subtarget:X86_64

wongsyrone, 2019-07-27:
The same here on Archlinux

panda-mute, 2019-07-28:
@Andy2244

Andy2244, 2019-07-28:
mhh [packages](https://downloads.openwrt.org/snapshots/packages/x86_64/packages/) and [faillogs](https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/) look fine for the buildbots on 25.07 ?

Did a quick run for the x86_64 target via my [package builder](https://github.com/Andy2244/openwrt-package-builder) and this worked. The base image is based on debian:stretch-slim, my live dev system is based on ClearLinux and also worked.

Not sure what i can do here, this looks like a waf <-> distro problem to me, maybe try with just python 2.7 installed?

Andy2244, 2019-07-28:
ah just remembered something, can you try with this patch?
https://github.com/Andy2244/openwrt-extra/blob/samba-4.9/samba4/patches/100-do-not-import-target-module-while-cross-compile.patch

I did remove it in the last update, since for my test systems all worked fine without it, maybe this is still needed for some distros?

EdiTurn, 2019-07-28:
Got same error during compile libtalloc.

panda-mute, 2019-07-28:
This should be the issue of changing to python3.

Andy2244, 2019-07-28:
> This should be the issue of changing to python3.

Not sure the 4.9.x branch is fully python2 compatible, not even sure it has python3 support at all. I suspect they updated some waf libs recently, maybe compare 4.9.8 to 4.9.11 in the waf folders?

Problem is on all my dev systems all builds fine, so seem to-be distro/system related.

panda-mute, 2019-07-28:
I am using master branch source code compile, 4.9.11 is ok when openwrt build don't switch to python.

You may check this https://github.com/openwrt/openwrt/commit/19938c8de7a062626796f53a2805608c0dd4edbd

Andy2244, 2019-07-28:
> 4.9.11 is ok when openwrt build don't switch to python3

I mean what python version is installed for the buildbots/sdk, since this still builds fine. I did not try to build 4.9.x with only python3 installed, since its not officially supported. Only 4.10/11 have python3 support, but those branches have some other major bugs atm and i need to wait for official fixes before i can try a switch again.

EXcomZkko, 2019-07-29:
Same here. (Debian Testing)

wongsyrone, 2019-07-29:
For Python 3, I will try changing the line
```py3
except Utils.WafError,e:
```
to
```py3
except Utils.WafError as e:
```
and try again to see if other errors occur

Andy2244, 2019-07-29:
@wongsyrone i tried to play around with arch, but the [openwrt docu](https://openwrt.org/docs/guide-developer/build-system/install-buildsystem) is way out of date.

I get, while trying to setup a test environment on latest arch.
```
Build dependency: Please install the GNU C Compiler (gcc) 4.8 or later
Build dependency: Please install the GNU C++ Compiler (g++) 4.8 or later
Build dependency: Please install GNU 'tar'
Build dependency: Please install GNU 'find'
Build dependency: Please install GNU 'bash'
Build dependency: Please install GNU 'patch'
Build dependency: Please install diffutils
Build dependency: Please install GNU fileutils
Build dependency: Missing seq command
Build dependency: Please install GNU 'awk'
Build dependency: Please install GNU 'grep'
Build dependency: Please install an extended getopt version that supports --long
Build dependency: Cannot find a file stat utility
Build dependency: Please install 'unzip'
Build dependency: Please install 'bzip2'
Build dependency: Please install GNU 'wget'
Build dependency: Please install Perl 5.x
Build dependency: Please install Python >= 3.5
Build dependency: Please install Python >= 3.5
Build dependency: Please install Git (git-core) >= 1.7.12.2
Build dependency: Please install the 'file' package
```

The strange part is most of the stuff is already installed, can you give me a hint what i need install for the sdk to work?

Andy2244, 2019-07-29:
ok just confirmed on debian testing, the error is because openwrt did drop the python2 requirement on host for the sdk. Yet we still need a working python2 for samba4.
The fix is to just build it for the package aka add it to the makefile:
`HOST_BUILD_DEPENDS:=python/host nfs-kernel-server/host e2fsprogs/host`

PS: Will be fixed with the next update.

flipreverse, 2019-07-30:
@Andy2244 This will fix the build issue. It, however, introduces another issue.
Wiresless regdb won't build, because it calls the python binary which is a symlink to python2.7. This is due to the fact that samba4 adds python2 as a dependency.
I fixed it by adding the following to `/package/firmware/wireless-regdb/Makefile`:
`include $(TOPDIR)/feeds/packages/lang/python/python3-package.mk`
and changed ` $(PYTHON) $(PKG_BUILD_DIR)/db2fw.py $(PKG_BUILD_DIR)/regulatory.db $(PKG_BUILD_DIR)/db.txt` to ` $(PYTHON3) $(PKG_BUILD_DIR)/db2fw.py $(PKG_BUILD_DIR)/regulatory.db $(PKG_BUILD_DIR)/db.txt`

Andy2244, 2019-07-30:
Mhh thats strange the file uses `#!/usr/bin/env python` which is generally assumed to work with python2, while `#!/usr/bin/env python3` is used for explicit python3 support. At least thats what i have seen on some python projects.

So given the age of the project i'm surprised it wont build with python2?

flipreverse, 2019-07-30:
I'm sorry. I missed a point in my explanation. I forced wireles-regdb to use python3 since python2 is missing a module called builtins. The error when running `build_dir/target-mips_24kc_musl/wireless-regdb-2019.06.03/regulatory.db` is: `No module named builtins`.

Andy2244, 2019-07-30:
Yeah i guess thats the fix than, if it only builds with python3 anyway than it has to ensure it actually uses python3 and not rely on whatever symlink exists.

flipreverse, 2019-07-30:
Ic. Who is responsible for fixing this?
How should a fix might look like?

Andy2244, 2019-07-30:
Maintainer is listed as Felix Fietkau <nbd@nbd.name>
@nbd168 is that you? If so can you have a look at this please?

panda-mute, 2019-07-31:
@ynezz

ynezz, 2019-07-31:
This https://github.com/ynezz/openwrt/commit/e523a9d1f4d3391ac8c173c6b5d687bce0c529e7 should make wireless-regdb use proper python, python3, please try it and let me know if it's ok, thanks.

MartB, 2019-07-31:
cant even build my openwrt so its not only samba its everything kinda.

PR #9600 (2019-08-01): samba4: add python2 host dependency
Maintainer: me
Compile tested: x86_64 (snapshots)
Run tested:

Description:
Make sure we have a working python2 version for samba-4.9, since its no longer a sdk requirement.
Fixes #9579
closed 2019-08-01 by diizzyy