openwrt/packages #9581: wireless-regdb cross compile error

adaaaaaa, 2019-07-27:
```Traceback (most recent call last):
  File "/home/captain/openwrt/build_dir/target-x86_64_musl/wireless-regdb-2019.06.03/db2fw.py", line 6, in <module>
    from dbparse import DBParser
  File "/home/captain/openwrt/build_dir/target-x86_64_musl/wireless-regdb-2019.06.03/dbparse.py", line 3, in <module>
    from builtins import bytes
ImportError: No module named builtins
make[3]: *** [Makefile:34: /home/captain/openwrt/build_dir/target-x86_64_musl/wireless-regdb-2019.06.03/.built] Error 1
make[3]: Leaving directory '/home/captain/openwrt/package/firmware/wireless-regdb'
time: package/firmware/wireless-regdb/compile#0.18#0.04#1.25
make[2]: *** [package/Makefile:109: package/firmware/wireless-regdb/compile] Error 2
make[2]: Leaving directory '/home/captain/openwrt'
make[1]: *** [package/Makefile:103: /home/captain/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/captain/openwrt'
make: *** [/home/captain/openwrt/include/toplevel.mk:218：world] 错误 2
```

neheb, 2019-07-27:
That's a transition problem. Basically, staging_dir/host/bin/python points to python2, when it should point to python3. Wipe your build directory and try again.

ynezz, 2019-07-28:
> Basically, `staging_dir/host/bin/python` points to python2, when it should point to python3.

This case with `STAGING_DIR_HOST/host/bin/python` should be handled in [prereq step](https://github.com/openwrt/openwrt/commit/3b68fb57c938af3948ae4c2da61501183fbef649). What is probably not handled currently is `STAGING_DIR_HOSTPKG/bin (staging_dir/hostpkg/bin/python)` which can probably point to python2 if the user build python2 from the package feeds and then somehow this path takes precedence during the build, although it shouldn't:

```
rules.mk:TARGET_PATH_PKG:=$(STAGING_DIR)/host/bin:$(STAGING_DIR_HOSTPKG)/bin:$(TARGET_PATH)
include/package.mk:  $(1) : export PATH=$$(TARGET_PATH_PKG)
```

diizzyy, 2019-07-28:
wireless-regdb isn't in package repo, closing

closed 2019-07-28 by diizzyy

brucealthompson, 2019-08-05:
Please reopen this bug.

I have the exact same build issue using the latest source. The build issue 
is with the following target platform:
              Target System (Marvell EBU Armada)  --->                     │ │  
  │ │        Subtarget (Marvell Armada 37x/38x/XP)  --->                  │ │  
  │ │        Target Profile (Linksys WRT1900ACS (Shelby))  --->  

I believe the problem is that staging_dir/hostpkg/bin/python points to python2. When I changed staging_dir/hostpkg/bin/python to point to python3, the build problem went away.

ynezz, 2019-08-05:
@brucealthompson https://github.com/openwrt/openwrt/commit/b6bae4a2c9f11f7e55319c2b4c709396ce649688

brucealthompson, 2019-08-19:
I just pulled the "fix" referenced here from master. It does not work in my build environment. From looking at the github change logs it looks like the "fix" was to change the Makefile to use $(PYTHON) instead of python. In my build environment, $(PYTHON) translates to python. So there is no change in behavior of the build.

Please see my comment above for the  fix that works in my build environment. Here it is again:
I believe the problem is that staging_dir/hostpkg/bin/python points to python2. When I changed staging_dir/hostpkg/bin/python to point to python3, the build problem went away.

ynezz, 2019-08-19:
> looking at the github change logs it looks like the "fix" was to change the Makefile to use $(PYTHON) instead of python.

No, the change was following `$(PYTHON) -> $(STAGING_DIR_HOST)/bin/$(PYTHON)` which should contain a valid python3. If it doesn't, then please run `rm staging_dir/host/.prereq-build; make prereq`