openwrt/packages #9582: ModuleNotFoundError: No module named 'distutils.command'

teqlee, 2019-07-27:
Error compile with latest master+feeds with gcc 9 & binutils 2.32

```
make[3]: Entering directory '/home/myname/openwrtmaster/tools/scons'
./files/pywrap.sh /home/myname/openwrtmaster/build_dir/host/scons-3.0.5/setup.py install --prefix=/home/myname/openwrtmaster/staging_dir/host
Traceback (most recent call last):
  File "/home/myname/openwrtmaster/build_dir/host/scons-3.0.5/setup.py", line 26, in <module>
    import distutils.command.build_scripts
ModuleNotFoundError: No module named 'distutils.command'
make[3]: *** [Makefile:35: /home/myname/openwrtmaster/staging_dir/host/stamp/.scons_installed] Error 1
make[3]: Leaving directory '/home/myname/openwrtmaster/tools/scons'
time: tools/scons/compile#0.09#0.02#0.11
make[2]: *** [tools/Makefile:159: tools/scons/compile] Error 2
make[2]: Leaving directory '/home/myname/openwrtmaster'
make[1]: *** [tools/Makefile:155: /home/myname/openwrtmaster/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnyyyynyyyyyyyyyyyyyyyyyynynnyyynnyyy] Error 2
make[1]: Leaving directory '/home/myname/openwrtmaster'
make: *** [/home/myname/openwrtmaster/include/toplevel.mk:218: world] Error 2
```

dibdot, 2019-07-27:
Try something like `sudo apt install python3-distutils` (on debian systems)

teqlee, 2019-07-27:
Thanks @dibdot, it's worked.

closed 2019-07-27 by teqlee