openwrt/packages #9586: <softethervpn> Anyone tried this with 130-iconv.patch?

mriksman, 2019-07-28:
# Issue template (remove lines from top till here)

@ps790 @neheb 
Environment: (put here arch, model, OpenWrt version)
Cross-developing withing Windows 10 Ubuntu
Target is ar71xx, OpenWRT 18.06

Description:
First time trying to cross-compile to get a binary of softethervpn. Total n00b.

After completing `make prepare`, I performed a `make /packages/softethervpn V=99`, but it said there was an issue with `130-iconv.patch`. I removed it from the `/patches` directory and compile continued on.Not sure what the issue is with it (or even what the patch is supposed to fix).

neheb, 2019-07-29:
I see the problem. Will push a fix.

neheb, 2019-07-29:
Pushed. Please check again.

closed 2020-01-12 by neheb