openwrt/packages #9587: libhttp-parser - (cross) compile fails

leonardpitzu, 2019-07-28:
libhttp-parser - cross compile fails
Maintainer : @nxhack

Evnironment:
- openwrt snapshot -  r10572-209b368254
- MacOS Mojave
- Linksys WRT3200ACM

Description:

```
/Volumes/work/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-9.1.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/9.1.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: unrecognised option: -install_name
collect2: error: ld returned 1 exit status
make[4]: *** [library] Error 1
make[3]: *** [/Volumes/work/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/http-parser-2.9.2/.built] Error 2
```

nxhack, 2019-07-29:
Thank you for the report.

https://github.com/openwrt/packages/pull/9598

nxhack, 2019-07-31:
@leonardpitzu 

Merged. Please check it.

leonardpitzu, 2019-07-31:
@nxhack 
positively tested on macos mojave. issue can be closed . Thanks for your support.

closed 2019-07-31 by neheb