openwrt/packages #9589: wpa_supplicant with pcsc support (for EAP-SIM and EAP-AKA)

francescogmb, 2019-07-28:
Maintainer: Jouni Malinen <j@w1.fi>
Environment: v18.06.1

Description:
to enable wpa_supplicant with EAP-SIM or EAP-AKA I edited
the file wpa_supplicant-full.config and set:
CONFIG_PCSC=y
CONFIG_EAP_AKA=y or CONFIG_EAP_SIM=y

The problem is that file file pcsc_funcs.c
is not updated:
two included are lost (to recall pcsc) but afther patch this you get this error:
or similar
```
make[4]: 'wpa_cli' is up to date.
  CC  ../src/utils/pcsc_funcs.c
../src/utils/pcsc_funcs.o: In function `scard_transmit':
pcsc_funcs.c:(.text.scard_transmit+0x22): undefined reference to `SCardTransmit'
pcsc_funcs.c:(.text.scard_transmit+0x44): undefined reference to `g_rgSCardT0Pci'
pcsc_funcs.c:(.text.scard_transmit+0x4c): undefined reference to `g_rgSCardT1Pci'
../src/utils/pcsc_funcs.o: In function `scard_deinit':
pcsc_funcs.c:(.text.scard_deinit+0xa): undefined reference to `SCardDisconnect'
pcsc_funcs.c:(.text.scard_deinit+0x14): undefined reference to `SCardReleaseContext'
../src/utils/pcsc_funcs.o: In function `scard_init':
pcsc_funcs.c:(.text.scard_init+0x22): undefined reference to `SCardEstablishContext'
pcsc_funcs.c:(.text.scard_init+0x34): undefined reference to `SCardListReaders'
pcsc_funcs.c:(.text.scard_init+0x52): undefined reference to `SCardListReaders'
pcsc_funcs.c:(.text.scard_init+0xe8): undefined reference to `SCardConnect'
pcsc_funcs.c:(.text.scard_init+0x106): undefined reference to `SCardBeginTransaction'
pcsc_funcs.c:(.text.scard_init+0x16c): undefined reference to `SCardEndTransaction'
pcsc_funcs.c:(.text.scard_init+0x354): undefined reference to `SCardEndTransaction'
collect2: error: ld returned 1 exit status
Makefile:1923: recipe for target 'wpa_supplicant' failed
make[4]: *** [wpa_supplicant] Error 1
```
The problem may be in  the variables declaration  enviroment.

note: no problem compiling the singole pcsc service.

pilovis, 2019-10-19:
Same problem here :(

abda11ah, 2023-07-30:
Any news about this ?

How is the EAP-SIM support on the latest OpenWRT (connecting to a Wi-Fi network by a SIM card authentication) ?