openwrt/packages #9621: python3 - cross compile fails

leonardpitzu, 2019-08-01:
Maintainer : @jefferyto

Evnironment:
- openwrt snapshot - r10572-209b368254
- MacOS Mojave
- Linksys WRT3200ACM

```
(cd /Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4/; if [ -x configure ]; then cp -fpR /Volumes/work/openwrt/scripts/config.{guess,sub} /Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4// && CC="gcc" CFLAGS="-O2 -I/Volumes/work/openwrt/staging_dir/host/include -I/Volumes/work/openwrt/staging_dir/hostpkg/include -I/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" CXX="g++" CPPFLAGS="-I/Volumes/work/openwrt/staging_dir/host/include -I/Volumes/work/openwrt/staging_dir/hostpkg/include -I/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" LDFLAGS="-L/Volumes/work/openwrt/staging_dir/host/lib -L/Volumes/work/openwrt/staging_dir/hostpkg/lib -L/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib $(pkg-config --static --libs libcrypto libssl) -Wl,-rpath=/Volumes/work/openwrt/staging_dir/hostpkg/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-apple-darwin18.7.0 --host=x86_64-apple-darwin18.7.0 --build=x86_64-apple-darwin18.7.0 --program-prefix="" --program-suffix="" --prefix=/Volumes/work/openwrt/staging_dir/hostpkg --exec-prefix=/Volumes/work/openwrt/staging_dir/hostpkg --sysconfdir=/Volumes/work/openwrt/staging_dir/hostpkg/etc --localstatedir=/Volumes/work/openwrt/staging_dir/hostpkg/var --sbindir=/Volumes/work/openwrt/staging_dir/hostpkg/bin --without-cxx-main --without-pymalloc --with-threads --prefix=/Volumes/work/openwrt/staging_dir/hostpkg --exec-prefix=/Volumes/work/openwrt/staging_dir/hostpkg --with-system-expat=/Volumes/work/openwrt/staging_dir/hostpkg --with-ensurepip=upgrade CONFIG_SITE= ; fi )
configure: WARNING: unrecognized options: --with-threads
checking build system type... x86_64-apple-darwin18.7.0
checking host system type... x86_64-apple-darwin18.7.0
checking for python3.7... python3.7
checking for --enable-universalsdk... no
checking for --with-universal-archs... no
checking MACHDEP... checking for --without-gcc... no
checking for --with-icc... no
checking for x86_64-apple-darwin18.7.0-gcc... gcc
checking whether the C compiler works... no
configure: error: in `/Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[2]: *** [/Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4/.configured] Error 77
make[2]: Leaving directory `/Volumes/work/openwrt/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#8.94#2.50#13.86
make[1]: *** [package/feeds/packages/python3/host/compile] Error 2
```

jefferyto, 2019-08-01:
I guess this is same as #7171 but for Python 3?

@commodo I think the patch in https://github.com/openwrt/packages/issues/7171#issuecomment-469287780 might work but I don't have a Mac running any recent version of macOS so I cannot test it.

jefferyto, 2019-08-01:
@leonardpitzu Can you post your config.log?

leonardpitzu, 2019-08-01:
@jefferyto @commodo it's not the same patch. see attached logs - the "original" one is without the patch while the "patched" one is with the patch in #7171 (which by the way still applies to py 2.7).
[original_config.log](https://github.com/openwrt/packages/files/3456833/original_config.log)
[config_patched.log](https://github.com/openwrt/packages/files/3456834/config_patched.log)

error without patch
```
(cd /Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4/; if [ -x configure ]; then cp -fpR /Volumes/work/openwrt/scripts/config.{guess,sub} /Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4// && CC="gcc" CFLAGS="-O2 -I/Volumes/work/openwrt/staging_dir/host/include -I/Volumes/work/openwrt/staging_dir/hostpkg/include -I/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" CXX="g++" CPPFLAGS="-I/Volumes/work/openwrt/staging_dir/host/include -I/Volumes/work/openwrt/staging_dir/hostpkg/include -I/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" LDFLAGS="-L/Volumes/work/openwrt/staging_dir/host/lib -L/Volumes/work/openwrt/staging_dir/hostpkg/lib -L/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib $(pkg-config --static --libs libcrypto libssl) -Wl,-rpath=/Volumes/work/openwrt/staging_dir/hostpkg/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-apple-darwin18.7.0 --host=x86_64-apple-darwin18.7.0 --build=x86_64-apple-darwin18.7.0 --program-prefix="" --program-suffix="" --prefix=/Volumes/work/openwrt/staging_dir/hostpkg --exec-prefix=/Volumes/work/openwrt/staging_dir/hostpkg --sysconfdir=/Volumes/work/openwrt/staging_dir/hostpkg/etc --localstatedir=/Volumes/work/openwrt/staging_dir/hostpkg/var --sbindir=/Volumes/work/openwrt/staging_dir/hostpkg/bin --without-cxx-main --without-pymalloc --with-threads --prefix=/Volumes/work/openwrt/staging_dir/hostpkg --exec-prefix=/Volumes/work/openwrt/staging_dir/hostpkg --with-system-expat=/Volumes/work/openwrt/staging_dir/hostpkg --with-ensurepip=upgrade CONFIG_SITE= ; fi )
configure: WARNING: unrecognized options: --with-threads
checking build system type... x86_64-apple-darwin18.7.0
checking host system type... x86_64-apple-darwin18.7.0
checking for python3.7... python3.7
checking for --enable-universalsdk... no
checking for --with-universal-archs... no
checking MACHDEP... checking for --without-gcc... no
checking for --with-icc... no
checking for x86_64-apple-darwin18.7.0-gcc... gcc
checking whether the C compiler works... no
configure: error: in `/Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[2]: *** [/Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4/.configured] Error 77
make[2]: Leaving directory `/Volumes/work/openwrt/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#8.94#2.50#13.86
make[1]: *** [package/feeds/packages/python3/host/compile] Error 2
```

error with patch
```
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libpython3.7.a(pymath.o) has no symbols
gcc -L/Volumes/work/openwrt/staging_dir/host/lib -L/Volumes/work/openwrt/staging_dir/hostpkg/lib -L/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib -L/Volumes/work/openwrt/staging_dir/hostpkg/lib -lssl -lcrypto -Wl, -L/Volumes/work/openwrt/staging_dir/host/lib -L/Volumes/work/openwrt/staging_dir/hostpkg/lib -L/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib -L/Volumes/work/openwrt/staging_dir/hostpkg/lib -lssl -lcrypto -Wl,   -Wl,-stack_size,1000000  -framework CoreFoundation -o python Programs/python.o libpython3.7.a -ldl  -framework CoreFoundation -lz     
ld: warning: directory not found for option '-L/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib'
ld: warning: directory not found for option '-L/Volumes/work/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib'
Undefined symbols for architecture x86_64:
  "_libintl_bindtextdomain", referenced from:
      _PyIntl_bindtextdomain in libpython3.7.a(_localemodule.o)
  "_libintl_dcgettext", referenced from:
      _PyIntl_dcgettext in libpython3.7.a(_localemodule.o)
  "_libintl_dgettext", referenced from:
      _PyIntl_dgettext in libpython3.7.a(_localemodule.o)
  "_libintl_gettext", referenced from:
      _PyIntl_gettext in libpython3.7.a(_localemodule.o)
  "_libintl_setlocale", referenced from:
      _PyLocale_setlocale in libpython3.7.a(_localemodule.o)
      _PyLocale_localeconv in libpython3.7.a(_localemodule.o)
  "_libintl_textdomain", referenced from:
      _PyIntl_textdomain in libpython3.7.a(_localemodule.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [python] Error 1
make[2]: *** [/Volumes/work/openwrt/build_dir/hostpkg/Python-3.7.4/.built] Error 2
make[2]: Leaving directory `/Volumes/work/openwrt/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#74.61#24.13#108.37
make[1]: *** [package/feeds/packages/python3/host/compile] Error 2
```

jefferyto, 2019-08-06:
I spent the last few days setting up my Mac to try this... I thought I would share what I found so far.

(I'm running Mac OS X 10.11 El Capitan, followed [this guide](https://openwrt.org/docs/guide-developer/buildroot.exigence.macosx) to set up my environment using Homebrew. Tried compiling both host Python 2 and 3.)

* The Mac linker does have an `-rpath` option, but it only accepts a space between the option flag and argument (`-rpath XXX`) and not an equals sign (`-rpath=XXX`). So changing the LDFLAGS to `-Wl$(comma)-rpath$(comma)$(STAGING_DIR_HOSTPKG)/lib` will work (and should work on Linux as well). I haven't verified that this rpath setting works correctly on Mac though.

* The rpath setting shouldn't affect linking to the SSL libraries, because those are in `$(STAGING_DIR_HOST)/lib` whereas the rpath command is for `$(STAGING_DIR_HOSTPKG)/lib`. (The Python 3 change that added the rpath command (6ade5a1e3adb189d062b707e2b8f731c9faab844) specifically mentions libffi, and that is stored in `$(STAGING_DIR_HOSTPKG)/lib`.)

  (According to 4e2f4a76e35a3bfe880abf6f425ef688a25aebc2 host Python should use it's own libffi rather than the one from the packages feed though, so why was the rpath setting necessary? :thinking:)

  (Also wonder if adding `$(STAGING_DIR_HOSTPKG)/lib` to the library search path is a change that needs to be made in OpenWrt core :thinking:)

* By making the rpath change mentioned above, I can get past configure and into compilation, but it fails (for both Python 2 and 3) with an `unknown type name 'u_int'` error. This is going to take a bit to explain:

  * The `u_int` type is used in `/usr/include/sys/random.h`.
  * Python wants to include this header (in [Python 2](https://github.com/python/cpython/blob/2.7/Python/random.c#L6-L8) and [Python 3](https://github.com/python/cpython/blob/3.7/Python/bootstrap_hash.c#L15-L17)) because both `HAVE_SYS_RANDOM_H` and `HAVE_GETENTROPY` was set during configure.
  * The `(defined(HAVE_GETRANDOM) || defined(HAVE_GETENTROPY))` condition was [specifically added to fix this error](https://bugs.python.org/issue29057) by stopping this header file from being included. Neither `getrandom()` nor `getentropy()` should be defined on Mac.
  * But in our case, `getentropy()` is defined because `HOST_LDFLAGS` added `-lssl -lcrypto` and `getentropy()` is available in libcrypto. This is why configure detected `getentropy()` and set `HAVE_GETENTROPY`.

So now I'm at a bit of a loss as to how to proceed. (Apologies @leonardpitzu I know this doesn't go very far in addressing your issue.)

@commodo Do you recall why ee88853e04b3caa2bdbfd73a5d991baebf59df10 was necessary? It looks like this only adds `-lssl -lcrypto` (and maybe some other libraries) but not a `-static` flag, so the libraries are not statically linked (though I haven't verified this yet).

leonardpitzu, 2019-08-09:
@jefferyto great debug session. i would say that this needs some fixing if openwrt still considers macos as a supported dev platform (and my 2cents point to further support macos as more and more developers migrate to this platform). Python2 is *still* needed but i think not a lot of effort should be put into it as it will anyways die January 1st 2020.
I am a HW developer myself and have 2 major allergies - Makefiles and Java so i am not much of help around patching this unless it's *easy* (i tend to do *stupid* makefiles and  use the manual at every step 😅)

commodo, 2019-08-12:
> @commodo Do you recall why [ee88853](https://github.com/openwrt/packages/commit/ee88853e04b3caa2bdbfd73a5d991baebf59df10) was necessary? It looks like this only adds `-lssl -lcrypto` (and maybe some other libraries) but not a `-static` flag, so the libraries are not statically linked (though I haven't verified this yet). (I found the PR that this commit was part of (#3818), still trying to understand the causes and effects sweat_smile)

TL;DR: that is needed to be able to build Python[3]-host together with the SSL module for OS X & Ubuntu 12.04 [which needed some support up until a few years ago];  pip[3] need to support `https://` for when downloading packages, and you need openssl-dev/libressl for that;

there used to be a time when (for OpenWrt) you would need to install `libssl-dev` ( openssl development toolkit ) to be able to build some things;
this caused weird reports with various distros, that have different versions of openssl-dev;

at some point in time, for OS X, openssl-dev was deprecated for homebrew, which is usually popular for building OpenWrt on Mac; not sure about macports or other;
that's about the time that libressl was introduced in the `tools/` sub-folder [initially for Mac];
then it got extended to Linux as well, and you pretty much shouldn't need to install openssl-dev via apt/yum/brew on your system anymore;
libressl only outputs static libs, so Python[3] use[s] those static libs; and that's how it needs to be linked; and this worked for a while, but I think I also had a Mac around to be able to test this; I don't have one at-hand [unfortunately]

commodo, 2019-08-12:
one thing that stands out is the fact that Python host builds as a framework; I see a `-framework` directive.
from other experiences OS X frameworks are a bit buggy;

can you try to disable it by adding maybe a `--disable-framework` somewhere around here https://github.com/openwrt/packages/blob/master/lang/python/python3/Makefile#L302 ?

commodo, 2019-08-12:
> from other experiences OS X frameworks are a bit buggy;
> 

hmm; buggy may not be the word; annoying would be the word;

jefferyto, 2019-08-12:
I tried removing `$$$$(pkg-config --static --libs libcrypto libssl)` earlier. It seemed to compile fine in my Linux environment. (Also tried compiled youtube-dl, python-cryptography and python-crypto using that host Python, and also manually tried `import ssl` inside host Python).

When I tried it on Mac, I got past the `unknown type name 'u_int'` error, then hit the `Undefined symbols` error (for `_libintl_bindtextdomain`, etc.) that @leonardpitzu got. (This was because I had compiled (target) Python 2 earlier, which caused host gettext-full to be compiled. Host Python wanted to use this gettext-full but I think there is an incompatibility; host (and target) Python 3 builds fine in a clean buildroot. I haven't had time to fully debug this yet.)

I'll try `--disable-framework` as well, but if things work correctly without `$$$$(pkg-config --static --libs libcrypto libssl)` I would prefer to remove it.

commodo, 2019-08-12:
> if things work correctly without `$$$$(pkg-config --static --libs libcrypto libssl)` I would prefer to remove it.

maybe things would work fine with current Python versions;
I don't have strong objections to try to remove it;

PR #9762 (2019-08-17): [openwrt-19.07] python,python3: Fix host Python compilation for macOS
Maintainer: me, @commodo 
Compile tested: none (cherry-picked from #9760)
Run tested: none

Description:
* Remove `$$$$(pkg-config --static --libs libcrypto libssl)` from `HOST_LDFLAGS`

  Having this leads to an `unknown type name 'u_int'` error on Mac. Removing it doesn't appear to affect Python's ability to find buildroot LibreSSL.

* Change `-Wl,-rpath=...` to `-Wl,-rpath,...` in `HOST_LDFLAGS`

  The equals sign version is not supported by the Mac linker (appears to be an GNU extension). The comma version is supported;  [... cut]
commit e4507482 (2019-08-17): python,python3: Fix host Python compilation for macOS

* Remove $$$$(pkg-config --static --libs libcrypto libssl) from
  HOST_LDFLAGS

  Having this leads to an "unknown type name 'u_int'" error on Mac.
  Removing it doesn't appear to affect Python's ability to find
  buildroot LibreSSL.

* Change -Wl,-rpath=... to -Wl,-rpath,... in HOST_LDFLAGS

  The equals sign version is not supported by the Mac linker (appears to
  be an GNU extension). The comma version is supported; -rpath and its
  argument will be separated by a space when passed to the linker.

* Add ac_cv_header_libintl_h=no to HOST [... cut]
PR #9760 (2019-08-17): python,python3: Fix host Python compilation for macOS
Maintainer: me, @commodo 
Compile tested: armvirt-64, 2019-08-16 snapshot sdk (Linux) / x86-64, OpenWrt master (Mac)
Run tested: manual testing with host Python (`import ssl`, `ssl.OPENSSL_VERSION`), checked rpath

Description:
* Remove `$$$$(pkg-config --static --libs libcrypto libssl)` from `HOST_LDFLAGS`

  Having this leads to an `unknown type name 'u_int'` error on Mac. Removing it doesn't appear to affect Python's ability to find buildroot LibreSSL.

* Change `-Wl,-rpath=...` to `-Wl,-rpath,...` in `HOST_LDFLAGS`

  The equals sign ve [... cut]
commit 229c23a4 (2019-08-17): python,python3: Fix host Python compilation for macOS

* Remove $$$$(pkg-config --static --libs libcrypto libssl) from
  HOST_LDFLAGS

  Having this leads to an "unknown type name 'u_int'" error on Mac.
  Removing it doesn't appear to affect Python's ability to find
  buildroot LibreSSL.

* Change -Wl,-rpath=... to -Wl,-rpath,... in HOST_LDFLAGS

  The equals sign version is not supported by the Mac linker (appears to
  be an GNU extension). The comma version is supported; -rpath and its
  argument will be separated by a space when passed to the linker.

* Add ac_cv_header_libintl_h=no to HOST [... cut]
closed 2019-08-17 by neheb