openwrt/packages #9630: rtl-sdr  : zero copy issues fixed in master at osmocom, not yet in a release

DanielRIOT, 2019-08-02:
Maintainer: @TimelessNL :   Jasper Scholte <NightNL@outlook.com>
Environment: ARM

Info only, not in a release yet but useful for those that get this error when using rtl_sdr based things on ARM targets
Reading samples in async mode...
```
...
Allocating 15 zero-copy buffers
[  149.076639] Unhandled fault: external abort on non-linefetch (0x008) at 0xb6e6f000
[  149.076691] pgd = b35b6ccd
[  149.083110] [b6e6f000] *pgd=8e53e835, *pte=9250f71f, *ppte=9250fe7f
Bus error
```

Description:
 Osmocom recently pushed a fix for the zero-copy buffer issue on Arm targets to their master branch ( its not a in a release yet ) but this fixes a crash on OpenWRT ARM targets ( IPQ4019, Rpi etc ) for people that run rtl_sdr or related applications ( rtl_433, adsb, rtl_ais etc ) on their routers.

[commit is here ](https://github.com/osmocom/rtl-sdr/commit/81833a1cf6288fee93a9157c0f60cafb5ec340b9)

TimelessNL, 2019-08-02:
I can take a look at this commit and see how easy (related changesets) to backport this somewhere next week. But I'm not the official maintainer of rtl-sdr, only of rtl-433.

DanielRIOT, 2019-08-05:
no problem , I don't think its critical - there are probably very few people doing this on routers...

I've been running the master version on my WPJ428 (IPQ4028 ARM v7 CPU).

I'd wait until the release from rtl-sdr and then update that package.

This issue was only put here as a interm solution for the few that use rtl-sdr based applications ( like rtl-433) on routers / rpi and are getting this error- no need for everyone to go down the rabbithole.

TimelessNL, 2019-08-05:
Ah indeed. People facing this issue can use your suggestion and compile their own bugfix I suppose. Otherwise we may import some patches while the next release may be very soon.

Strykar, 2020-12-19:
@DanielRIOT Current rtl-sdr package is based on 0.6.0, is this fixed?