openwrt/packages #9631: banip: /etc/banip/banip.whitelist keeps growing on each router reboot

R-Adrian, 2019-08-02:
Maintainer: @dibdot 
Environment:
```
system type             : Qualcomm Atheros QCA9558 ver 1 rev 0
machine                 : TP-LINK TL-WR1043ND v2
processor               : 0
cpu model               : MIPS 74Kc V5.0
```
using a custom build (made from snapshot)
 SuperWRT SNAPSHOT, r10641-1d2b2e744e
```
#cat /proc/version
Linux version 4.14.134 (contact@superwrt.download) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r9875-3e803499c3)) #0 Wed Jul 31 14:53:41 2019
```

**Description:**
i use dual-stack PPPoE IPv4 + IPv6 dynamic addresses from my ISP and have a daily reboot scheduled in crontab.

A while ago i discovered that the banip whitelist (`/etc/banip/banip.whitelist`) keeps growing and on each reboot the ip addresses (v4 and v6) of the active wan interfaces keep getting added to the whitelist and are never removed.

Since my ISP allocates different addresses each time that the PPPoE session is re-created, that means the whitelist keeps growing forever, a little bit at each reboot.

Am I doing something wrong or is the whitelist really supposed to be cleaned up manually from time to time?

current configuration of /etc/config/banip is:
```
config banip 'global'
        option ban_automatic '1'
        option ban_fetchutil 'uclient-fetch'
        option ban_iface 'wan'
        option ban_enabled '1'

config banip 'extra'
        option ban_debug '0'
        option ban_nice '10'
        option ban_backup '0'
        option ban_maxqueue '1'

(...snip... 
and then are sections for the definitions of the individual lists themselves... using the standard format there. )
```

dibdot, 2019-08-02:
Works as designed! :smile: 
New ISP/Uplink subnets will be always automatically added to the whitelist  & whitelist ipset to prevent major lock outs. Maybe I could add an extra option in a future version, that these kind of entries are only stored temporary in the whitelist ipset but not in the whitelist itself.
Anyway, you have to maintain the whitelist always manually ...

R-Adrian, 2019-08-02:
yes, please, 

storing them temporary only in the ipset instead of appending them to the file will limit the wear-out of the router's flash memory chips. (minuscule as it is, it can add up)

Also, having to manually scrub the whitelist stale IP addresses is becoming annoying.

PR #9669 (2019-08-06): banip: update 0.1.5
Maintainer: me / @dibdot
Compile tested: -
Run tested: GL.iNet GL-AR750S, OpenWrt SNAPSHOT r10644-cb49e46a8a

Description:
* add extra options to control auto-addons to
  blacklist & whitelist ('ban_autoblacklist' & 'ban_autowhitelist',
  both enabled by default). If disabled auto-addons are only stored
  temporary in the black/whitelist ipset but not in the list itself, fixes #9631
* remove old, no longer needed procd workaround
* remove 'zeus' source from default config (discontinued)
commit c8e16134 (2019-08-06): banip: update 0.1.5

* add extra options to control auto-addons to
  blacklist & whitelist ('ban_autoblacklist' & 'ban_autowhitelist',
  both enabled by default). If disabled auto-addons are only stored
  temporary in the black/whitelist ipset but not in the list itself,
  fixes #9631
* remove old, no longer needed procd workaround
* remove 'zeus' source from default config (discontinued)
closed 2019-08-06 by diizzyy