openwrt/packages #9636: branch:master,vsftpd is error:500 OOPS: child died

zdz2019, 2019-08-03:
Use the x86_64 platform,
Use branch:master commit ba8aeb02eae26771b89390403462e6841621069e
package:vsftpd_3.0.3-3
Never working properly, the error is 500 OOPS: child died

neheb, 2019-08-03:
I was the last person to touch this. I looked at my patch. I see nothing that can cause issues. Assuming you compile your own builds, things to try are to remove patches as well as overriding Os with O2.

zdz2019, 2019-08-04:
I think this problem may be caused by a kernel failure.
I use the latest master source code test,All are default, no other patches or modifications added.
I found these error messages in the system log:

Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.339295] WARNING: CPU: 0 PID: 34 at __nf_unregister_net_hook+0x43/0x110
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.346239] Modules linked in: pppoe ppp_async snd_usb_audio snd_hda_intel snd_hda_codec_realtek snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_codec pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm snd_usbmidi_lib snd_hda_core slhc r8169 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.418275]  nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_raw iptable_mangle iptable_filter ip_tables e1000e crc_ccitt bnx2 asn1_decoder arptable_filter arpt_mangle arp_tables fuse sg evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat ip6t_NPT ip6t_MASQUERADE nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_rt ip6t_mh
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.489843]  ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip_gre gre ixgbe igbvf igb e1000 mdio ip_tunnel tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore vfat fat udf crc_itu_t hfsplus raid456 async_raid6_recov async_pq async_xor xor raid6_pq async_memcpy async_tx raid10 raid1 raid0 md_mod nls_utf8 nls_iso8859_1 nls_cp936 nls_cp437 sha1_ssse3 sha1_generic authenc uas sdhci_pltfm xhci_plat_hcd ehci_platform xfs libcrc32c reiserfs exfat button_hotplug ptp realtek pps_core mii libphy
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.548849] CPU: 0 PID: 34 Comm: kworker/u8:2 Not tainted 4.19.62 #0
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.555258] Hardware name:  YL-1900L4-V2/YL-1900L4-V2, BIOS 5.6.5 05/08/2017
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.562338] Workqueue: netns cleanup_net
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.566298] RIP: 0010:__nf_unregister_net_hook+0x43/0x110
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.571734] Code: f9 ff ff 48 85 c0 0f 84 e0 00 00 00 49 89 c5 49 89 ff 48 c7 c7 e0 fd 07 82 41 89 f6 e8 76 41 0e 00 4d 8b 65 00 4d 85 e4 75 13 <0f> 0b 48 c7 c7 e0 fd 07 82 e8 bf 3b 0e 00 e9 af 00 00 00 41 0f b7
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.590631] RSP: 0018:ffffc9000013bd88 EFLAGS: 00010246
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.595887] RAX: 0000000000000000 RBX: ffffffffa05b7040 RCX: 0000000000000000
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.603074] RDX: ffff88813b1dac00 RSI: 0000000000000002 RDI: ffffffff8207fde0
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.610264] RBP: ffffc9000013bdb0 R08: 0000000000000001 R09: ffff88813be89580
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.617460] R10: ffffc9000013bcf0 R11: 0000000000000001 R12: 0000000000000000
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.624656] R13: ffff888135ba8e00 R14: 0000000000000002 R15: ffff888135ba8000
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.631853] FS:  0000000000000000(0000) GS:ffff88813ba00000(0000) knlGS:0000000000000000
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.640002] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.645783] CR2: 000000000041e420 CR3: 0000000002008000 CR4: 00000000001006f0
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.652970] Call Trace:
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.655430]  nf_unregister_net_hook+0x3b/0x40
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.659822]  nf_unregister_net_hooks+0x2a/0x40
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.664304]  0xffffffffa05b5045
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.667459]  ops_exit_list.isra.9+0x39/0x60
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.671673]  cleanup_net+0x177/0x260
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.675264]  process_one_work+0x1d2/0x320
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.679311]  worker_thread+0x20d/0x3c0
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.683091]  kthread+0x113/0x120
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.686350]  ? process_one_work+0x320/0x320
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.690564]  ? kthread_create_worker_on_cpu+0x40/0x40
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.695647]  ret_from_fork+0x35/0x40
Mon Aug  5 00:32:57 2019 kern.warn kernel: [  140.699260] ---[ end trace 4dcba9d46f65c6f3 ]---
Mon Aug  5 00:37:52 2019 kern.info kernel: [  435.034417] nf_conntrack: default automatic helper assignment has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT target to attach helpers instead.

neheb, 2020-02-23:
This looks like a dupicate of https://github.com/openwrt/packages/issues/10057 . Closing.

closed 2020-02-23 by neheb

Veyrdite, 2022-11-15:
This bug still exists @DazzyWalkman @neheb.  It looks like I should probably post this here and not in #10057 

An alternative fix is to opkg install libseccomp procd-seccomp. These are currently not deps of this package but it doesn't work by default without them.  Installing them gets past (at least) this one particular issue with vsftpd.