openwrt/packages #9637: lcdproc-dirvers:  Error relocating /usr/lib/lcdproc/hd44780.so: outb: symbol not found

damocles-git, 2019-08-03:
the lcdproc-drivres seems broken:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.4'
DISTRIB_REVISION='r7808-ef686b7292'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 18.06.4 r7808-ef686b7292'
```
```
Symbols for /usr/lib/lcdproc/hd44780.so:
	ldd (0x77886000)
	libusb-0.1.so.4 => /usr/lib/libusb-0.1.so.4 (0x77852000)
	libftdi1.so.2 => /usr/lib/libftdi1.so.2 (0x77834000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x77816000)
	libugpio.so.1 => /usr/lib/libugpio.so.1 (0x77802000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x777de000)
	libc.so => ldd (0x77886000)
Error relocating /usr/lib/lcdproc/hd44780.so: sock_recv: symbol not found
Error relocating /usr/lib/lcdproc/hd44780.so: sock_close: symbol not found
Error relocating /usr/lib/lcdproc/hd44780.so: report: symbol not found
Error relocating /usr/lib/lcdproc/hd44780.so: sock_connect: symbol not found
Error relocating /usr/lib/lcdproc/hd44780.so: outb: symbol not found
Error relocating /usr/lib/lcdproc/hd44780.so: sock_send: symbol not found
Error relocating /usr/lib/lcdproc/hd44780.so: inb: symbol not found

Symbols for /usr/lib/lcdproc/curses.so:
	ldd (0x770e8000)
	libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0x7708c000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77068000)
	libc.so => ldd (0x770e8000)
Error relocating /usr/lib/lcdproc/curses.so: report: symbol not found

Symbols for /usr/lib/lcdproc/lis.so:
	ldd (0x774f0000)
	libftdi1.so.2 => /usr/lib/libftdi1.so.2 (0x774bc000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x7749e000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7747a000)
	libc.so => ldd (0x774f0000)
Error relocating /usr/lib/lcdproc/lis.so: report: symbol not found

Symbols for /usr/lib/lcdproc/glcd.so:
	ldd (0x770a8000)
	libusb-0.1.so.4 => /usr/lib/libusb-0.1.so.4 (0x7707e000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7705a000)
	libc.so => ldd (0x770a8000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x7703c000)
Error relocating /usr/lib/lcdproc/glcd.so: report: symbol not found
Error relocating /usr/lib/lcdproc/glcd.so: outb: symbol not found
Error relocating /usr/lib/lcdproc/glcd.so: inb: symbol not found

Symbols for /usr/lib/lcdproc/ula200.so:
	ldd (0x77948000)
	libftdi1.so.2 => /usr/lib/libftdi1.so.2 (0x77916000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x778f8000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x778d4000)
	libc.so => ldd (0x77948000)
Error relocating /usr/lib/lcdproc/ula200.so: report: symbol not found

Symbols for /usr/lib/lcdproc/futaba.so:
	ldd (0x774b6000)
	libusb-0.1.so.4 => /usr/lib/libusb-0.1.so.4 (0x7748e000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x77470000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7744c000)
	libc.so => ldd (0x774b6000)
Error relocating /usr/lib/lcdproc/futaba.so: report: symbol not found

Symbols for /usr/lib/lcdproc/IOWarrior.so:
	ldd (0x77af6000)
	libusb-0.1.so.4 => /usr/lib/libusb-0.1.so.4 (0x77acc000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77aa8000)
	libc.so => ldd (0x77af6000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x77a8a000)
Error relocating /usr/lib/lcdproc/IOWarrior.so: report: symbol not found

Symbols for /usr/lib/lcdproc/shuttleVFD.so:
	ldd (0x77d3a000)
	libusb-0.1.so.4 => /usr/lib/libusb-0.1.so.4 (0x77d12000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77cee000)
	libc.so => ldd (0x77d3a000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x77cd0000)
Error relocating /usr/lib/lcdproc/shuttleVFD.so: report: symbol not found

Symbols for /usr/lib/lcdproc/picolcd.so:
	ldd (0x77754000)
	libusb-0.1.so.4 => /usr/lib/libusb-0.1.so.4 (0x7772a000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x7770c000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x776e8000)
	libc.so => ldd (0x77754000)
Error relocating /usr/lib/lcdproc/picolcd.so: report: symbol not found

Symbols for /usr/lib/lcdproc/i2500vfd.so:
	ldd (0x77244000)
	libftdi1.so.2 => /usr/lib/libftdi1.so.2 (0x77212000)
	libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x771f4000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x771d0000)
	libc.so => ldd (0x77244000)
Error relocating /usr/lib/lcdproc/i2500vfd.so: report: symbol not found

```

neheb, 2019-08-03:
ping @haraldg @pprindeville

haraldg, 2019-08-03:
I don't think there is a problem here. The drivers are supposed to be loaded by dlopen() from the server, where all these symbols should be defined. The drivers aren't expected to run in a stand alone fashion.

damocles-git, 2019-08-07:
I'm sorry I haven't been clear on this report, the error also is given when the LCDd daemon is started:

<pre>
Wed Aug  7 18:06:40 2019 user.err LCDd: Could not open driver module /usr/lib/lcdproc/hd44780.so: Error relocating /usr/lib/lcdproc/hd44780.so: outb: symbol not found
Wed Aug  7 18:06:40 2019 user.err LCDd: Driver [hd44780] binding failed
Wed Aug  7 18:06:40 2019 user.err LCDd: Could not load driver hd44780
Wed Aug  7 18:06:40 2019 user.err LCDd: There is no output driver
Wed Aug  7 18:06:40 2019 user.crit LCDd: Critical error while initializing, abort.
</pre>

since the 'outb' function is a system function I think that something is missing, even compiling the package using the openwrt sdk returns this error.

neheb, 2019-08-07:
That is an actual bug.

haraldg, 2019-08-07:
damocles-git writes:

> since the 'outb' function is a system function I think that something is missing, even compiling the package using the openwrt sdk returns this error.

Thanks for the clarification. Actually outb should be inlined.

Maybe you have compiled lcdproc with optimization disabled?
(I think this is a bug in the upstream build system to use outb for
not optimized builds.)

-- 
Nationalratswahl: Ich trete als unabhängiger Experte für Klimaschutz
und freie Software an! Ich will mit Vorzugsstimmen ins Parlament kommen,
weil wenn ich es nicht mache, kümmert sich womöglich niemand darum.
https://haraldgeyer.at

neheb, 2019-08-07:
I looked at musl. outb is only defined for i386 and x86-64.

closed 2019-08-09 by neheb