openwrt/packages #9655: libtalloc - python3 error

MartB, 2019-08-04:
Maintainer: @thess
Environment: (put here arch, model, OpenWrt version)

Description:
Python3 doesnt work with this package version 2.2.0 works but causes other issues.

```
Applying ./patches/100-Remove_libbsd_dependency_check.patch using plaintext: 
patching file lib/replace/wscript
touch /home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14/.prepared_7db349feb615f3f20408b0134713c031_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14/.configured_*
rm -f /home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/target-mipsel_24kc_musl/stamp/.libtalloc_installed
(cd /home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14; echo -e >cache.txt " \nChecking simple C program: OK \nrpath library support: OK \n-Wl,--version-script support: OK \nChecking getconf LFS_CFLAGS: NO \nChecking for large file support without additional flags: OK \nChecking for -D_LARGE_FILES: OK \nChecking correct behavior of strtoll: NO \nChecking for working strptime: OK \nChecking for C99 vsnprintf: OK \nChecking for HAVE_SHARED_MMAP: OK \nChecking for HAVE_MREMAP: OK \nChecking for HAVE_INCOHERENT_MMAP: NO \nChecking for HAVE_SECURE_MKSTEMP: OK \nChecking for HAVE_IFACE_GETIFADDRS: OK \nChecking for kernel change notify support: OK \nChecking for Linux kernel oplocks: OK \nChecking for kernel share modes: OK \nChecking if can we convert from CP850 to UCS-2LE: OK \nChecking if can we convert from UTF-8 to UCS-2LE: OK \nChecking whether we can use Linux thread-specific credentials with 32-bit system calls: OK \nChecking whether we can use Linux thread-specific credentials: OK \nChecking whether setreuid is available: OK \nChecking whether setresuid is available: OK \nChecking whether seteuid is available: OK \nChecking whether fcntl locking is available: OK \nChecking for the maximum value of the 'time_t' type: OK \nChecking whether the realpath function allows a NULL argument: OK \nChecking whether POSIX capabilities are available: OK \nChecking for ftruncate extend: OK \nvfs_fileid checking for statfs() and struct statfs.f_fsid: OK \ngetcwd takes a NULL argument: OK \nChecking value of NSIG: \"65\" \nChecking value of _NSIG: \"65\" \nChecking value of SIGRTMAX: \"64\" \nChecking value of SIGRTMIN: \"34\" \nChecking if toolchain accepts -fstack-protector: OK \n" " \nChecking uname machine type: \"mipsel\" \nChecking uname release type: \"4.14.134\" \nChecking uname sysname type: \"Linux\" \nChecking uname version type: \"4.14.134\" \n" ; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -O2 -pipe -mno-branch-likely -march=1004kc -fno-caller-saves -fno-plt -fomit-frame-pointer -DNDEBUG -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14=talloc-2.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-O2 -pipe -mno-branch-likely -march=1004kc -fno-caller-saves -fno-plt -fomit-frame-pointer -DNDEBUG -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14=talloc-2.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-O2 -pipe -mno-branch-likely -march=1004kc -fno-caller-saves -fno-plt -fomit-frame-pointer -DNDEBUG -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14=talloc-2.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/target-mipsel_24kc_musl/include -I/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/usr/include -I/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/include/fortify -I/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/include " LDFLAGS="-L/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/target-mipsel_24kc_musl/lib -L/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/usr/lib -L/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/lib -znow -zrelro "  ./buildtools/bin/waf configure --prefix=/usr --sysconfdir=/etc --localstatedir=/var --with-libiconv="" --cross-compile --cross-answers=/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14/cache.txt --disable-rpath --disable-rpath-install )
Traceback (most recent call last):
  File "./buildtools/bin/waf", line 75, in <module>
    import Scripting
  File "/home/martb/Dokumente/customfirmwares/openwrt-lam/build/erx/build_dir/target-mipsel_24kc_musl/talloc-2.1.14/third_party/waf/wafadmin/Scripting.py", line 146
    except Utils.WafError, e:
                         ^
SyntaxError: invalid syntax
```

This package is used by freeradius3 so all users of this package are affected by this dependency issue.

MartB, 2019-08-07:
Am i doing sth wrong or why is seemingly noone interested in this package, can i use freeradius without it or is there an alternative for it?

yogo1212, 2019-08-09:
@MartB i'm having the same problem
but before i looked here i started patching the various errors....
work for sisyphus. here's my progress:
https://github.com/openwrt/packages/pull/9686

neheb, 2020-01-17:
ping @Andy2244 got any idea?

I ask since this shares samba code.

Andy2244, 2020-01-17:
Yes as i stated in #9686 they would have needed the same cross-compile patch, but this was now merged upstream. So building from latest source should work, also check my samba makefile, there is a new answer i needed specifically four our buildbot to pass.

Robby-, 2020-01-24:
I too am having the same problem, trying to compile the freeradius3 package.

neheb, 2020-01-25:
This should be rebased when https://github.com/openwrt/packages/pull/9686 gets merged.

Robby-, 2020-02-04:
This issue can probably be closed now as this was fixed by #9686.

closed 2020-02-04 by neheb