openwrt/packages #9716: CircleCI: Builds have been failing

jefferyto, 2019-08-13:
Looks like the first build to fail with the same/similar error is https://circleci.com/gh/openwrt/packages/4600:

```
src-git base https://github.com/openwrt/openwrt.git;master
src-link packages /home/build/openwrt_packages
src-git luci https://github.com/openwrt/luci.git;master
Updating feed 'base' from 'https://github.com/openwrt/openwrt.git;master' ...
Cloning into './feeds/base'...
remote: Enumerating objects: 9087, done.        
remote: Counting objects: 100% (9087/9087), done.        
remote: Compressing objects: 100% (7626/7626), done.        
remote: Total 9087 (delta 1626), reused 4800 (delta 829), pack-reused 0
Receiving objects: 100% (9087/9087), 12.33 MiB | 10.95 MiB/s, done.
Resolving deltas: 100% (1626/1626), done.
Create index file './feeds/base.index' 
make: *** [staging_dir/host/.prereq-build] Error 1
Collecting target info: doneUpdating feed 'packages' from '/home/build/openwrt_packages' ...
Create index file './feeds/packages.index' 
make: *** [staging_dir/host/.prereq-build] Error 1
Collecting target info: doneUpdating feed 'luci' from 'https://github.com/openwrt/luci.git;master' ...
Cloning into './feeds/luci'...
Warning: Permanently added the RSA host key for IP address '140.82.113.3' to the list of known hosts.

remote: Enumerating objects: 4140, done.        
remote: Counting objects: 100% (4140/4140), done.        
remote: Compressing objects: 100% (2105/2105), done.        
Receiving objects: 100% (4140/4140), 4.16 MiB | 0 bytes/s, done.
remote: Total 4140 (delta 915), reused 2998 (delta 562), pack-reused 0
Resolving deltas: 100% (915/915), done.
Checking out files: 100% (2252/2252), done.
Create index file './feeds/luci.index' 
make: *** [staging_dir/host/.prereq-build] Error 1
Collecting target info: donemake: *** [staging_dir/host/.prereq-build] Error 1
Exited with code 2
```

karlp, 2019-08-13:
This was (completely?) addressed in https://github.com/openwrt/packages/pull/9659/commits/a6d5b5d1c1c6ea922afcfff1efac7fb55c9664c4 but yeah, still ongoingi I think

jefferyto, 2019-08-13:
That's part of #9659 which hasn't been merged yet.

karlp, 2019-08-13:
I know, I'm just linking it as a fix is enroute for this.

jefferyto, 2019-08-13:
@karlp Thanks for linking - I was updating my local repo and wondering why I wasn't seeing this fix :laughing:

karlp, 2019-08-14:
This has merged now, can we close this one?

jefferyto, 2019-08-14:
Sure

closed 2019-08-14 by jefferyto