openwrt/packages #9723: It seems that there's no php* packages in the newest opkg source

guzhiyong, 2019-08-14:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
I have updated to the newest source and when I try _opkg list php*_ I got nothing.
It seems that there's no php packages in the newest opkg source
```
Formating code blocks by wrapping them with pairs of ```
```

mhei, 2019-08-14:
You must describe in more detail, what you've done. The issue template helps you to fill in expected information to allow supporting you, also the helping texts should be removed before submitting the issue. Please fix this.
BTW: did you run "opkg update" before "opkg list..."?

neheb, 2019-08-14:
Yeah that was my bad. I messed up when backporting an icu patch, which broke PHP. It should be fine soon.

mhei, 2019-08-20:
No response from @guzhiyong, closing since this was probably a temporary issue.

closed 2019-08-20 by mhei