openwrt/packages #9730: prometheus-node-exporter-lua: Please add 'architecture'

systemcrash, 2019-08-14:
Ping @klukonin

When scraping, I get e.g.:

```
...
node_openwrt_info{revision="r7808-ef686b7292",model="TP-LINK Archer C7 v4",id="OpenWrt",board_name="archer-c7-v4",release="18.06.4"} 1
...
```

Could someone please add the architecture field? Like what we see when logging in to the device via GUI:

Model blah
Architecture blah
Firmware Version blah
Kernel Version blah

producing e.g.

```
...
node_openwrt_info{revision="r7808-ef686b7292",model="TP-LINK Archer C7 v4",architecture="Qualcomm Atheros QCA9558 ver 1 rev 0",id="OpenWrt",board_name="archer-c7-v4",release="18.06.4"} 1
...
```

If ```architecture``` doesn't belong under ```node_openwrt_info``` I can live with it anywhere else.

systemcrash, 2019-08-14:
Architecture seems to be pulled like so:

https://github.com/openwrt/luci/blob/36f226548a4b0eee66695a163c728089e60d28ca/modules/luci-mod-status/luasrc/view/admin_status/index/10-system.htm

This references ```boardinfo.system```

systemcrash, 2019-08-14:
I guess that stuff goes here:

https://github.com/openwrt/packages/blob/master/utils/prometheus-node-exporter-lua/files/usr/lib/lua/prometheus-collectors/openwrt.lua

systemcrash, 2019-08-14:
Also located when running:
```ubus call system board```

klukonin, 2019-08-14:
Yes, it's possible.
I'm not sure about which method would be the best.
May be ubus call will be optimal.

Also you should call a maintainer of this package.
I'm just a guy ho did a small commit for personal purposes.

PR #9739 (2019-08-15): prometheus-node-exporter-lua: add target & system to OpenWrt collector
Maintainer: me
Compile & run tested: GL.iNet GL-AR150 OpenWrt SNAPSHOT r10756+1-7546be6007

Description:
Before:
node_openwrt_info{revision="r10756+1-7546be6007",model="GL.iNet GL-AR150",id="OpenWrt",board_name="glinet,gl-ar150",release="SNAPSHOT"} 1

After:
node_openwrt_info{revision="r10756+1-7546be6007",target="ath79/generic",board_name="glinet,gl-ar150",id="OpenWrt",model="GL.iNet GL-AR150",release="SNAPSHOT",system="Atheros AR9330 rev 1"} 1

Fixes #9730, replace #9735
commit a14bed1b (2019-08-15): prometheus-node-exporter-lua: add target & system to OpenWrt collector

Before:
node_openwrt_info{revision="r10756+1-7546be6007",model="GL.iNet GL-AR150",id="OpenWrt",board_name="glinet,gl-ar150",release="SNAPSHOT"} 1

After:
node_openwrt_info{revision="r10756+1-7546be6007",target="ath79/generic",board_name="glinet,gl-ar150",id="OpenWrt",model="GL.iNet GL-AR150",release="SNAPSHOT",system="Atheros AR9330 rev 1"} 1

Fixes #9730, replace #9735
PR #9751 (2019-08-16): [19.07] prometheus-node-exporter-lua: add target & system to OpenWrt collector
Maintainer: me
Compile & run tested: GL.iNet GL-AR150 / OpenWrt 19.07-SNAPSHOT r10306+1-d97c6d9f09

Description:

Before:
node_openwrt_info{revision="r10756+1-7546be6007",model="GL.iNet GL-AR150",id="OpenWrt",board_name="glinet,gl-ar150",release="SNAPSHOT"} 1

After:
node_openwrt_info{revision="r10756+1-7546be6007",target="ath79/generic",board_name="glinet,gl-ar150",id="OpenWrt",model="GL.iNet GL-AR150",release="SNAPSHOT",system="Atheros AR9330 rev 1"} 1

Fixes #9730, replace #9735

(cherry picked from commit a14bed [... cut]
commit f07e5b5b (2019-08-16): prometheus-node-exporter-lua: add target & system to OpenWrt collector

Before:
node_openwrt_info{revision="r10756+1-7546be6007",model="GL.iNet GL-AR150",id="OpenWrt",board_name="glinet,gl-ar150",release="SNAPSHOT"} 1

After:
node_openwrt_info{revision="r10756+1-7546be6007",target="ath79/generic",board_name="glinet,gl-ar150",id="OpenWrt",model="GL.iNet GL-AR150",release="SNAPSHOT",system="Atheros AR9330 rev 1"} 1

Fixes #9730, replace #9735

(cherry picked from commit a14bed1bc959116617b0eb75d2d9a46551a480e6)
closed 2019-08-15 by champtar