openwrt/packages #9740: libgd: please add webp support by default

eliz82, 2019-08-15:
Package: libgd https://github.com/openwrt/packages/blob/master/libs/libgd/Makefile
Maintainer: @jow- / Jo-Philipp Wich <jo@mein.io>
Environment: Netgear R7800 with Nginx and PHP

Description:
As Microsoft Edge added WebP support in November 2018 and Mozilla Firefox added support in January 2019 , WebP format is almost universally supported on client side.  https://caniuse.com/#feat=webp

To start to adopt this more efficient image format, server side support is also necessary. 
The package php7-mod-gd use libgd.

Dependencies:
https://github.com/libgd/libgd
https://github.com/webmproject/libwebp

jow-, 2019-08-15:
Can you provide a patch? I won't have time to work on that.

rmandrad, 2019-09-14:
compilation on wrt3200acm (mvebu) fails .. fixed by adding --disable-neon to the makefile

stefaang, 2019-09-27:
@neheb I see you've disabled several libs in libwebp... I'll have a look for a libwebp-full package including the cwebp converter tool.

neheb, 2019-09-27:
Which do you need?

stefaang, 2019-09-28:
I'd be mainly interested in jpeg --> webp for now.

mhei, 2020-01-12:
My last upgrade of PHP probably broke your use-case since PHP now relies on the external libgd and not anymore on the bundled/internal library.
Meanwhile we also have a libgd-full package, maybe this helps to bake a nice dependency chain without making the whole thing too complex...

neheb, 2020-02-23:
Seems everything got merged. Closing.

closed 2020-02-23 by neheb