openwrt/packages #9749: opkg:  dnsmasq conflicts with dnsmasq-full

fantom-x, 2019-08-15:
Environment: Image builder with 19.07-SNAPSHOT for Gl.iNet gl-ar750

I am still trying to figure out what package got a new dependency (**dnsmasq-full**) recently, but now an image cannot be put together without explicitly removing **dnsmasq** from the package list first like below:

**" -dnsmasq dnsmasq-full "**

```
Collected errors:
 * check_data_file_clashes: Package dnsmasq wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/etc/hotplug.d/ntp/25-dnsmasqsec
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/etc/init.d/dnsmasq
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/usr/lib/dnsmasq/dhcp-script.sh
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/usr/sbin/dnsmasq
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/usr/share/acl.d/dnsmasq_acl.json
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/usr/share/dnsmasq/dhcpbogushostname.conf
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /build/openwrt-imagebuilder/build_dir/target-mips_24kc_musl/root-ar71xx/usr/share/dnsmasq/rfc6761.conf
	But that file is already provided by package  * dnsmasq-full
 * opkg_install_cmd: Cannot install package dnsmasq.
```

fantom-x, 2019-08-15:
FYI, looks like **travelmate** pulls in **dnsmasq-full** during the packaging with the image builder.

dibdot, 2019-08-16:
let's close this one ... travelmate explicit depends on "dnsmasq", seems to be a problem in imagebuilder.

closed 2019-08-16 by dibdot

fantom-x, 2019-08-16:
This is a problem on its own. What if I want to add dnsmaq-full, but in order to make the firmware build I will need to explicitly remove dnsmasq. I have created several issue like that for other packages and they were fixed eventually.

dibdot, 2019-08-16:
dnsmasq is a core package and not hosted here in the package repo. If you're still convinced there's a problem here, then open a ticket at bugs.openwrt.org (same applies to your image builder ticket).

acecilia, 2020-03-27:
Hello, just hit this issue myself. I solved it by running (note `-dnsmasq` after `luci-app-travelmate`):

```
make image PROFILE=asl26555-16M PACKAGES="uhttpd uhttpd-mod-ubus libiwinfo-lua luci-base luci-app-firewall luci-mod-admin-full luci-theme-bootstrap zram-swap -ppp -ppp-mod-pppoe -iptables -ip6tables -odhcp6c -kmod-ipv6 -kmod-ip6tables -odhcpd -opkg relayd luci-proto-relay wpad -wpad-mini luci-app-travelmate -dnsmasq"
```

hmh, 2022-12-28:
A few years later, and it only got worse. On OpenWRT 22, to even *compile* dnsmasq-full, you *must* use the Image Builder.  The only way the compile finishes is to have both dnsmasq and dnsmasq-full as *modules*, the build breaks should you try to install to the image either one.

This means any images you tell the build to generate will lack dnsmasq(!).

You will also have this kind of breakage on ustream-* (colliding with ustream-wolfssl), etc.

There are two workarounds: (1) give up on non-image-builder usage of OpenWRT.  This is clearly what happened in dev-land, otherwise this kind of breakage would never survived for a month, let alone made it to a stable release.  And in *some* cases, you have (2): give up on per-device rootfs, and build for a single device -- this one I have not tried yet, but it looks like it might work from the braindamage that forces DEFAULT_* packages to be either {M} or {*}, and triggers the bug.