openwrt/packages #9750: opkg: image builder pulls in dnsmasq-full instead of dnsmasq

fantom-x, 2019-08-16:
The latest version of **travelmate** has introduced an explicit dependency on **dnsmasq**, but the image builder is adding **dnsmasq-full** to the image instead. That increases the image size by ~0.5MB and it actually does not work without a workaround #9749 .

dibdot, 2019-08-16:
Strange, maybe a problem in imagebuilder with "PROVIDES"!? I don't now ... anyway, as a workaround you should set explicit dnsmasq in first place as well, e.g.:

`make image PROFILE="glinet_gl-ar750s" PACKAGES="dnsmasq travelmate"`

fantom-x, 2019-08-16:
Thx, your workaround worked very well. But I would still like to keep the other issue open as it is a separate one: dnsmasq-full can be added explicitly and then the build breaks.

This issue is about pulling in **dnsmasq-full** instead of **dnsmasq**.

dibdot, 2019-08-29:
Please open a image builder related ticket at bugs.openwrt.org.

closed 2019-08-29 by dibdot