openwrt/packages #9753: miniupnpd: Unable to add IPv6 Firewall rules (AddPinhole)

ffrediani, 2019-08-16:
When trying to add IPv6 Firewall rules on OpenWrt 17.01.6 which runs miniupnpd version 2.0.20170421-2 it doesn't seem to work as at all.

I used the miniupnpc client from http://miniupnp.free.fr/ and run the following example command:
upnpc -6 -A "" "" 2001:db8:1234::5678 12345 tcp 300

The two error messages I get are:
AddPinhole([]: -> [2001:db8:1234::5678]:12345) failed with code -3 (UnknownError)
AddPinhole([]: -> [2001:db8:1234::5678]:12345) failed with code 401 (Invalid Action)

When looking at the miniupnpd daemon logs in the router I see:
"daemon.notice miniupnpd[3622]: SoapMethod: Unknown: AddPinhole"

Although for miniupnpd version 2.0 doesn't have the option specifically to choose IGD version to advertise - v1 or v2 - (differently from miniupnpd 2.1 in OpenWrt 18.06 which does), it is possible to see in the logs that miniupnpd 2.0 listen also on a IPv6 Address.
As support for IPv6 was added to miniupnpd back in 2012 I suspect that maybe binary has not been compiled with --igd2 option as discussed in their forum.

As this package is optional and installed only after the generic factory/sysupgrade firmware deployment would it be the case to upgrade only the version of miniupnpd package to 2.1 used in 17.01 repository so when when it gets installed via opkg it has the same behaviour of 18.06 which works better for adding IPv6 rules dynamically.

Obs: I don't think is fair or right to disregard any issue that have relation to anything below 18.06 and force people out of 17.01 so abruptly. There has been discussions on the dev mail list about the issues of running 18.06 on many devices gives which are very usable and an extension to 17.01 life until 18.06 is more stable or usable on these devices was discussed as well.
I think this is one of the cases that is simple to solve just by upgrading the package version and doesn't even require any new 17.01.x version builds.

After all 17.01 has full IPv6 support and as such UPnP/PCP should work properly for more coming applications that require that for incoming IPv6 connections and to avoid lay people to add rules manually. Even though there are issues to maintain these rules when IPv6 Prefix Delegation received from the ISP are dynamic.

ffrediani, 2019-08-16:
@ldir-EDB0 @neheb this one as well is a rather different situation which perhaps may be easier to solve by just upgrading the package to version 2.1 or by compiling it with --igd2 option

Reference: https://miniupnp.tuxfamily.org/forum/viewtopic.php?t=1892&highlight=igd2

neheb, 2020-01-24:
Is this still an issue?

ffrediani, 2020-01-24:
Yes it remains. The last message remains valid.

neheb, 2020-01-24:
Not sure I understand. --igd2 as well as --ipv6 are already passed: https://github.com/openwrt/packages/blob/master/net/miniupnpd/Makefile#L50

ffrediani, 2020-01-24:
Are you sure it is passed in 17.01 ?
In 18.06 works fine, but not in 17.01 which is where the issue shows up.

neheb, 2020-01-24:
Ah no. I think 17.01 is EOL at this point.

I just checked. miniupnpd is not in this repository for 17.01.

ffrediani, 2020-01-24:
Well, when it was reported it wasn't EOL.
miniupnpd is part of OpenWrt for a while, including in version 17.01. Maybe the issue is that it is not in a expected folder (ex: ar71xx/generic/packages), but in mips_24kc folder. I have 17.01.x installed here and I use miniupnp. Even in 15.05.1 it is available.

It can be found at: http://downloads.openwrt.org/releases/17.01.4/packages/mips_24kc/packages/

neheb, 2020-01-24:
It's here: https://github.com/openwrt-routing/packages/tree/lede-17.01/miniupnpd

When initially introduced here, igd2 and ipv6 were enabled: https://github.com/openwrt/packages/commit/a2e2eeb409a71027e27b858c27a617bb11557ece#diff-22dddd1975965a186aa1f32d4f345dbfR46

Anyway, I'll close for now. Ask the openwrt-routing people to fix it if they can.

closed 2020-01-24 by neheb

ffrediani, 2020-01-24:
Why are you closing it if you are not sure ?
This is not even a openwrt-routing thing. miniupnpd isn't a routing subject really.

Don't you want to have to deal with this issue and just wish to clean up stuff really ? It remains an issue which can be fixed by the way it was described above.

Did a pretty detailed report with the solution and you simply close it because you couldn't understand the issue ? Is it the proper way of doing things here ?

ffrediani, 2020-01-24:
Why are you closing it if you are not sure ?

Don't you want to have to deal with this issue and just wish to clean up stuff really ? It remains an issue which can be fixed by the way it was described above.

Did a pretty detailed report with the solution and you simply close it because you couldn't understand the issue ? Is it the proper way of doing things here ?

neheb, 2020-01-24:
> In 18.06 works fine, but not in 17.01 which is where the issue shows up.

Note that the last commit in my link is: https://github.com/openwrt-routing/packages/commit/3e916a365214a6700a636bac551c41c8f57db7b3

In any case, there's nothing I can do from my end.

ffrediani, 2020-01-24:
Yes, this was reported for 17.01 and the intent was to fix for it.
There is a major different between 18.06 and 17.01 versions to several hardware and 17.01 is still well used, and sometimes the only version that works fine or these hardware so why I was trying to get it fixed in this version so it could work the same it does in  18.06.

Regarding the commit there it no harm to have it enabled by default which is exactly what 18.06 does as it works fine in that version for IPv6 stuff as described in initial message.

What would be the issue for fixing it for 17.01 ? Perhaps I couldn't get it fully yet.

neheb, 2020-01-24:
The proper place to report would be https://github.com/openwrt-routing/packages

I have no commit access there. The amount of issues involving miniupnpd there is worrying.

neheb, 2020-01-24:
The packages feed is listed above the routing feed in feeds.conf.default, but even if I import it, I don't think the buildbots would build anything.