openwrt/packages #9761: lighttpd: mod_webdav unable to COPY or MOVE

languagegame, 2019-08-16:
Maintainer: @flyn-org 
Packages: lighttpd and lighttpd-mod-webdav
Environment: openwrt-19.07 and master

Description:
Please upgrade the lighttpd packages to the latest version 1.4.54, which incorporates fixes for the mod_webdav module.  The current version 1.4.53 has a bug in the mod_webdav so that it does not correctly support COPY or MOVE requests over HTTP.

Here are more details. Version 1.4.53 of lighttpd has a bug in mod_webdav in terms of how it parses the "Destination:" HTTP header.  The parsing engine assumes the uri would have an extra slash after the hostname at the front, which is incorrect.  This bug makes the module fail for any COPY or MOVE request over the webdav protocol.  This bug was fixed in 1.4.54, so I'm hoping that whoever maintains this package can upgrade it to the most recent version for openwrt-19.07 and master.

The new version of lighttpd is 1.4.54 and its mod_webdav module fixes this.  Please update the package for branches 19.07 and master.

languagegame, 2019-08-16:
I uncovered this bug in version 1.4.53 by using the webdav litmus test.  I also replicated the bug using a Windows 10 client.  In both cases, creating and deleting files worked correctly, but any COPY or MOVE command (which requires a "Destination:" http header in the request) fails.  

I then tested to manually change the Destination header and resubmit, and it was able to work by adding an extra slash in the uri, which is incorrect for HTTP but works in this version of lighttpd.  For example, a header of "Destination: http://hostname/filename" fails as the destination target for a COPY command.  However, If I manually modify the header and resubmit with an extra slash like the following, it works, such as "Destination: http://hostname//filename".

This bug was fixed by the author of the program in version 1.4.54, so I'm hoping this can be upgraded in openwrt for the latest 19.07 and master branches.

ja-pa, 2019-09-05:
Should be fixed by #9891

languagegame, 2019-09-06:
Yes, thanks.  Is it possible to get this added to the 19.07 branch as well?

languagegame, 2019-09-08:
Works great on 19.07.  This can be closed

closed 2019-09-08 by languagegame