openwrt/packages #9766: BUG.Single netcard vmware openwrt router upload speed very slow on special case

wood350, 2019-08-18:
package name: openwrt-18.06.4-x86-64-combined-ext4.img.gz
I use a vmware virtual os running openwrt 18.06.4，The physical pc only installed one net card. local area net contains a pc of ip 192.168.1.3,Thers is a ftp server running on it,I just map wan ip port 3939 to 192.168.1.3:21, I found the download speed very slow at wan side,only 60KB/S. I tried a nother physical pc with two netcards runing openwrt 18.06.4, this issue dispeared. the speed at wan side up to 2MB/S。

DazzyWalkman, 2019-08-20:
This issue is outside the scope of this openwrt package repo.
It's all about your vmware host version and type of virtualized nic for openwrt guest.
VMXNET2/3 are prefered.
For detail, see https://kb.vmware.com/s/article/1001805

diizzyy, 2019-08-23:
Closed, not related to package repo

closed 2019-08-23 by diizzyy