openwrt/packages #9767: [dhcrelay] Cant use ppp* as upstream interface (Unsupported device type 512 for "pptp-vpn")

LipkeGu, 2019-08-18:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (lantiq, xrx200 (FB7360-SL), 19-07)

Description:
as defined in the config (uci) we cant use ppp0 interfaces as upstream interfaces ex: for wan.

```Internet Systems Consortium DHCP Relay Agent 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/
Listening on LPF/eth0.4/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.4/9c:c7:a6:e8:63:28
Listening on LPF/eth0.3/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.3/9c:c7:a6:e8:63:28
Listening on LPF/eth0.2/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.2/9c:c7:a6:e8:63:28
Listening on LPF/eth0.1/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.1/9c:c7:a6:e8:63:28
Unsupported device type 512 for "pptp-vpn"

If you think you have received this message due to a bug rather
than a configuration issue please read the section on submitting
bugs on either our web page at www.isc.org or in the README file
before submitting a bug.  These pages explain the proper
process and the information we find helpful for debugging.

exiting.
root@fb7360:~#

```

LipkeGu, 2019-08-18:
It seems dhcrelay doesnt listening on ppp interfaces

https://dhcp-users.isc.narkive.com/PDJUgg48/dhcp-relay-listening-on-a-machine-containing-ppp#post11

LipkeGu, 2019-08-19:
I ported the fedora-patch to openwrt and recompiled the image...
with this patch dhcrelay stays now connected on ppp interfaces like dsl or pptp

`--- a/common/bpf.c	2018-02-21 15:30:46.000000000 +0100
+++ b/common/bpf.c	2019-08-18 21:07:37.000000000 +0200
@@ -650,6 +650,16 @@
                         memcpy(&hw->hbuf[1], LLADDR(sa), sa->sdl_alen);
                         break;
 #endif /* IFT_FDDI */
+#if defined(IFT_PPP)
+                case IFT_PPP:
+			hw->hlen = 0;
+			hw->hbuf[0] = HTYPE_RESERVED;
+			hw->hbuf[1] = 0xde;
+			hw->hbuf[2] = 0xad;
+			hw->hbuf[3] = 0xbe;
+			hw->hbuf[4] = 0xef;
+			break;
+#endif
                 default:
                         log_fatal("Unsupported device type %d for \"%s\"",
                                   sa->sdl_type, name);
--- a/common/discover.c	2018-02-21 15:30:46.000000000 +0100
+++ b/common/discover.c	2019-08-18 20:30:25.000000000 +0200
@@ -629,8 +629,7 @@
 		    (local_family == AF_INET6 &&
 		    !(info.flags & IFF_MULTICAST)) ||
 #endif
-		      info.flags & IFF_LOOPBACK ||
-		      info.flags & IFF_POINTOPOINT) && !tmp) ||
+		      info.flags & IFF_LOOPBACK) && !tmp) ||
 		    (!(info.flags & IFF_UP) &&
 		     state != DISCOVER_UNCONFIGURED))
 			continue;
--- a/common/lpf.c	2018-02-21 15:30:46.000000000 +0100
+++ b/common/lpf.c	2019-08-18 20:39:46.000000000 +0200
@@ -563,6 +563,16 @@
 			hw->hbuf[0] = HTYPE_FDDI;
 			memcpy(&hw->hbuf[1], sa->sa_data, 6);
 			break;
+#if defined(ARPHRD_PPP)
+	        case ARPHRD_PPP:
+			hw->hlen = 0;
+			hw->hbuf[0] = HTYPE_RESERVED;
+			hw->hbuf[1] = 0xde;
+			hw->hbuf[2] = 0xad;
+			hw->hbuf[3] = 0xbe;
+			hw->hbuf[4] = 0xef;
+			break;
+#endif
 		default:
 			log_fatal("Unsupported device type %ld for \"%s\"",
 				  (long int)sa->sa_family, name);
`

```
root@fb7360:~# /etc/init.d/dhcrelay4 restart
Requesting: pptp-vpn as upstream: Y downstream: N
Requesting: eth0.1 as upstream: N downstream: Y
Requesting: eth0.2 as upstream: N downstream: Y
Requesting: eth0.3 as upstream: N downstream: Y
Requesting: eth0.4 as upstream: N downstream: Y
Internet Systems Consortium DHCP Relay Agent 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/
Listening on LPF/eth0.4/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.4/9c:c7:a6:e8:63:28
Listening on LPF/eth0.3/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.3/9c:c7:a6:e8:63:28
Listening on LPF/eth0.2/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.2/9c:c7:a6:e8:63:28
Listening on LPF/eth0.1/9c:c7:a6:e8:63:28
Sending on   LPF/eth0.1/9c:c7:a6:e8:63:28
Listening on LPF/pptp-vpn/
Sending on   LPF/pptp-vpn/
Sending on   Socket/fallback
Forwarded BOOTREQUEST for 00:13:77:3c:e9:30 to 10.20.0.1
Forwarded BOOTREQUEST for 00:13:77:3c:e9:30 to 10.20.0.1
Forwarded BOOTREQUEST for 00:13:77:3c:e9:30 to 10.20.0.1
Forwarded BOOTREQUEST for 00:13:77:3c:e9:30 to 10.20.0.1
```

neheb, 2020-01-17:
@pprindeville is the maintainer.

Please add a PR if needed.