openwrt/packages #9770: kea: Issue stopping, reloading and showing the status using keactrl

tiagogaspar8, 2019-08-18:
Maintainer: BangLang Huang<banglang.huang@foxmail.com>, Rosy Song<rosysong@rosinson.com>
Environment: ARM Latest OpenWRT Trunk (from two weeks ago)

Description:

When I run keactrl stop, reload or status I get this (differs a bit from command to command:

```
root@OpenWrt:~# keactrl status
ps: unrecognized option: p
BusyBox v1.31.0 () multi-call binary.

Usage: ps

Show list of processes

        w       Wide output
DHCPv4 server: inactive
ps: unrecognized option: p
BusyBox v1.31.0 () multi-call binary.

Usage: ps

Show list of processes

        w       Wide output
DHCPv6 server: inactive
DHCP DDNS: inactive
ps: unrecognized option: p
BusyBox v1.31.0 () multi-call binary.

Usage: ps

Show list of processes

        w       Wide output
Control Agent: inactive
Kea DHCPv4 configuration file: /etc/kea/kea-dhcp4.conf
Kea DHCPv6 configuration file: /etc/kea/kea-dhcp6.conf
Kea DHCP DDNS configuration file: /etc/kea/kea-dhcp-ddns.conf
Kea Control Agent configuration file: /etc/kea/kea-ctrl-agent.conf
keactrl configuration file: /etc/kea/keactrl.conf
```

neheb, 2019-08-18:
ping @rosysong

neheb, 2019-09-10:
ping @rosysong @hbl0307106015

neheb, 2019-09-10:
Error comes from here: https://github.com/isc-projects/kea/blob/e34d0e7048b3a30b8e4ba45c751afdb0b335ca9b/src/bin/keactrl/keactrl.in#L140

One way to work around this is to install procps-ng-ps

tiagogaspar8, 2019-09-10:
Then shouldn't procps-ng-ps be in the dependencies of keactrl ?

neheb, 2019-09-10:
It was probably an oversight. I'm sure there's an alternative. But for right now, that's the solution.

hbl0307106015, 2019-09-16:
@neheb @tiagogaspar8  Well, the check_running() use 'ps -p'  which does not support on busybox/ps.
I think there's another method for checking whether a process is running,
1. Creating a xxx.pid file at /var/run directory when start the program. (the program and procd script can do this)

2. In checking(), just read the xxx.pid file and verify if /proc/${pid} directory exist

@neheb What do you think ?

neheb, 2019-09-16:
Sounds sensible.

tiagogaspar8, 2019-09-16:
I saw that BusyBox doesn't support it after some search, I've also tried to add procps-ng-ps as a dependency of kea but it selects the whole procps-ng package Wich we don't need.

So your suggestion is make a patch?
Also we should update kea to the last version wile at it, I've tried to change the version but patches need to be modified

hbl0307106015, 2019-09-18:
@tiagogaspar8  and how about make a pull request to the kea ?

tiagogaspar8, 2019-09-22:
I did it this friday

tiagogaspar8, 2019-10-08:
Issue fixed.

closed 2019-10-08 by tiagogaspar8